melexis/mlx90640-library #2: Address space is not auto incrementing?

nseidle, 2018-06-04:
Another hiccup I hit implementing the driver: On many I2C sensors the read pointer is auto-incrementing between blocks of reads. I don't believe the MLX90640 auto increments the address pointer so you need to write the address before every read.

I don't believe this is documented in the datasheet or the driver. A note about this in the datasheet (or driver) would be helpful for those who are writing drivers for other platforms.

For example: You cannot read 50 bytes like this

```
I2C_read()
{
    write address of the location to read
    stop

    read 32 bytes (max buffer size on many arduino implementations)
    read 18 bytes (fails because address pointer has reset to 0x00)
}
```
This is how I implemented it for a Teensy 3.5:

```
//Read a number of words from startAddress. Store into Data array.
//Returns 0 if successful, -1 if error
int MLX90640_I2CRead(uint8_t _deviceAddress, unsigned int startAddress, unsigned int nWordsRead, uint16_t *data)
{

  //Caller passes number of 'unsigned ints to read', increase this to 'bytes to read'
  uint16_t bytesRemaining = nWordsRead * 2;

  //It doesn't look like sequential read works. We need to re-issue the address command each time

  uint16_t dataSpot = 0; //Start at beginning of array

  //Setup a series of chunked I2C_BUFFER_LENGTH byte reads
  while (bytesRemaining > 0)
  {
    Wire.beginTransmission(_deviceAddress);
    Wire.write(startAddress >> 8); //MSB
    Wire.write(startAddress & 0xFF); //LSB
    if (Wire.endTransmission(false) != 0) //Do not release bus
    {
      Serial.println("No ack read");
      return (0); //Sensor did not ACK
    }

    uint16_t numberOfBytesToRead = bytesRemaining;
    if (numberOfBytesToRead > I2C_BUFFER_LENGTH) numberOfBytesToRead = I2C_BUFFER_LENGTH;

    Wire.requestFrom((uint8_t)_deviceAddress, numberOfBytesToRead);
    if (Wire.available())
    {
      for (uint16_t x = 0 ; x < numberOfBytesToRead / 2; x++)
      {
        //Store data into array
        data[dataSpot] = Wire.read() << 8; //MSB
        data[dataSpot] |= Wire.read(); //LSB

        dataSpot++;
      }
    }

    bytesRemaining -= numberOfBytesToRead;

    startAddress += numberOfBytesToRead / 2;
  }

  return (0); //Success
}
```

slavysis, 2018-06-07:
I have added a note in the MLX90640 driver document - chapter 2.3.3

Best regards

mranostay, 2018-10-31:
@slavysis  Was wonder if you could help explain this output with the regmap subsystem (writing linux support in the video-i2c.c v4l2 driver currently). So all the pixel data has 0xFF for the LSB always, and was wonder if that is normal? NOTE: the reads are starting at 0x400 (it is just a hexdump output).

Reference: https://gist.github.com/d68eeb49a7f89ccb1568cd8043e3c05d

mranostay, 2018-11-08:
That is just the raw data from the sensor pointed at objects at room temperature. Haven't gotten around to processing the temperature data.. still need to write that userspace support. 

Ref: https://patchwork.linuxtv.org/patch/52679/

slavysis, 2018-11-08:
For a room temperature object, having all negative values is normal.

Best regards