melexis/mlx90640-library #4: Sample data to validate API

adibacco, 2018-06-14:
In reference to the following code snippet:

_static uint16_t eeMLX90640[832];
static uint16_t mlx90640Frame[834];
paramsMLX90640 mlx90640;
static float mlx90640To[768];
MLX90640_DumpEE (slaveAddress, eeMLX90640);
MLX90640_ExtractParameters(eeMLX90640, &mlx90640);
MLX90640_GetFrameData (0x33, mlx90640Frame);
tr = MLX90640_GetTa(mlx90640Frame, &mlx90640) – TA_SHIFT; 
MLX90640_CalculateTo(mlx90640Frame, &mlx90640, emissivity, tr, mlx90640To);_ 

It would be really nice if you could upload a file with the **eeMLX90640, mlx90640Frame, mlx90640, mlx90640To** and **tr**

Just as test to prove that the API are doing their job.

slavysis, 2018-06-15:
That is a great suggestion and I am currently working on it. I will upload it soon.

Best regards

adibacco, 2018-06-16:
Just to start I'm attaching eeprom contents as reported in MPLX90640 datasheet "Revision 11, 4 may 18".

[eeData_content.txt](https://github.com/melexis/mlx90640-library/files/2108152/eeData_content.txt)

slavysis, 2018-06-18:
I have uploaded a xlsx file with input data - EEPROM data, frame 0 data and frame 1 data in the first sheet.
The second sheet contains all the parameters that should be extracted based on that input EEPROM data.
In the third sheet you can find the object temperatures calculated based on the input data and the extracted parameters. I hope this will be helpful and if you have some suggestions how to make this file
a better tool, please let me know.

Best regards

adibacco, 2018-06-22:
Really awesome data. They are very useful, what emissivity did you use?

slavysis, 2018-06-25:
Emissivity is 1.0.

Best regards

closed 2020-10-29 by slavysis