melexis/mlx90640-library #5: Steps to read data onto Raspberry Pi?

LoopyFoot, 2018-06-19:
Hello sir,

I am trying to read the data on a Raspberry Pi, and I am confused about what I need to do in order to take your files and read the data from my MLX90640 sensor.

From a high level, what steps are needed in order to read the data from the MLX90640 using my Raspberry Pi?  I have the sensor successfully connected to my Raspberry Pi using the i2cdetect command, and I see it at address 0x33.

However, after that, I cloned your github repository; but I am unsure what to do next.  Any information or support is gratefully appreciated!  For example, do I need to create a Makefile and compile the project in Geany?

abood91, 2018-06-20:
Hello @LoopyFoot 
If you can connect melexis to the existing rpi i2c then you could clone my repo and use the code (modify at will ) https://github.com/abood91/RPiMLX90640

LoopyFoot, 2018-06-20:
What does your CMakeLists.txt file look like?  Should this be located in the RPiMLX90640 directory?

abood91, 2018-06-21:
Hello @LoopyFoot  First of all on my previous testing writeBMP did not really work so you need to create that functionality yourself (in python its much easier ). read the main.cpp carefully there is couple of comments added, in general `Colorize()` function writes xxxxxxxx.rgb.txt file to ./img/ folder and `display_Img()` writes xxxxxxxxx.temp.txt to that folder too. check the comments

Gadgetoid, 2018-06-27:
I've got a false coloured, direct-to-framebuffer example up and running on the Raspberry Pi - see: https://github.com/pimoroni/mlx90640-library/blob/master/fbuf.cpp

This uses continuous mode, plus some fine-tuned timing in order to get the data out of the sensor with the fewest retries and the most sleep period. It works up to 64hz but that's really pushing the limit (and the resulting image is quite noisy). It sets the Pi's i2c bus to 1mhz, which, is incompatible with SMBus.

Timing is very, very finicky and if you don't sync with the data availability on the sensor then it'll churn through unnecessary i2c reads and just throw the data away. 

I calculate the time it takes to read the sensor and display to the framebuffer so it can sleep for the appropriate amount of time. There's also a slight offset in there, since hitting 31.25ms dead on at 32hz seems to result in the data not being ready (implying the sensor isn't actually running at 32hz but perhaps closer to 31).

Here it is in action: https://twitter.com/Gadgetoid/status/1011984760623763456

LoopyFoot, 2018-06-28:
Thanks for sharing the library!  I think that I have it up and running on my Pi.

I was wondering if you have the functionality to save the array of data in a .csv file?

Gadgetoid, 2018-06-29:
That's one of the first things I wrote before shifting to a framebuffer output- albeit I was outputting binary data and converting it to a series of PNG images and then finally an animated GIF with Python.

It's possible to capture one or more frames and save their output into a binary file. I don't see any reason why you couldn't do the same with CSV although 768 columns of temperature data would be some painful data to process!

What's your end goal with the CSV data? Reading it into another language for processing?

dcomins, 2018-11-09:
Hi Gadgetoid,
Is there any way to add in your display more colours (p.ex: blue) and adapt the display resolution to a 1920 x 1080p screen?

dcomins, 2018-11-09:
The problem is: When I try to interpolate data more dan WIDTH*4 or HEIGHT*4 the display dose not show the entire image.

#define OUTPUT_W (int)(24*n)
#define OUTPUT_H (int)(32*n)

n=4 is the maximum number allowed.

[image]

thresholdout, 2018-12-12:
> Hello @LoopyFoot First of all on my previous testing writeBMP did not really work so you need to create that functionality yourself (in python its much easier ). read the main.cpp carefully there is couple of comments added, in general `Colorize()` function writes xxxxxxxx.rgb.txt file to ./img/ folder and `display_Img()` writes xxxxxxxxx.temp.txt to that folder too. check the comments
> I have added also `#ifdef REMOVE_1H_OLD_FILES` you should enable this only if you want to delete the files that is older than 1 hour

do you know how to read data from mlx90640 with python? thanks!

koks123456, 2019-03-14:
Hello, I'm a newer in using Raspberry pi, can l  ask how to read data from mlx90640 in raspberry pi? Can sir show the step to run the functions code inside this library? thank you

xbst, 2019-10-12:
hello @Gadgetoid. I am using https://github.com/pimoroni/mlx90640-library.git library for my camera on pi 3. I can get the readings and and outputs of step and test.cpp. I want to display the running image using rawrgb or fbub but the fbuf gives me cnt > 4 error continuously. wondering if I need to any better GUI to display or I am doing something wrong. 
[image]

zephyrusws, 2019-10-20:
check /boot/config.txt
dtparam=i2c1_baudrate=1000000

ScotsDon, 2020-07-26:
I have same problem as SushanTiwari above.  I have set dtparam=i2c1_baudrate=1000000 but to no avail.  I don't understand what FBUF actually does but it shouldn't give these errors - someone please advise.

mlx-kva, 2020-08-05:
On a raspberry pi, you might want to consider to use the pure python package:
https://github.com/melexis-fir/mlx9064x-driver-py

ScotsDon, 2020-08-06:
> 
> 
> On a raspberry pi, you might want to consider to use the pure python package:
> https://github.com/melexis-fir/mlx9064x-driver-py

Thanks, I'll definitely try that.  

However, following advice from Pimoroni, I reduced the frame rate from 8 to 4 FPS in the source codes of the various examples that were giving the error.  This basically solves the issue, although I still get a few CNT > 4 errors.  It needs further investigation to assess the cause of the problem since I feel it should do 8 FPS.  So I need to check the I2C baud rate really is 1 Meg and to try to determine if outputting to a 1600x1200 HDMI monitor has any effect - a lower resolution display might work better.

closed 2020-11-03 by slavysis