melexis/mlx90640-library #6: Dump eeprom data inconsistent

intechsysltd, 2018-06-21:
Hello!

Thanks for the API, I'm trying to get it work together with our boards.
After the driver settings, I'm able to read data from the eeprom but it is inconsistent with your example data xlsx.
My test code is only calls MLX90640_DumpEE() and MLX90640_ExtractParameters() methods after i2c init. I attached the picture of the analyser's output, and the first 12 extracted parameters and the data from the eeprom.
The i2c driver's clock is about 375kHz.

[params.txt](https://github.com/melexis/mlx90640-library/files/2122688/params.txt)
[eeprom.txt](https://github.com/melexis/mlx90640-library/files/2122689/eeprom.txt)
[image]

slavysis, 2018-06-21:
Hello,

It is normal that the EEPROM content will not be the same as the one in the sample data as it is downloaded from two different MLX90640 devices. The sample data is meant to be used in the following way: 
 1. Load the sample EEPROM data into the MCU memory allocated for it.
 2. Call the MLX90640_ExtractParameters() function
 3. Download the calculated parameters and compare to the ones in the sample data file.
 4. Load the sample frame 0 data into the MCU memory allocated for the frame data and call
            tr = MLX90640_GetTa() - 8;
            MLX90640_CalculateTo(pFrame, ();
 5. Download the calculated frame temperatures and compare with the ones in the sample data file
 6.  Load the sample frame 1 data into the MCU memory allocated for the frame data and call
            tr = MLX90640_GetTa() - 8;
            MLX90640_CalculateTo(pFrame, ();
 7. Download the calculated frame temperatures and compare with the ones in the sample data file

If all the values you get are the same as the ones in the sample data file, the driver has compiled correctly and all you addressing is intact. All you will have to do is to simply get live data from the sensor and put it in the appropriate memory location.

Best regards

adibacco, 2018-06-25:
In my case, my software is working with sample data  but when I get live data from the sensor, I get a lot of negative temperatures on the pixels that are near the border of the sensor. What could be the problem?

slavysis, 2018-06-25:
Did you check that all the parameters are properly extracted = the same as the ones in the sample data?
When using the live data you should also use the EEPROM of the device that you are reading the live data from and extract the parameters for it.

Best regards

closed 2020-11-03 by slavysis

sachinbs1995, 2021-09-01:
hai
I am using wave shares MLX90640 thermal sensor I am getting error parameters extraction failed, and got to know that EEPROM values are corrupted plz help me to restore the EEPROM values.

nkurochkin, 2022-12-01:
The same problem ! Is there any dump of EEPROM available or paramsMLX90640 structure ?