melexis/mlx90640-library #8: Calibration data

Zdeno5, 2018-07-02:
I have some weird temperatures and alpha parameters. 
What can i make with it?
Some fixed alpha data or...?
With the example data it works fine.
I have 2 sensors and the data are same.
The image is measured on smooth homogen surface.

[image]
[image]
[image]

Zdeno5, 2018-07-02:
https://github.com/Zdeno5/MLX90640/blob/master/DATA.csv

slavysis, 2018-07-02:
I can't find the EEPROM data that you read from the device. Could you please share it as well.

Best regards

Zdeno5, 2018-07-02:
https://github.com/Zdeno5/MLX90640/blob/master/EEPROM.csv

slavysis, 2018-07-03:
Could you please check the performance if you use continuous mode?
In step mode the calibration applies only to certain conditions and you may observe some strange values.

Best regards

Zdeno5, 2018-07-03:
https://github.com/Zdeno5/MLX90640/blob/master/Continousmode.csv

In continuous mode are strange values too.
Is it possible to get data like this?

slavysis, 2018-07-03:
Well, that is strange. Could you please try with the default configuration - 0x1901. 

Best regards

Zdeno5, 2018-07-09:
It works with 0x1901.
How can i get it work with 2 subpages?
I used 0x1A0F, but it gives strange values.

slavysis, 2018-07-09:
The device works with two sub-pages by default. It is alternating the sub-pages automatically (unless programmed not to do so). You need to have the 4 LSBs to b'0001' in order to have alternating sub-pages (0x1901 does that). I would say if you need to,you should only change the refresh rate and/or the resolution. All the other bits would be better to remain at the default values. Other wise you might configure the devices to work in some special mode that generates what you call 'strange values'.

Best regards

closed 2018-07-10 by Zdeno5

pronot, 2019-01-29:
With MLX90640-41 Eval board and the Software for EVB90640, the temperature reading shows room temperature over 600C which below 30C actually. I think somehow EEPROM data is garbled. Is there a way to reset/reload it by the Software for EVB90640?

slavysis, 2019-01-30:
The driver is not writing the configuration in the EEPROM. So if your configuration differs from the one in the EEPROM, you would need to set it every time after POR. Normally, 0x1901 should be the default value in the EEPROM also. Could you please send me a dump of the EEPROM?

pronot, 2019-01-30:
Also with the MLX90640-41 Eval Board, i see 650C for room temperature on the Windows software of the board (image below). EEPROM data may be garbled somehow?

[image]

I exported EEPROM data from MLX90640-41 Eval Board software also. 

[mlx90640-eeprom-export.log](https://github.com/melexis/mlx90640-library/files/2813208/mlx90640-eeprom-export.log)

When i read EEPROM register from my MCU with:

`status = MLX90640_I2CRead(MLX90640_I2C_ADR, 0x240C, 1, &value);`

i read the value as 0x1100

slavysis, 2019-02-01:
It seems like only the configuration value in the EEPROM is corrupted. So, if you just include in your code after POR a line to set 1901 in the register, the sensor should be working just fine.

Heni13, 2019-05-15:
Hello ,
I have a problem dealing with my MLX90640-41 Eval Board :
when using the software the Mesure was blocked between two images and the mesurements are false 
[image]
[image]
it ca be the configuration .
How can i solve this problem?
what is the function or the code line that i can put in the console to have the best  configuration ?
or can i reload a program into the stm32 chip in the eval board to reinitialise the board and have significant mesure.
Thank you

slavysis, 2019-05-15:
As you are using the MLX90640 eval board, could you please contact Melexis sales explaining your issue.

Best regards

Heni13, 2019-05-16:
I am asking if someone knows how we can code the console of the MLX90640 Eval Board 
[image]
or where i can find the commands of this console

Heni13, 2019-05-19:
good morning,
i found the MLX90640 library :   
MLX90640_API.cpp
MLX90640_API.h
MLX90640_I2C_Driver.cpp
MLX90640_SWI2C_Driver.cpp
MLX90640_I2C_Driver.h
but in the drivers there is #include"mbed.h"  and i didn't find this header file  can you help me 
thank you.

slavysis, 2019-05-20:
The I2C driver is platform specific. As MBED is one of the widely used platforms, the published driver uses mbed.h and all related mbed specific IO functions. If you are using a different platform you would have to
change the I2C driver accordingly so that it ensures proper communication with the MLX90640 device. This would most likely mean you will also have to use the proper header files for the platform (MCU) you are using

Best regards

Heni13, 2019-05-20:
I m using STM32f746zg .
so i should change the i2c driver ?
thank you

slavysis, 2019-05-20:
Yes, you will have to change the I2C driver accordingly. Please note that the function parameters and
result should remain the same and only the IO operations and library usage should be different.

Best regards