melexis/mlx90640-library #9: Striped frames

drajmund, 2018-07-06:
I am writing regarding the SparkFun MLX90640 Arduino examples available on GitHub which are based on the mlx90640-library. I have successfully uploaded the Example2_OutputToProcessing sketch to an ESP32 board ("ESP32 Dev Module" from Arduino boards) and it works (parameters: 8Hz, Chess mode, 19 bit), but the frames coming from the ESP32 are striped (see example attached). I have read the documentation and analyzed the driver source code, but I have no clue why it is happening. The same sensor with the Melexis Evaluation board produces perfect frames without stripes on the PC software provided by Melexis (MLxCIRT 90640 3). 

I tried the sketch with different parameters (changing the refresh rate, resolution, mode, timing between subpage readings), but I am now I have no ideas how to fix it. Could you please give me any advice?

[image]

Gadgetoid, 2018-07-06:
I haven't seen anything quite like this- although images are usually pretty noisy and have a chessboard pattern emphasised by anything fast moving. Are you sure it's not the interpolation introducing errors?

abood91, 2018-07-09:
I have came cross issue with different cameras, usually this happens if the cabling is bad or a bit loose. try fiddling with the cables.

Vitt61, 2018-08-30:
HI, try to replace the component to exlcude the hardware problem.
For software I think to values of compensation in eeprom not used.

emaAnfu, 2018-09-06:
Hi, I resolved the same problem placing two capacitors (1u and 100n) between Vdd and ground of the MLX90640 sensor, as its datasheet suggests.

drajmund, 2018-09-06:
Thank you for your advices. I have designed a small panel for the sensor and it works perfectly with it. The wiring was quite bad (too long wires and loosely connected).

Here is the result:
[image]

closed 2020-11-03 by slavysis