melexis/mlx90640-library #14: Defective pixels processing

GeorgeZheng2018, 2018-08-17:
Hi  @slavysis  

Thanks for your excellent library, which is very helpful for me to test and understand the sensor.

In current library, it is only identifying the defective pixels from the EEPROM datas and checking their adjacent pixels. Do you have the plan to add the defective pixels processing in the library?

I remember  you mentioned that max I2C clock freq.  is 400KHz for reading Mlx90640's EEPROM info. and 1MHz for reading RAM info. It is not found in MLX90640's specification. Why does it need to limit
the communication rate for EEPROM reading? 

Thanks.

slavysis, 2018-08-23:
I am planning to add defective pixels processing, but I cannot tell when it will be available.
The limitation is required to guarantee stable readings from the EEPROM. Thanks for the 
specification tip.

Best regards

cimba007, 2018-09-29:
I also have a defective pixel. Keep the good work on this excellent library up @slavysis  ;-)

slavysis, 2018-10-30:
There is already a function for correcting the defective pixels.

Best regards

closed 2020-11-03 by slavysis