melexis/mlx90640-library #16: Limited ranges on checks

Kiogora, 2018-10-02:
I received  these compiler warnings while using the esp-idf toolchain to perform the tests.

```
In function 'ExtractGainParameters':
mlx90640_api.c:613:15: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if(gainEE > 32767)
               ^

mlx90640_api.c:785:19: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (offsetRef > 32767)
                   ^

mlx90640_api.c: In function 'ExtractKtaPixelParameters':
mlx90640_api.c:855:17: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (KtaRoCo > 127)

```
Are the limited ranges of the datatypes used intended?

Kiogora, 2018-10-02:
Just checked the code cleanly and I notice these are sanity checks in case of overflow issues on the data passed. I shall now close this issue.

closed 2018-10-02 by Kiogora

sagar448, 2019-11-01:
@Kiogora Hey I know this is very long time ago but did you manage to run the MLX90640 on the esp32 esp-idf? If so do you have the code for it? It would really help!

Kiogora, 2019-11-01:
Check https://github.com/Kaiote-opensource/OpenHAP/tree/master/firmware/components/drivers_thermal_imagers