melexis/mlx90640-library #21: Repeated memory reads

jrddupont, 2019-01-07:
Hey, 
I am looking through the api trying to understand it and looking for places I can speed it up and whatnot. I noticed that [in this section the api ](https://github.com/melexis/mlx90640-library/blob/master/functions/MLX90640_API.cpp#L65) reads from the camera 4x. I reduced it to 1 time on my end and it still functions normally. Is there a reason for these extra reads? 
Thanks

slavysis, 2019-01-08:
Well, it does not really read 4 times. The driver is trying to get a valid data frame (no new frame data is available while the current one is being read). As this might create an endless loop, the driver allows that for just 5 loops. If for 5 loops a valid frame data cannot be obtained, an error is being thrown. If you have properly set the device and the I2C reading speed, one pass through that loop is enough as it anyways normally loops just once. I am working on a fast calculation solution, but I am not really sure when it would be ready.

jrddupont, 2019-01-10:
Ah, I see. Thank you for the information

closed 2019-01-10 by jrddupont