melexis/mlx90640-library #22: Is there a way to get the temperature reading from the camera?

Daz9, 2019-01-26:
I am trying to develop a python application that can take the temperature data from the sensor like a thermometer. I do not need the thermal image I just need the temperature reading.  

I used the fork of this library "pimoroni/mlx90640-library" but the hotspot.cpp code does not get compiled when running make and the python code examples do not work due to an error with the wrapper code I think.  ImportError: dynamic module does not define init function (init_MLX90640)

Any help with the issues or advice on how to do what I'm asking would be greatly appreciated

slavysis, 2019-01-28:
What do you mean by "I just need the temperature reading."? Is it that you need only Ta and not the temperatures of the pixels?

Daz9, 2019-01-28:
Hey thank you for replying,

Yeah, I'm only needing to perform Ta rather than generate pixels with the temperature values. An IR thermometer may have been better suited for the task but I only have the mlx90640 available to me.

If there is a way for me to extract the temperature at a certain point in the sensor like this image it would be ideal:
 
[image]

slavysis, 2019-01-29:
It looks like you still need to measure the object temperature from a distance, but you do not need the full frame, just some pixels, is that right? If that is right, normally you could do that with the MLX90640 device, but the driver is processing the whole frame only as this is the most efficient.

Regarding python, did you look at the issue just before the one you posted:  [https://github.com/melexis/mlx90640-library/issues/20](url)

closed 2020-11-03 by slavysis