melexis/mlx90640-library #29: Changing I2C Address

nlsa, 2019-03-09:
I tried to change the I2C address of my MLX90640 by the following steps, using methods from the mlx90640-library:
Addressing the device at 0x33,
  Write 0x0000 to register 0x240F
  Sleep 10 ms or longer
  Write 0xBE32 to register 0x240F
  Sleep
  power-off
Upon power-on, the device only responds with ACK to I2C address 0x00, returning either 0x008 or 0x009.  NACK from all other addresses 0x01 through 0x7F.  Repeating the above steps to 0x00 doesn't change anything.
Question 1: Have I ruined the device?
Question 2: Could someone please provide a working example that changes the I2C address so I don't repeat my mistake on another device?
Thank you -

slavysis, 2019-03-11:
When you say 'sleep' do you mean just waiting for 10ms without doing anything?
Are you powering down the MLX90640 device after you write 0x0000 and before you write the new slave address?
Do you read the EERPOM address after writing to it in order to verify that the correct value is written?

Best regards

nlsa, 2019-03-11:
Thank you for the quick reply, Slavysis!
I stepped through the code in the debugger.  The "sleep" period was whatever time it took for the step with my finger.
No, I did not power-down after setting it to 0x000... was I supposed to? (I thought that the power-down was only after writing the new valid address).
Yes, I did read after writing and it was not what I wrote, first clue that something was wrong.
Best,
Michael

slavysis, 2019-03-12:
Hi,

You should NOT power off the device when the slave address in the EEPROM is 0x00. So what you did seems right. The only issues is that for some reason you were not able to write the new slave address.
So when you powered off the device the slave address was 0x00 and that put the device in a special mode. Could you please contact Melexis sales and post a question explaining the issue. 

Best regards

nlsa, 2019-03-15:
I contacted Melexis with my question, pointing back to this thread.  They told me to contact Digi-Key (from whom I purchased the device) about the problem and ask _them_ to contact Melexis on my behalf.  Unimpressive customer service, I say.

nlsa, 2019-04-02:
Melexis responded directly with a generous offer to replace the device.  My faith in their customer service has been redeemed.  Thank you!

ackoc23, 2019-04-03:
Where you able to change the I2C address ?
How many MLX90640 can you connect together ?

nlsa, 2019-04-03:
Hi,
I haven’t tried again.  Awaiting a couple of new devices.  You should be able to configure up to 127 different I2C addresses.
I’m a little apprehensive because I thought that I was doing it right when I bricked the first one.
Best,
Michael

This message is intended only for the addressee and may contain information that is company confidential or privileged. Any technical data in this message may be exported only in accordance with the U.S. International Traffic in Arms Regulations (22 CFR Parts 120-130) or the Export Administration Regulations (15 CFR Parts 730-774). Unauthorized use is strictly prohibited and may be unlawful. If you are not the intended recipient, or the person responsible for delivering to the intended recipient, you should not read, copy, disclose or otherwise use this message. If you have received this email in error, please delete it, and advise the sender immediately.

nlsa, 2019-04-04:
> Do you know if there's any arduino example sketch on how to change the I2C address ?

Hi,
Google found a couple of discussions but not complete source code.
https://forum.arduino.cc/index.php?topic=570057.0
https://discourse.processing.org/t/how-to-change-i2c-address-of-mlx90640-32x24-thermal-camera-sensor/6238
I am not using Arduino so I can’t help, sorry
MRO

nlsa, 2019-04-11:
I succeeded in changing the address using the steps listed above.  The C# .NetMF code is below.  I stepped through it in the Visual Studio debugger so that I could confirm each step.
 public ChangeAddress()
        {
            using (OutputPort i2cPort = new OutputPort(Pins.GPIO_PIN_SDA, true)) // Toggles SDA.  This is necessary on Netduino Plus 2
            {                                                                    // to wake up I2C.  I don't know why.
                i2cPort.Write(false);
                Thread.Sleep(100);
            }
            Netduino.Foundation.Communications.I2CBus _I2C;
            _I2C = new Netduino.Foundation.Communications.I2CBus(0x33, 400);                                  // configure I2C 
            UInt16 Result = _I2C.ReadUShort(0x240F, Netduino.Foundation.Communications.ByteOrder.BigEndian);  // should be 0xBE33 (original address is 0x33)
            _I2C.WriteUShort(0x240F, 0, Netduino.Foundation.Communications.ByteOrder.BigEndian);              // erase register
            Thread.Sleep(10);
            Result = _I2C.ReadUShort(0x240F, Netduino.Foundation.Communications.ByteOrder.BigEndian);         // should be zero
            _I2C.WriteUShort(0x240F, 0xBE32, Netduino.Foundation.Communications.ByteOrder.BigEndian);         // write new address, 0x32
            Thread.Sleep(10);
            Result = _I2C.ReadUShort(0x240F, Netduino.Foundation.Communications.ByteOrder.BigEndian);         // should be 0xBE32
        }

AKremor, 2019-06-25:
> Hi,
>
> You should NOT power off the device when the slave address in the EEPROM is 0x00. So what you did seems right. The only issues is that for some reason you were not able to write the new slave [...]

Hi @slavysis,

What is the sensor doing in the 'special mode' that it enters when power cycled with address 0x00 set? Is there a way the user can return the sensor to normal operations if it enters the special mode, or is the sensor permanently unusable?

Thanks,
Anthony

slavysis, 2019-06-26:
Hi,
Unfortunately, there is no way the user can bring the sensor back to normal operation.

Best regards

p4twu, 2019-10-17:
Hi,

I'm also trying to change the i2c address of my MLX90640 in order to connect more than one device.

If I run my little program [setaddr.cpp](https://gist.github.com/p4twu/c729b884be2305d78e9de39c2480389a) the address gets changed, but it reverts back to the default address 0x33 after disconnecting and reconnecting the device to power.

Please note, that I changed the address at two locations (EEPROM 0x240F and register 0x8010). 

Can anyone help?

Thank you!
Patrick

closed 2020-11-03 by slavysis

karelv, 2025-08-04:
See raspberry pi specific forums for support on how to use that piece of hardware.
MLX90640 indeed answers to 0x33 by default.
It requires a 100nF and a 10uF decoupling capacitor near the sensor.
Other than that it pretty straight forward, connect SDA of the sensor with SDA of the master, same for SCL.
Make sure there is a pullup resistor on SDA & SCL, and the VDD is 3.3V.