melexis/mlx90640-library #31: Endianness reversal in I2C read

Kiogora, 2019-03-27:
The following code within the I2C driver reverses the endianness. I believe it is good to make note of this to users of this library to adapt accordingly.
```
    for(cnt=0; cnt < nMemAddressRead; cnt++)
    {
        i = cnt << 1;
        *p++ = (uint16_t)data_rd[i]*256 + (uint16_t)data_rd[i+1];
    }
```
This can  be seen in the first I2C buffer of the EEPROM dump that shows data as transferred over the wire and the reconstructed buffer returned from the I2C read function after the above code segment is applied.

```
D (2482) I2C UINT8 BUFFER: 00 a4 69 9f 00 00 20 61 00 04 03 20 03 e0 16 26 
D (2482) I2C UINT8 BUFFER: b2 25 61 86 04 99 00 00 ff ff 00 00 00 00 be 33 
D (2492) I2C UINT8 BUFFER: 43 20 ff b1 01 01 01 01 01 01 f1 f1 e0 f1 c0 e0 
D (2492) I2C UINT8 BUFFER: 00 01 01 02 01 02 f1 02 f1 f2 f1 f2 e0 e2 c0 d1 
D (2502) I2C UINT8 BUFFER: 88 95 37 70 dd a9 ff ff 33 22 33 33 11 22 dd ff 
D (2512) I2C UINT8 BUFFER: dd bb 00 ff 22 11 32 33 22 33 11 22 ff 01 bc ee 
D (2512) I2C UINT8 BUFFER: 17 72 2f ac 25 57 a0 7d 76 56 f1 8c 41 3d 3c 3b 
D (2522) I2C UINT8 BUFFER: 24 52 f8 7f ff ff 06 2a ec 00 97 97 97 97 2a fb 
D (2532) I2C UINT8 BUFFER: 08 40 05 30 23 e0 f8 5e fb fe 00 90 10 30 ff ff 
D (2542) I2C UINT8 BUFFER: 00 60 ff ff 10 2e f3 fe f8 0e ff e0 11 0e f7 ee 
D (2542) I2C UINT8 BUFFER: 04 00 ff ff 0c be ec 8e f5 00 f8 60 08 60 e4 1e 
D (2552) I2C UINT8 BUFFER: 03 70 f8 60 0c 40 eb 70 f4 1e ff 90 08 12 e4 7e 
D (2562) I2C UINT8 BUFFER: 0c 32 05 1e 07 be 14 30 03 c2 00 8e f0 2e 10 10 
D (2562) I2C UINT8 BUFFER: ff ff ff ff f4 0e 0f d0 03 c2 03 ae f8 be 17 a0 
D (2572) I2C UINT8 BUFFER: 0f d2 0f de f4 9e 0c 70 00 d2 00 4e f4 5e 04 00 
D (2582) I2C UINT8 BUFFER: 0f 52 04 50 fc 1e 0f 42 03 e2 0b 7e fb f0 ff ff 
D (2582) I2C UINT8 BUFFER: ff ff 0c 20 ff ff ff b0 ff 5e 04 20 ff ff f7 ce 
D (2592) I2C UINT8 BUFFER: 00 2e 04 40 17 a0 f3 be ff 8e ff 90 10 70 fb 2e 
D (2602) I2C UINT8 BUFFER: 07 a0 07 60 13 e0 f0 0e fc 3e f8 30 ff ff e7 de 
D (2612) I2C UINT8 BUFFER: ff ff 00 02 13 d0 e7 a0 f7 b0 ff 20 13 52 e7 fe 
D (2612) I2C UINT8 BUFFER: 08 22 0c 70 07 de 14 12 07 a2 08 80 f4 1e 10 40 
D (2622) I2C UINT8 BUFFER: 04 70 04 80 fb de 0f f0 03 c2 ff ff f8 ae ff ff 
D (2632) I2C UINT8 BUFFER: 0f e2 0f a0 f8 3e 10 50 04 72 00 70 f8 0e 04 30 
D (2632) I2C UINT8 BUFFER: 0b 92 08 60 00 20 07 f0 00 02 ff ff ff a0 04 40 
D (2642) I2C UINT8 BUFFER: 13 d0 0c 50 27 40 ff ff ff ff 0f f0 1f 40 ff ff 
D (2652) I2C UINT8 BUFFER: 08 70 04 70 17 c0 f7 ce 07 ae 07 40 14 80 03 40 
D (2662) I2C UINT8 BUFFER: 13 a0 0f 90 1f 90 fb e0 0b d0 00 40 10 30 f3 80 
D (2662) I2C UINT8 BUFFER: 03 c0 04 10 17 e0 ff ff ff b0 03 60 13 70 ff ff 
D (2672) I2C UINT8 BUFFER: 10 12 08 a0 07 8e 17 d0 0b e2 0c 50 03 8e 0c 90 
D (2682) I2C UINT8 BUFFER: 08 b2 04 be fb fe 0f e0 07 f2 ff 90 f4 de 1b 80 
D (2682) I2C UINT8 BUFFER: 17 d2 0f ee ff ff 14 10 0c 04 00 70 f4 5e ff ff 
D (2692) I2C UINT8 BUFFER: ff ff 08 50 00 10 0b f2 07 f2 0b b0 ff b0 04 70 
D (2702) I2C UINT8 BUFFER: ff ff 0c 60 27 60 03 b0 03 90 0c 40 1b c0 f8 2e 
D (2702) I2C UINT8 BUFFER: 00 60 ff ff 1b b0 f7 ce 07 40 ff d0 18 10 ff ff 
D (2712) I2C UINT8 BUFFER: 0f c2 0f 90 14 20 fb de 07 e2 00 02 0c 20 f3 6e 
D (2722) I2C UINT8 BUFFER: 0a c2 08 00 23 82 f3 80 03 80 07 60 1b 12 f0 00 
D (2732) I2C UINT8 BUFFER: ff ff 00 b0 03 9e 13 f2 ff d2 04 80 f7 ee ff ff 
D (2732) I2C UINT8 BUFFER: fc 80 00 90 f7 de 07 f0 03 72 f8 0e f8 4e 13 a2 
D (2742) I2C UINT8 BUFFER: 0f e4 0b d0 f0 4e ff ff 04 02 00 40 f0 50 ff ff 
D (2752) I2C UINT8 BUFFER: 12 f4 08 40 07 c0 0f b2 0b b2 ff ff ff 50 ff ff 
D (2752) I2C UINT8 BUFFER: 07 f0 0c 92 23 c0 fc 20 03 c0 10 30 14 00 f8 60 
D (2762) I2C UINT8 BUFFER: 00 90 08 50 1b 80 f7 a0 07 70 07 82 ff ff 02 de 
D (2772) I2C UINT8 BUFFER: 0b a0 0b f0 10 12 f4 00 07 f2 00 10 0c 60 ff ff 
D (2772) I2C UINT8 BUFFER: 0b 22 0c 72 18 12 ef d0 03 90 0b 62 13 d2 f3 f0 
D (2782) I2C UINT8 BUFFER: fc 22 f8 d0 f7 fe 08 42 fb e2 ff ff ec 3e 04 92 
D (2792) I2C UINT8 BUFFER: f8 b2 fc 60 f3 ae 07 c2 ff 82 fb b0 ff ff ff ff 
D (2802) I2C UINT8 BUFFER: 07 d2 00 20 f0 3e 08 32 00 34 fc 50 e8 8e fc 60 
D (2802) I2C UINT8 BUFFER: 0b 34 fc b0 ff ff 07 e0 ff ff 07 b0 f0 00 08 20 
D (2812) I2C UINT8 BUFFER: 0f e0 0c 90 27 70 fc 30 13 52 0c 60 17 b0 ff ff 
D (2822) I2C UINT8 BUFFER: 00 70 0b c2 17 50 f7 80 0b 50 ff a0 17 f2 fb b0 
D (2822) I2C UINT8 BUFFER: 0f a2 0b f0 17 d0 ff ff 00 32 07 d0 0c 02 ef e0 
D (2832) I2C UINT8 BUFFER: 03 42 04 82 17 c0 e8 20 f4 20 03 d0 ff ff ec 00 
D (2842) I2C UINT8 BUFFER: 00 12 f4 b0 f7 90 00 60 03 84 f4 80 eb de 04 02 
D (2842) I2C UINT8 BUFFER: f4 82 ff ff ff ff fb a0 ff 62 f3 c0 ec 0e 03 d0 
D (2852) I2C UINT8 BUFFER: 03 b4 fc 00 ef e0 00 30 f8 54 fb f0 ff ff ff ff 
D (2862) I2C UINT8 BUFFER: 03 64 f8 b0 f7 f0 f8 40 f4 32 fc 00 f3 80 fc 10 
D (2872) I2C UINT8 BUFFER: f8 40 ff ff 14 10 f4 90 ff b0 0c 52 0f e0 f4 20 
D (2872) I2C UINT8 BUFFER: f8 60 00 82 13 70 f3 50 ff 72 ff c0 ff ff fb 40 
D (2882) I2C UINT8 BUFFER: fc 02 07 f0 0c 00 f4 2e f8 52 00 40 04 10 eb c0 
[... 74 lines cut ...]
D (3502) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a30   12 10 a0 08 8e 07 d0 17  e2 0b 50 0c 8e 03 90 0c  |..........P.....|
D (3522) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a40   b2 08 be 04 fe fb e0 0f  f2 07 90 ff de f4 80 1b  |................|
D (3532) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a50   d2 17 ee 0f ff ff 10 14  04 0c 70 00 5e f4 ff ff  |..........p.^...|
D (3542) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a60   ff ff 50 08 10 00 f2 0b  f2 07 b0 0b b0 ff 70 04  |..P...........p.|
D (3552) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a70   ff ff 60 0c 60 27 b0 03  90 03 40 0c c0 1b 2e f8  |..`.`'....@.....|
D (3562) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a80   60 00 ff ff b0 1b ce f7  40 07 d0 ff 10 18 ff ff  |`.......@.......|
D (3572) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a90   c2 0f 90 0f 20 14 de fb  e2 07 02 00 20 0c 6e f3  |.... ....... .n.|
D (3582) RECONSTRUCTED UINT16 DATA:: 0x3ffb5aa0   c2 0a 00 08 82 23 80 f3  80 03 60 07 12 1b 00 f0  |.....#....`.....|
D (3592) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ab0   ff ff b0 00 9e 03 f2 13  d2 ff 80 04 ee f7 ff ff  |................|
D (3602) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ac0   80 fc 90 00 de f7 f0 07  72 03 0e f8 4e f8 a2 13  |........r...N...|
D (3612) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ad0   e4 0f d0 0b 4e f0 ff ff  02 04 40 00 50 f0 ff ff  |....N.....@.P...|
D (3622) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ae0   f4 12 40 08 c0 07 b2 0f  b2 0b ff ff 50 ff ff ff  |..@.........P...|
D (3632) RECONSTRUCTED UINT16 DATA:: 0x3ffb5af0   f0 07 92 0c c0 23 20 fc  c0 03 30 10 00 14 60 f8  |.....# ...0...`.|
D (3642) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b00   90 00 50 08 80 1b a0 f7  70 07 82 07 ff ff de 02  |..P.....p.......|
D (3662) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b10   a0 0b f0 0b 12 10 00 f4  f2 07 10 00 60 0c ff ff  |............`...|
D (3672) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b20   22 0b 72 0c 12 18 d0 ef  90 03 62 0b d2 13 f0 f3  |".r.......b.....|
D (3682) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b30   22 fc d0 f8 fe f7 42 08  e2 fb ff ff 3e ec 92 04  |".....B.....>...|
D (3692) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b40   b2 f8 60 fc ae f3 c2 07  82 ff b0 fb ff ff ff ff  |..`.............|
D (3702) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b50   d2 07 20 00 3e f0 32 08  34 00 50 fc 8e e8 60 fc  |.. .>.2.4.P...`.|
D (3712) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b60   34 0b b0 fc ff ff e0 07  ff ff b0 07 00 f0 20 08  |4............. .|
D (3722) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b70   e0 0f 90 0c 70 27 30 fc  52 13 60 0c b0 17 ff ff  |....p'0.R.`.....|
D (3732) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b80   70 00 c2 0b 50 17 80 f7  50 0b a0 ff f2 17 b0 fb  |p...P...P.......|
D (3742) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b90   a2 0f f0 0b d0 17 ff ff  32 00 d0 07 02 0c e0 ef  |........2.......|
D (3752) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ba0   42 03 82 04 c0 17 20 e8  20 f4 d0 03 ff ff 00 ec  |B..... . .......|
````

Kiogora, 2019-03-27:
> The following code within the I2C driver reverses the endianness of data. I believe it is good to make note of this to users of this library to adapt according to the endianness of the platform in use.
> 
> ```
>     for(cnt=0; cnt < nMemAddressRead; cnt++)
>     {
>         i = cnt << 1;
>         *p++ = (uint16_t)i2cData[i]*256 + (uint16_t)i2cData[i+1];
>     }
> ```
> 
> This can be seen in the first I2C buffer of the EEPROM dump that shows data as transferred over the wire and the reconstructed buffer returned from the I2C read function after the above code segment is applied.
> 
> ```
> D (2482) I2C UINT8 BUFFER: 00 a4 69 9f 00 00 20 61 00 04 03 20 03 e0 16 26 
> D (2482) I2C UINT8 BUFFER: b2 25 61 86 04 99 00 00 ff ff 00 00 00 00 be 33 
> D (2492) I2C UINT8 BUFFER: 43 20 ff b1 01 01 01 01 01 01 f1 f1 e0 f1 c0 e0 
> D (2492) I2C UINT8 BUFFER: 00 01 01 02 01 02 f1 02 f1 f2 f1 f2 e0 e2 c0 d1 
> D (2502) I2C UINT8 BUFFER: 88 95 37 70 dd a9 ff ff 33 22 33 33 11 22 dd ff 
> D (2512) I2C UINT8 BUFFER: dd bb 00 ff 22 11 32 33 22 33 11 22 ff 01 bc ee 
> D (2512) I2C UINT8 BUFFER: 17 72 2f ac 25 57 a0 7d 76 56 f1 8c 41 3d 3c 3b 
> D (2522) I2C UINT8 BUFFER: 24 52 f8 7f ff ff 06 2a ec 00 97 97 97 97 2a fb 
> D (2532) I2C UINT8 BUFFER: 08 40 05 30 23 e0 f8 5e fb fe 00 90 10 30 ff ff 
> D (2542) I2C UINT8 BUFFER: 00 60 ff ff 10 2e f3 fe f8 0e ff e0 11 0e f7 ee 
> D (2542) I2C UINT8 BUFFER: 04 00 ff ff 0c be ec 8e f5 00 f8 60 08 60 e4 1e 
> D (2552) I2C UINT8 BUFFER: 03 70 f8 60 0c 40 eb 70 f4 1e ff 90 08 12 e4 7e 
> D (2562) I2C UINT8 BUFFER: 0c 32 05 1e 07 be 14 30 03 c2 00 8e f0 2e 10 10 
> D (2562) I2C UINT8 BUFFER: ff ff ff ff f4 0e 0f d0 03 c2 03 ae f8 be 17 a0 
> D (2572) I2C UINT8 BUFFER: 0f d2 0f de f4 9e 0c 70 00 d2 00 4e f4 5e 04 00 
> D (2582) I2C UINT8 BUFFER: 0f 52 04 50 fc 1e 0f 42 03 e2 0b 7e fb f0 ff ff 
> D (2582) I2C UINT8 BUFFER: ff ff 0c 20 ff ff ff b0 ff 5e 04 20 ff ff f7 ce 
> D (2592) I2C UINT8 BUFFER: 00 2e 04 40 17 a0 f3 be ff 8e ff 90 10 70 fb 2e 
> D (2602) I2C UINT8 BUFFER: 07 a0 07 60 13 e0 f0 0e fc 3e f8 30 ff ff e7 de 
> D (2612) I2C UINT8 BUFFER: ff ff 00 02 13 d0 e7 a0 f7 b0 ff 20 13 52 e7 fe 
> D (2612) I2C UINT8 BUFFER: 08 22 0c 70 07 de 14 12 07 a2 08 80 f4 1e 10 40 
> D (2622) I2C UINT8 BUFFER: 04 70 04 80 fb de 0f f0 03 c2 ff ff f8 ae ff ff 
> D (2632) I2C UINT8 BUFFER: 0f e2 0f a0 f8 3e 10 50 04 72 00 70 f8 0e 04 30 
> D (2632) I2C UINT8 BUFFER: 0b 92 08 60 00 20 07 f0 00 02 ff ff ff a0 04 40 
> D (2642) I2C UINT8 BUFFER: 13 d0 0c 50 27 40 ff ff ff ff 0f f0 1f 40 ff ff 
> D (2652) I2C UINT8 BUFFER: 08 70 04 70 17 c0 f7 ce 07 ae 07 40 14 80 03 40 
> D (2662) I2C UINT8 BUFFER: 13 a0 0f 90 1f 90 fb e0 0b d0 00 40 10 30 f3 80 
> D (2662) I2C UINT8 BUFFER: 03 c0 04 10 17 e0 ff ff ff b0 03 60 13 70 ff ff 
> D (2672) I2C UINT8 BUFFER: 10 12 08 a0 07 8e 17 d0 0b e2 0c 50 03 8e 0c 90 
> D (2682) I2C UINT8 BUFFER: 08 b2 04 be fb fe 0f e0 07 f2 ff 90 f4 de 1b 80 
> D (2682) I2C UINT8 BUFFER: 17 d2 0f ee ff ff 14 10 0c 04 00 70 f4 5e ff ff 
> D (2692) I2C UINT8 BUFFER: ff ff 08 50 00 10 0b f2 07 f2 0b b0 ff b0 04 70 
> D (2702) I2C UINT8 BUFFER: ff ff 0c 60 27 60 03 b0 03 90 0c 40 1b c0 f8 2e 
> D (2702) I2C UINT8 BUFFER: 00 60 ff ff 1b b0 f7 ce 07 40 ff d0 18 10 ff ff 
> D (2712) I2C UINT8 BUFFER: 0f c2 0f 90 14 20 fb de 07 e2 00 02 0c 20 f3 6e 
> D (2722) I2C UINT8 BUFFER: 0a c2 08 00 23 82 f3 80 03 80 07 60 1b 12 f0 00 
> D (2732) I2C UINT8 BUFFER: ff ff 00 b0 03 9e 13 f2 ff d2 04 80 f7 ee ff ff 
> D (2732) I2C UINT8 BUFFER: fc 80 00 90 f7 de 07 f0 03 72 f8 0e f8 4e 13 a2 
> D (2742) I2C UINT8 BUFFER: 0f e4 0b d0 f0 4e ff ff 04 02 00 40 f0 50 ff ff 
> D (2752) I2C UINT8 BUFFER: 12 f4 08 40 07 c0 0f b2 0b b2 ff ff ff 50 ff ff 
> D (2752) I2C UINT8 BUFFER: 07 f0 0c 92 23 c0 fc 20 03 c0 10 30 14 00 f8 60 
> D (2762) I2C UINT8 BUFFER: 00 90 08 50 1b 80 f7 a0 07 70 07 82 ff ff 02 de 
> D (2772) I2C UINT8 BUFFER: 0b a0 0b f0 10 12 f4 00 07 f2 00 10 0c 60 ff ff 
> D (2772) I2C UINT8 BUFFER: 0b 22 0c 72 18 12 ef d0 03 90 0b 62 13 d2 f3 f0 
> D (2782) I2C UINT8 BUFFER: fc 22 f8 d0 f7 fe 08 42 fb e2 ff ff ec 3e 04 92 
> D (2792) I2C UINT8 BUFFER: f8 b2 fc 60 f3 ae 07 c2 ff 82 fb b0 ff ff ff ff 
> D (2802) I2C UINT8 BUFFER: 07 d2 00 20 f0 3e 08 32 00 34 fc 50 e8 8e fc 60 
> D (2802) I2C UINT8 BUFFER: 0b 34 fc b0 ff ff 07 e0 ff ff 07 b0 f0 00 08 20 
> D (2812) I2C UINT8 BUFFER: 0f e0 0c 90 27 70 fc 30 13 52 0c 60 17 b0 ff ff 
> D (2822) I2C UINT8 BUFFER: 00 70 0b c2 17 50 f7 80 0b 50 ff a0 17 f2 fb b0 
> D (2822) I2C UINT8 BUFFER: 0f a2 0b f0 17 d0 ff ff 00 32 07 d0 0c 02 ef e0 
> D (2832) I2C UINT8 BUFFER: 03 42 04 82 17 c0 e8 20 f4 20 03 d0 ff ff ec 00 
> D (2842) I2C UINT8 BUFFER: 00 12 f4 b0 f7 90 00 60 03 84 f4 80 eb de 04 02 
> D (2842) I2C UINT8 BUFFER: f4 82 ff ff ff ff fb a0 ff 62 f3 c0 ec 0e 03 d0 
> D (2852) I2C UINT8 BUFFER: 03 b4 fc 00 ef e0 00 30 f8 54 fb f0 ff ff ff ff 
> D (2862) I2C UINT8 BUFFER: 03 64 f8 b0 f7 f0 f8 40 f4 32 fc 00 f3 80 fc 10 
> D (2872) I2C UINT8 BUFFER: f8 40 ff ff 14 10 f4 90 ff b0 0c 52 0f e0 f4 20 
[... 76 lines cut ...]
> D (3502) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a30   12 10 a0 08 8e 07 d0 17  e2 0b 50 0c 8e 03 90 0c  |..........P.....|
> D (3522) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a40   b2 08 be 04 fe fb e0 0f  f2 07 90 ff de f4 80 1b  |................|
> D (3532) RECONSTRUCTED UINT16 DATA:: 0x3ffb5a50   d2 17 ee 0f ff ff 10 14  04 0c 70 00 5e f4 ff ff  |..........p.^...|
[... cut ...]
> D (3712) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b60   34 0b b0 fc ff ff e0 07  ff ff b0 07 00 f0 20 08  |4............. .|
> D (3722) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b70   e0 0f 90 0c 70 27 30 fc  52 13 60 0c b0 17 ff ff  |....p'0.R.`.....|
> D (3732) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b80   70 00 c2 0b 50 17 80 f7  50 0b a0 ff f2 17 b0 fb  |p...P...P.......|
> D (3742) RECONSTRUCTED UINT16 DATA:: 0x3ffb5b90   a2 0f f0 0b d0 17 ff ff  32 00 d0 07 02 0c e0 ef  |........2.......|
> D (3752) RECONSTRUCTED UINT16 DATA:: 0x3ffb5ba0   42 03 82 04 c0 17 20 e8  20 f4 d0 03 ff ff 00 ec  |B..... . .......|
> ```

slavysis, 2019-03-29:
Hello,

Actually, the I2C protocol reverses the endianness of the data as it transmits LSbyte first and MSbit first and the function restores the data in the original endianness. You are right that I should explicitly note it so that the restored EEPROM data is properly used within the other functions. I will add that note to the pdf document.

Best regards

Kiogora, 2019-03-30:
Hi @slavysis , thank you for the insight.

The issue was that my I2C HAL code automatically reversed the endianness to LSByte first before committing the data to the I2C peripheral, therefore using the HAL code and this code below together did a double reversal on the data's endianness.

```
    for(cnt=0; cnt < nMemAddressRead; cnt++)
    {
        i = cnt << 1;
        *p++ = (uint16_t)i2cData[i]*256 + (uint16_t)i2cData[i+1];
    }
```
I shall now close this issue personally and thanks for adding this note on the next PDF revision.

closed 2019-03-30 by Kiogora