melexis/mlx90640-library #32: Documentation of EEPROM address 0x240A

Kiogora, 2019-03-27:
CheckEEPROMValid() does the following check:

`deviceSelect = eeData[10] & 0x0040;` 

which fails, but the data in the EEPROM is actually correct when parameters are etracted. I have this verified by checking the extracted parameters. As noted on the endianness issue in #31 , My data both in both big and little endian is 0x9904 and 0x0499, but this check will always fail.

Without documentation of this EEPROM address in the datasheet, it is difficult to know the cause of the failure. Maybe it is meant to be:

`deviceSelect = eeData[10] & 0x0004`?

slavysis, 2019-03-29:
Hello,

Actually with both endiannesses it should pass. The complete code is:

int CheckEEPROMValid(uint16_t *eeData)  
 {
     int deviceSelect;
     deviceSelect = eeData[10] & 0x0040;
     if(deviceSelect == 0)
     {
         return 0;
     }
     
     return -7;    
 } 

When the EEPROM data is valid the function returns 0.

learning88, 2019-03-30:
me too, my eeData[10] =0x04CD

Kiogora, 2019-03-30:
@slavysis Thank you for the reply. I did look into the code suficiently. I notice the  eeData[10] value in the MLX90640 example data.xlsx that you committed has eeData[10] =0x04CD, my eeData[10]=0x0499, @learning88 eeData[10] =0x04CD .

These cases seem to have a pattern, any insight is appreciated

slavysis, 2019-04-08:
Hi,
The encoding at this address is going to change so that check will be modified or removed in the future versions of the driver. Apparently @learning88 already got new devices. You should simply remove the if statement in the MLX90640_ExtractParameters() function. The code should be something like this:

int MLX90640_ExtractParameters(uint16_t *eeData, paramsMLX90640 *mlx90640)
{
    int error = 0;
  
        ExtractVDDParameters(eeData, mlx90640);
        ExtractPTATParameters(eeData, mlx90640);
        ExtractGainParameters(eeData, mlx90640);
        ExtractTgcParameters(eeData, mlx90640);
        ExtractResolutionParameters(eeData, mlx90640);
        ExtractKsTaParameters(eeData, mlx90640);
        ExtractKsToParameters(eeData, mlx90640);
        ExtractCPParameters(eeData, mlx90640);
        ExtractAlphaParameters(eeData, mlx90640);
        ExtractOffsetParameters(eeData, mlx90640);
        ExtractKtaPixelParameters(eeData, mlx90640);
        ExtractKvPixelParameters(eeData, mlx90640);
        ExtractCILCParameters(eeData, mlx90640);
        error = ExtractDeviatingPixels(eeData, mlx90640);  
    
    return error;

}

@Kiogora I am pretty sure that eeData[10] = 0x0499 in the example data that I committed. So this is the same value as the one that you are getting and this value should return error = 0 and it should work just fine in your case. 

Best regards

Kiogora, 2019-04-14:
@slavysis Thanks for this. I do confirm that the data in your xlsx file is 0x499, sorry for the mixup. The function works well. I shall now close this issue.

closed 2019-04-14 by Kiogora