melexis/mlx90640-library #34: MLX90640 Read Data Not Fast Enough

ackoc23, 2019-05-22:
Hello,
I am Multiplexing a few MLX90640, trying to incorporate them into a few different projects,
I have noticed that the frame rate I get is very limited (less than 3Hz) compared to what I read online and the MLX specs, but it gets better if I cancel the following loop, and only run the content of it one time instead of twice  

for (byte x = 0 ; x < 2 ; x++) {
    uint16_t mlx90640Frame[834];
    int status = MLX90640_GetFrameData(MLX90640_address, mlx90640Frame);
    float Ta = MLX90640_GetTa(mlx90640Frame, &mlx90640);
    float tr = Ta - TA_SHIFT; //Reflected temperature based on the sensor ambient temperature    
    MLX90640_CalculateTo(mlx90640Frame, &mlx90640, emissivity, tr, mlx90640To);
 }    

If I don't run the loop twice I have noticed that I get errors in the images, 
Do I have to run this twice to get precise results?

I have tried this both with the Teensy3.2(96MHz) and ESP32(240MHz) and got similar results.
This also happens When I don't multiplex,  that tells me that it is related to the time it takes the API's in the loop to run

Thanks for your help

slavysis, 2019-05-22:
Hi,

I am not familiar with your setup, application and code, so my comment is based on assumptions.
Did you change the refresh rate of the MLX90640 device? If yes, did you make sure the correct refresh rate was selected in the register? MLX90640_GetFrameData() waits for new data to be available, so if the
device is set to slower refresh rate, you will get slower refresh rates.
It is normal to get slower refresh rate if you run the whole calculation twice.
It depends on your application how many times you need to run the loop.
What are the errors in the image that you get? Is it that you are missing data for half of the frame?
Is it that you are getting wrong temperature values?
Normally if your code/read are fast enough, each call of MLX90640_GetFrameData() will
result in new data for a sub-page. The sub-pages alternate automatically. The MLX90640_GetFrameData()
function returns the sub-page of the frame data that was read - this is helpful if you have some filtering
or other image processing. In that way if you skip a sub-page you can still do proper filtering and/or
processing without distorting the image.
Do you power off/on the MLX90640 device somewhere along the main loop?

Best regards

ackoc23, 2019-05-22:
Thanks for the fast replay,
I have tried to set the refresh rate of the MLX90640 device to 16Hz and 8Hz by:

`MLX90640_SetRefreshRate(MLX90640_address, 0x04); //Set rate to 8Hz
 MLX90640_SetRefreshRate(MLX90640_address, 0x05); //Set rate to 16Hz`

When you say "It depends on your application how many times you need to run the loop."
What is gained when the loop runs more than once?

I don't think that I am powering off/on the MLX90640 device somewhere along the main loop, How do you power off/on the MLX90640 during the loop?

This is my main loop:

`void loop()
{  
  image_msg.header.frame_id = frameid_one;  
  image_msg.header.stamp = nh.now();
    
  for (byte x = 0 ; x < 1 ; x++) {
    uint16_t mlx90640Frame[834];
    int status = MLX90640_GetFrameData(MLX90640_address, mlx90640Frame);
    float Ta = MLX90640_GetTa(mlx90640Frame, &mlx90640);
    float tr = Ta - TA_SHIFT; //Reflected temperature based on the sensor ambient temperature    
    MLX90640_CalculateTo(mlx90640Frame, &mlx90640, emissivity, tr, mlx90640To);
  }
  
  int Heat_Pixel=0;
  float max_v = 0;  
  float low_v = 1000; 
  const int data_length = image_msg.height*image_msg.step;
  uint8_t data[data_length];
  
  for (int i = 0 ; i < data_length ; i++) {       
    data[i]= mlx90640To[i]; 
    if ( data[i] > max_v ) {
      max_v = data[i];      
    }
    if ( data[i] < low_v ) {
      low_v = data[i];      
    }    
  }
  
  if ( (max_v - low_v) > 4 ) {
    for (int i = 0 ; i < data_length ; i++) {  
      if ( data[i] > (max_v - 3) ) {
        image_msg.data[i]=255;
        Heat_Pixel++;
      }   
      if ( data[i] < (max_v - 3) ) {
        image_msg.data[i]=0;
      }     
    }
  }        
  else {
    for (int i = 0 ; i < data_length ; i++) {  
      image_msg.data[i]=0;      
    }  
  }
  
  //publish message
  if (Heat_Pixel>=1) {
    thermal_image.publish(&image_msg); 
  }
  else {
    for (int i = 0 ; i < data_length ; i++) {  
      image_msg.data[i]=0;             
    }
    thermal_image.publish(&image_msg);   
  }
  nh.spinOnce();  
}`

Most of the time (0.5 sec) is taken by the for loop that takes the reading at the start of the main loop
It would be great if I could get a few Hz from the sensor,
Thanks for your help,

slavysis, 2019-05-23:
Hi,

How long does it take you to just read the frame data (without the calculation). What is your i2c speed
when getting the frame data? 
I am not aware why you are running that loop twice. The only reasonable
explanation I can think of is that your whole main loop is not fast enough and you cannot get the consecutive subpages. So that is why you are running that loop twice - in the first iteration you are getting one of the subpages and then probably you are in time to also get the other subpage in the second itteration. In this way after that loop you should probably get data for the whole frame of 768 pixels. I would expect that reading out the data takes you around 15-16 ms at 1MHz I2C clock (x2 as you need to do it twice for the 2 subpages. The calculations should take you around 30-40ms. So in total I would expect that for a full frame you should be able to get at least 10Hz. Of course this also depends on the speed of the other stuff you are doing.
I see that you are making some sort of a 'black-and-white' image where all pixels above certain limit are white and all pixels below that limit are black. So I guess that if you do not take consecutive frames you may end-up in a situation where the data for one subpage(newly acquired) is a bit different than the data for the other (previously acquired) and that 'black-and-white' image gets messed up (this is just an assumption). Thus you are having troubles when running the loop one time - you are simply working with half of the frame being outdated. 
I would recommend that you first make sure you are running at the max speed of the i2c in order to be able to read out the data as fast as possible. After that I believe you should check how much time each step is taking and try to optimize if possible. If your whole main loop runs faster than the refresh rate of the device, you should be able to run the for loop just once as you would be back in time to read the next subframe.

Best regards

ackoc23, 2019-05-24:
Hi,

I setup up my I2C speed and Refresh Rate as:

`Wire.begin(22,23,1000000); //Increase I2C clock speed to 1MHz - ESP32 , I2C Pin23 SCL, Pin22 SDA
  MLX90640_I2CFreqSet(1000); //Set I2C Freq, in kHz (1000=1MHz)
MLX90640_SetRefreshRate(MLX90640_address, 0x04);  //Set rate to 8Hz `

When I cheak the time to get the frame data as:
`double RTime=millis();`
`int status = MLX90640_GetFrameData(MLX90640_address, mlx90640Frame);`
` Serial.println(millis()-RTime);`

I get:

421.00
33.00
421.00
33.00
421.00
33.00
421.00
32.00
...
Meaning it takes 421ms and then 33ms and over again....

this is running on 240GHz so why is it so slow?

Thanks in advance,
Best regards

ackoc23, 2019-05-24:
One more pice of information is that the oveerall time of the main loop is:
447.00
61.00
448.00
62.00
447.00
62.00

So the rest of the code in the loop only takes 30ms, whereas MLX90640_GetFrameData takes over ~430ms

Thanks,

slavysis, 2019-05-27:
Did you measure the actual frequency of the i2c clock (using an oscilloscope)? It seems you are running at much lower frequency than 1MHz. The GetFrameData function is waiting for a new frame data to be available and then tries to read out the whole frame data before a next 'new' data is available. If this is not possible, an error is being thrown. Note that if the time-to-read the frame data is close to the refresh time of the sensor you may need a few tries before getting the valid data. This would result in an increased time. That is why it is important to make sure you are able to read all the pixels data as fast as possible.

Best regards

ackoc23, 2019-05-27:
What hardware would you suggest in order to read the data through I2C as fast as possible?
Can I use I2C faster than 1MHz? dos the sensor support this?
Is there another way to talk to the MLX90640 besides I2C?

Thanks,

slavysis, 2019-05-28:
Well, any hardware that supports 1000KHz i2c and repeated start condition generation. At 1Mhz you should be able to read the sub-frame data in about 16ms. Did you check what is the actual frequency of your i2c?

Best regards

closed 2020-11-03 by slavysis