melexis/mlx90640-library #37: EEPROM validity bit discrepancy in MLX90640ESF-BAB devices

sdbakker, 2019-06-04:
I have 3 BAA 120 degree and 3 BAB 60 degree devices. Using this library I have no problems using the BAA devices. However the BAB devices failed to extract the EEPROM parameters.

After closer investigation it seems that the function https://github.com/melexis/mlx90640-library/blob/0b76a54c480f565e7131fadaab7bb82a76f420c4/functions/MLX90640_API.cpp#L1277 
checks bit 6 of addres 0x240A for 0.  For all 3 BAB devices this bit is 1:
```
BAA 0x240A: 0x48D - 10010001101
BAB 0x240A: 0x4D9 - 10011011001
```
If I alter the EEPROM validity check to reflect this all else works like expected. 

I can not find any reference of this discrepancy between BAA and BAB devices in the datasheet, this library or errata. Is this a recent change, a fluke or some undocumented difference?

slavysis, 2019-06-05:
As there were some changes in the EEPROM encoding, the CheckEEPROMValid function will be removed from the driver (hopefully very soon). For more details, please refer to that post: [https://github.com/melexis/mlx90640-library/issues/32#issuecomment-480719293](url)

sdbakker, 2019-06-06:
@slavysis thanks for your reaction and apologies for the duplication! I somehow missed that issue.

closed 2019-06-06 by sdbakker