melexis/mlx90640-library #38: MLX90640_BadPixelsCorrection segfault with no bad pixels

Gadgetoid, 2019-06-11:
See https://github.com/pimoroni/mlx90640-library/issues/22 for context.

The long and short of it is that if `MLX90640_BadPixelsCorrection` gets a list of `{0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF, 0xFFFF}` it will fail since it doesn't check to see if the `brokenPixels` or `outlierPixels` arrays passed to it actually contain valid pixel indices.

I've fixed this rather ham-fistedly with a check against the first index for `0xFFFF` (see https://github.com/pimoroni/mlx90640-library/commit/93bb625bcd1a6269738459a85483d6057f2d6d7a#diff-26f614063f9bb2756155e3e26d51b772R120) but I believe this is something that `MLX90640_BadPixelsCorrection` should check.

slavysis, 2019-06-12:
Hi,

Actually the function does check for 0xFFFF:

while(pixels[pix] < 65535)
    {...} 

Perhaps it is possible that this check does not work properly for some compilers/platforms as it is comparing a 16bit unsigned value to 0xFFFF (which might be interpreted differently by some compilers).
Could you, please try to change this check in the MLX90640_BadPixelsCorrection to:

while(pixels[pix] != 0xFFFF)
    {...}

Best regards

Gadgetoid, 2019-06-12:
You're correct, changing `65535` to `0xFFFF` does the trick. I don't know why I didn't spot this!