melexis/mlx90640-library #42: No work in MLX90640_DumpEE

sjm1924, 2019-08-26:
Hi, I'm a developer to try use MLX90640 on STM32F767ZI Nucleo Board

I'm using STM32CubeIDE to use stm board and MLX90640.

With mlx90640-libarary of this git, I made a code for I2C Operation as below:

uint16_t user_i2c_read_Word(uint8_t id, uint16_t reg_addr)
{
	uint8_t data[2];
	uint8_t cmd[2];
	cmd[0] = (reg_addr & 0xFF00) >> 8;
	cmd[1] = reg_addr & 0x00FF;
//
	if(HAL_I2C_Master_Transmit(&hi2c2, (id << 1), cmd, 2, HAL_MAX_DELAY)!= HAL_OK) return -1;
	if(HAL_I2C_Master_Receive(&hi2c2, (id << 1) | 0x01, data, 2, HAL_MAX_DELAY)!= HAL_OK) return -1;

	return ((uint16_t) (data[0] << 8)) | ((uint16_t)((data[1]) & 0x00FF));

}

int MLX90640_I2CRead(uint8_t slaveAddr, uint16_t startAddress,uint16_t nMemAddressRead, uint16_t *data)
{
	uint16_t temp_address = startAddress;
		uint16_t temp_data;
		if(HAL_I2C_IsDeviceReady(&hi2c2, (slaveAddr << 1), 1, 2) != HAL_OK) return -1;

		for(int i=0; i < nMemAddressRead; i++){
			temp_data = user_i2c_read_Word(slaveAddr,temp_address);
			if(temp_data == -1) return -1;
			temp_address++;
			*(data + i) = temp_data;
		}
	  return 0;
}

int MLX90640_I2CWrite(uint8_t slaveAddr, uint16_t writeAddress, uint16_t data)
{
	uint8_t reg_m,reg_l,dat_m,dat_l;
		reg_m = (uint8_t) ((writeAddress & 0xFF00) >> 8);			//Address MSB
			reg_l = (uint8_t) (writeAddress & 0x00FF); 				//Address LSB
			dat_m = (uint8_t) ((data & 0xFF00) >> 8);	// Data MSB
			dat_l = (uint8_t) (data & 0x00FF);			//Data LSB
			uint8_t Byte4[4]={0};
			Byte4[0]=reg_m;
			Byte4[1]=reg_l;
			Byte4[2]=dat_m;
			Byte4[3]=dat_l;

	  static uint16_t dataCheck;

	  HAL_I2C_IsDeviceReady(&hi2c2, (slaveAddr << 1), 1, 2);
	  if(HAL_I2C_Master_Transmit(&hi2c2, (slaveAddr << 1), Byte4, 4, HAL_MAX_DELAY) != HAL_OK) return -1;

	  return 0;
}

But, when I use
MLX90640_Error=MLX90640_SetRefreshRate(MLX_I2C_ADDR, RefreshRate);
MLX90640_Error=MLX90640_SetChessMode(MLX_I2C_ADDR);

it works well with return value 0.
However, when I use
paramsMLX90640 mlx90640;
MLX90640_Error=MLX90640_DumpEE(MLX_I2C_ADDR, eeMLX90640);

It doesn't work any more. Just holding time....
Could you give me any advice to solve this problem?

I've already check: https://github.com/melexis/mlx90640-library/issues/12
At that issue, he use LL code for stm32f4, and LL structure of it with that of stm32f7 is differ.
Thus, I want to check first if the HAL code is alright or not.

Thank you for reading this issue.

slavysis, 2019-08-26:
Hi,

Did you check the communication with some oscilloscope or logical analyzer?

Best regards

imliubo, 2019-09-15:
Hi @sjm1924 :
    Have you sloved this problem?I meet the same problem with you.

sjm1924, 2019-09-17:
Hi, I Solve the problem by making LL code for MLX90640.

The code is below:

`uint16_t MLX90640_I2CReadWord(uint8_t slave_address, uint16_t start_address) {

	volatile uint8_t reg_m,reg_l,dat_m,dat_l;

	reg_m = (uint8_t) ((start_address & 0xFF00) >> 8);	//Address MSB
	reg_l = (uint8_t) (start_address & 0x00FF); 	    //Address LSB

	LL_I2C_SetSlaveAddr(I2C2, (slave_address<<1)); /* Prepare Address to send */
	LL_I2C_SetMasterAddressingMode(I2C2, LL_I2C_ADDRSLAVE_7BIT);
	LL_I2C_SetTransferRequest(I2C2, LL_I2C_REQUEST_WRITE); /* Request write */
	LL_I2C_DisableAutoEndMode(I2C2); /* Disable automatic STOP condition generation */
	LL_I2C_SetTransferSize(I2C2, 2); /* Set transfer size register */
	while(LL_I2C_IsActiveFlag_BUSY(I2C2)){} /* check I2C busy */
	LL_I2C_GenerateStartCondition(I2C2); /* generate I2C Start address and send address*/

	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)){
	}
	//Send start address MSB
	LL_I2C_TransmitData8(I2C2, reg_m);
	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)){
	}
	//Send start address LSB
	LL_I2C_TransmitData8(I2C2, reg_l);
	while (!LL_I2C_IsActiveFlag_TC(I2C2)){
	}

	LL_I2C_SetSlaveAddr(I2C2, (slave_address<<1)); /* Prepare Address to send */
	LL_I2C_SetMasterAddressingMode(I2C2, LL_I2C_ADDRSLAVE_7BIT);
	LL_I2C_SetTransferRequest(I2C2, LL_I2C_REQUEST_READ); /* Request read */
	LL_I2C_DisableAutoEndMode(I2C2); /* Enable autoend mode*/
	LL_I2C_SetTransferSize(I2C2, 2); /* Set transfer size register */
	/*while(LL_I2C_IsActiveFlag_BUSY(I2Cx)){} check I2C busy */
	LL_I2C_GenerateStartCondition(I2C2); /* generate I2C Start address and send address*/

	LL_I2C_AcknowledgeNextData(I2C2, LL_I2C_ACK);
	while (!LL_I2C_IsActiveFlag_RXNE(I2C2)){
	}
	dat_m = LL_I2C_ReceiveData8(I2C2);
	LL_I2C_AcknowledgeNextData(I2C2, LL_I2C_NACK);

	while (!LL_I2C_IsActiveFlag_RXNE(I2C2)){
	}
	dat_l = LL_I2C_ReceiveData8(I2C2);
	LL_I2C_GenerateStopCondition(I2C2);

	LL_I2C_ClearFlag_TXE(I2C2);
	LL_I2C_ClearFlag_OVR(I2C2);
	LL_I2C_ClearFlag_NACK(I2C2);
	LL_I2C_ClearFlag_STOP(I2C2);

	return ((uint16_t) (dat_m << 8)) | ((uint16_t)((dat_l) & 0x00FF));

}

int MLX90640_I2CRead(uint8_t slave_address, uint16_t start_address,
		uint16_t numb_words, uint16_t *data){

	uint16_t temp_address = start_address;
	uint16_t temp_data;
	for(int i=0; i < numb_words; i++){

		temp_data = MLX90640_I2CReadWord(slave_address,temp_address);
		temp_address++;
		HAL_Delay(0);
//		printf("i=%d ",i);
		*(data + i) = temp_data;

	}
	return 0;
}

void MLX90640_I2CFreqSet(int freq) {
	//i2c.frequency(1000*freq);
}

int MLX90640_I2CWrite(uint8_t slave_address, uint16_t start_address,
		uint16_t data) {

	uint8_t reg_m,reg_l,dat_m,dat_l;
	reg_m = (uint8_t) ((start_address & 0xFF00) >> 8);			//Address MSB
	reg_l = (uint8_t) (start_address & 0x00FF); 				//Address LSB
	dat_m = (uint8_t) ((data & 0xFF00) >> 8);	// Data MSB
	dat_l = (uint8_t) (data & 0x00FF);			//Data LSB

	LL_I2C_Disable(I2C2);
	LL_I2C_ClearFlag_STOP(I2C2);
	LL_I2C_Enable(I2C2);
	LL_I2C_HandleTransfer(I2C2, (slave_address<<1), LL_I2C_ADDRSLAVE_7BIT, 4, LL_I2C_MODE_AUTOEND,    LL_I2C_GENERATE_START_WRITE);
//	LL_I2C_SetTransferSize(I2C2,6);
//	LL_I2C_GenerateStartCondition(I2C2);

//	//Send device address
//	LL_I2C_TransmitData8(I2C2, slave_address);
	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)) {
	}

	//Send write address MSB
	LL_I2C_TransmitData8(I2C2, reg_m);
	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)) {
	}

	//Send write address LSB
	LL_I2C_TransmitData8(I2C2, reg_l);
	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)) {
	}

	//Send data MSB
	LL_I2C_TransmitData8(I2C2, dat_m);
	while (!LL_I2C_IsActiveFlag_TXIS(I2C2)) {
	}

	//Send data LSB
	LL_I2C_TransmitData8(I2C2, dat_l);
	while (!LL_I2C_IsActiveFlag_STOP(I2C2)) {
	}

//	LL_I2C_GenerateStopCondition(I2C2);

	return 0;
}`

And the result of this code is below:
[image]
[image]

The difference of two pictures is existing of a paper cup with hot water.

However, I met another problem.
I'm using stm32f767zi nucleo board.
When I set SystemClk as 216MHz, I can use I2C Clock under 600kHz.
If I set I2C Clock higher than 600kHz, the code is stuck on 

while (!LL_I2C_IsActiveFlag_TXIS(I2C2)){}

in MLX90640_I2CReadWord().

If I set SystemClk as 192MHz, I can use I2C Clock under 800kHz with same stats.

What should I do??

Furthermore, When I get just frame data as below:
`while (1)
  {
	  count_start=TIM6->CNT;
	  cnt++;

 	//////////////// IR Array //////////////
 	  int MLX90640_Error = 0;
 	 MLX90640_Error=MLX90640_GetFrameData(MLX_I2C_ADDR, dataSend.dataIR);

 	count_end=TIM6->CNT;
 	if (count_end<count_start){
 		count_end=count_end+32768;
 	}

 	duration=(count_end-count_start)/(float)SystemCoreClock*(htim6.Init.Prescaler+1);
 	printf("Count Start: %d, Count End: %d, Duration: %.2f s\r\n",count_start, count_end,duration);
}`

It takes 0.42s for one loop. But I have to use IR array with 10Hz.
What's the problem with this status?

sjm1924, 2019-09-17:
Hi @imliubo :
Thanks for comments.
I use below code to print the result of above pictures(a paper cup with hot water).
'''C
for(int row=0;row<32;row++){
 	 		for(int col=0;col<24;col++){
 	 			printf("%5d ",65535-dataSend.dataIR[32 * (23-col) + row]);
 	 		}
 	 		printf("\r\n");
 	 	}
''' 

I just need raw data with 10Hz.
If you use SWI2C, does it works faster than 10Hz?

imliubo, 2019-09-21:
Hi @sjm1924 ,
I update the code with HWI2C mode,it can be work with max 16Hz.You can try it.
[MLX90640-With-STM32](https://github.com/imliubo/MLX90640-With-STM32)