melexis/mlx90640-library #44: MLX90640 Eval Board

gutcZHAW, 2019-10-14:
Hi there

I want to read out different sensors with a raspberry pi 3 and store the data. One of this sensors is a MLX90640 which is mounted on an evaluation Board from melexis. To minimize the calculation load on the raspberry, i want the STM32 on the eval board to do all the necessary calculations.

Is there also a library (which runs on a Linux System) or a Documentation of the USB Interface of the Evaluation Board?

Thanks in advance for your support.

Best regards

adityac-view, 2019-10-21:
> Hi there
>
> I want to read out different sensors with a raspberry pi 3 and store the data. One of this sensors is a MLX90640 which is mounted on an evaluation Board from melexis. To minimize the [...]

Hi gutcZHAW, I've also been trying to find a solution for the same. If you've come around something useful, please do share. Thanks!

gutcZHAW, 2019-10-22:
Hi @adityac-view 

I contacted Melexis regarding this toppic. Unfortunately they don't provide a library to access the eval board which runs on a linux system. I only got the description of the psf which doesn't give any low level information of the usb interface. There isn't even a schematic of the eval board so it's pretty useless for my application.

I'm not sure yet if I should use this library and access the mlx90640 direct over i2c or if I should look for a better suited hardware.

closed 2020-11-03 by slavysis