melexis/mlx90640-library #45: I2C read problem

vthielen, 2019-10-16:
I am trying to read back the image data from the MLX90640 to an MSP430 TI MCU over I2C
The problem that I have is that the chip always replies the same data, even after resetting it (although I should be reading from RAM)
Even when I change the readback address, the data is always the same.
Therefore, I think the chip doesn't get the readback address, en responds with a page from its ROM.
Can somebody have a look at the traces in attachment?
Might the problem be that there is a stop condition after the requested address, instead of restart?
[image]
[image]

slavysis, 2019-10-31:
Indeed the problem is that instead of a 'Repeated start' condition the MCU is transmitting a 'Stop' condition. The MLX90640 terminates the communication after the stop condition and does not fetch the initial read address. If your library does not support 'repeated start', you could try using the software implementation of the I2C.

vthielen, 2019-11-13:
Hello Slaveyko,

Thanks for the reply!
I am using a restart now, and I am getting data that seems to respond to
the image now (values go up if I put my hand over it)
But the data I get back seems strange to me: MSByte is 0xFF everywhere (see
image below), and only LSB has data dat responds to the image.
Does this have something to do with the checkerboard pattern?
Or do I have to turn the gain of the chip down?

Kind regards,

Bart

[image: afbeelding.png]

slavysis, 2019-11-13:
The data is 2's complement, so having FF in the MS byte simply means that the data is negative. At higher object temperatures the signal will become positive.

Best regards

vthielen, 2020-01-03:
Hello Slaveyko,

I contacted you a while ago because I had problems reading out the MLX90640
chip.
The readout is now ok, but I have a problem with the temperature values
that I get: they are about 20 deg too low: I used it on my hand (37
degrees) in the data below, and get values of about 18 degrees.
Also, if I use the GetImage routine, I still see a residual checkerboard
pattern (see attachment) after combining both subframes, suggesting to me
that the two subframes are not equal.

Could this have anything to do with the settings in the control register?
I am using 0x191D to get subpage 1 and 0x190D to get subpage 0, after which
I write 0x0010 to the status register, but I have the impression that this
means the camera only takes one image (instead of continuous), and
therefore doesn't reach a good working temperature? (I read back 28 degrees)

At the bottom of this email I included the data from the temperature
calculation, generated by inserting two printf's before and after the line
that calculates the final temperature as shown below:

printf("pix%i To=%f ",pixelNumber,To);
To = sqrt(sqrt(irData / (alphaCompensated * alphaCorrR[range] * (1 +
params->ksTo[range] * (To - params->ct[range]))) + taTr)) - 273.15;
printf("IRdata=%f alphaCompensated=%f alphaCorrR=%f To=%f range=%i ksTo=%f
ct=%f\n",irData,alphaCompensated,alphaCorrR[range],To,range,params->ksTo[range],params->ct[range]);

[image: afbeelding.png]

irDataCP:-45.675446 -50.232594  gain:0.998309  alphaCorrR:1.008076 1.000000
0.939919 0.902271  taTr:8391447552.000000  alphascale:2048.000000
 vdd:3.554232  ta:27.945938  tr:-8.000000  emissivity:0.950000

pix0 To=11.706423 IRdata=-63.010338 alphaCompensated=0.000000
alphaCorrR=1.000000 To=11.706601 range=1 ksTo=-0.000200 ct=0.902270
pix2 To=13.226405 IRdata=-63.429398 alphaCompensated=0.000000
alphaCorrR=1.000000 To=13.226575 range=1 ksTo=-0.000200 ct=0.902270
pix4 To=15.732354 IRdata=-59.086304 alphaCompensated=0.000000
alphaCorrR=1.000000 To=15.732493 range=1 ksTo=-0.000200 ct=0.902270
pix6 To=16.579142 IRdata=-58.314812 alphaCompensated=0.000000
alphaCorrR=1.000000 To=16.579271 range=1 ksTo=-0.000200 ct=0.902270
pix8 To=18.003624 IRdata=-55.162258 alphaCompensated=0.000000
alphaCorrR=1.000000 To=18.003731 range=1 ksTo=-0.000200 ct=0.902270
pix10 To=18.554527 IRdata=-54.251087 alphaCompensated=0.000000
alphaCorrR=1.000000 To=18.554626 range=1 ksTo=-0.000200 ct=0.902270
pix12 To=19.054195 IRdata=-53.168243 alphaCompensated=0.000000
alphaCorrR=1.000000 To=19.054289 range=1 ksTo=-0.000200 ct=0.902270
pix14 To=19.532085 IRdata=-51.098534 alphaCompensated=0.000000
alphaCorrR=1.000000 To=19.532169 range=1 ksTo=-0.000200 ct=0.902270
pix16 To=19.250906 IRdata=-52.467487 alphaCompensated=0.000000
alphaCorrR=1.000000 To=19.250994 range=1 ksTo=-0.000200 ct=0.902270
pix18 To=18.880327 IRdata=-53.235611 alphaCompensated=0.000000
alphaCorrR=1.000000 To=18.880423 range=1 ksTo=-0.000200 ct=0.902270
pix20 To=18.800734 IRdata=-54.390774 alphaCompensated=0.000000
alphaCorrR=1.000000 To=18.800831 range=1 ksTo=-0.000200 ct=0.902270
pix22 To=18.423588 IRdata=-52.988235 alphaCompensated=0.000000
alphaCorrR=1.000000 To=18.423691 range=1 ksTo=-0.000200 ct=0.902270
pix24 To=17.791426 IRdata=-54.569183 alphaCompensated=0.000000
alphaCorrR=1.000000 To=17.791536 range=1 ksTo=-0.000200 ct=0.902270
pix26 To=16.138138 IRdata=-57.297649 alphaCompensated=0.000000
alphaCorrR=1.000000 To=16.138269 range=1 ksTo=-0.000200 ct=0.902270
pix28 To=15.686981 IRdata=-57.124283 alphaCompensated=0.000000
alphaCorrR=1.000000 To=15.687119 range=1 ksTo=-0.000200 ct=0.902270
pix30 To=13.199301 IRdata=-60.450211 alphaCompensated=0.000000
alphaCorrR=1.000000 To=13.199468 range=1 ksTo=-0.000200 ct=0.902270
pix33 To=12.220643 IRdata=-65.398674 alphaCompensated=0.000000
alphaCorrR=1.000000 To=12.220819 range=1 ksTo=-0.000200 ct=0.902270
pix35 To=14.382569 IRdata=-60.509827 alphaCompensated=0.000000
alphaCorrR=1.000000 To=14.382729 range=1 ksTo=-0.000200 ct=0.902270
pix37 To=15.709999 IRdata=-61.316479 alphaCompensated=0.000000
[... 706 lines cut ...]
alphaCorrR=0.902271 To=nan range=3 ksTo=-0.000200 ct=0.902270
pix745 To=7.527321 IRdata=-107.018982 alphaCompensated=0.000000
alphaCorrR=1.000000 To=7.527502 range=1 ksTo=-0.000200 ct=0.902270
pix747 To=9.333996 IRdata=-100.713875 alphaCompensated=0.000000
alphaCorrR=1.000000 To=9.334185 range=1 ksTo=-0.000200 ct=0.902270
pix749 To=10.046054 IRdata=-102.692696 alphaCompensated=0.000000
alphaCorrR=1.000000 To=10.046242 range=1 ksTo=-0.000200 ct=0.902270
pix751 To=10.611079 IRdata=-96.462494 alphaCompensated=0.000000
alphaCorrR=1.000000 To=10.611268 range=1 ksTo=-0.000200 ct=0.902270
pix753 To=10.227405 IRdata=-100.031609 alphaCompensated=0.000000
alphaCorrR=1.000000 To=10.227593 range=1 ksTo=-0.000200 ct=0.902270
pix755 To=10.240218 IRdata=-96.836060 alphaCompensated=0.000000
alphaCorrR=1.000000 To=10.240406 range=1 ksTo=-0.000200 ct=0.902270
pix757 To=10.479496 IRdata=-93.972252 alphaCompensated=0.000000
alphaCorrR=1.000000 To=10.479685 range=1 ksTo=-0.000200 ct=0.902270
pix759 To=9.828879 IRdata=-93.683510 alphaCompensated=0.000000
alphaCorrR=1.000000 To=9.829068 range=1 ksTo=-0.000200 ct=0.902270
pix761 To=8.431450 IRdata=-99.663010 alphaCompensated=0.000000
alphaCorrR=1.000000 To=8.431639 range=1 ksTo=-0.000200 ct=0.902270
pix763 To=7.620794 IRdata=-97.315567 alphaCompensated=0.000000
alphaCorrR=1.000000 To=7.620979 range=1 ksTo=-0.000200 ct=0.902270
pix765 To=5.690634 IRdata=-101.396118 alphaCompensated=0.000000
alphaCorrR=1.000000 To=5.690798 range=1 ksTo=-0.000200 ct=0.902270
pix767 To=4.710835 IRdata=-95.527527 alphaCompensated=0.000000
alphaCorrR=1.000000 To=4.710990 range=1 ksTo=-0.000200 ct=0.902270

slavysis, 2020-01-06:
Indeed, this setting is causing the behavior you observe. If you try using the default setting 0x1901, everything should be working just fine.

Best regards

closed 2020-11-03 by slavysis