melexis/mlx90640-library #46: Issue with MLX 90640 on mbed on STMF437

nusret103, 2019-11-06:
Hi,

we are working in one project with your library with dev board from ublox (u-blox c030) and we are trying to run the terminal camera library provided by you on git.
we can't make this library to work.

we are logging out som data and the data are:

irData[764] = 0, kv= 0.437500, kta= 0.007812, offset= ffffff9d, alpha= nan

 irData[765] = 3f80, kv= 0.375000, kta= 0.007080, offset= ffffff96, alpha= nan

 irData[766] = 0, kv= 0.437500, kta= 0.007324, offset= ffffff97, alpha= nan

 irData[767] = 3f80, kv= 0.375000, kta= 0.007568, offset= ffffff94, alpha= -340279649078051570000000000000000000000.000000

let us know if you have an idea what is the issue or didi you have any hello world app for this specific MCU

slavysis, 2019-11-07:
Can you please provide some data logs? As a start it would be nice if you can provide the EEPROM data and the extracted parameters values.

nusret103, 2019-11-07:
Hi,

I am providing the data as you requested,

BR,

Nusret

[parameters_values1.txt](https://github.com/melexis/mlx90640-library/files/3818955/parameters_values1.txt)
[parameters_values.txt](https://github.com/melexis/mlx90640-library/files/3818956/parameters_values.txt)
[EEPROM.txt](https://github.com/melexis/mlx90640-library/files/3818957/EEPROM.txt)

slavysis, 2019-11-07:
Hi,
I see only the EEPROM data and the frame data. Can you also provide the extracted parameters (like alpha, offset, kv, etc)?

Best regards

nusret103, 2019-11-07:
Hi

I am uploading the excel file with my data
The is one extra sheet where you have the thermal images according to your excel file,
I try multiple data the image is the same always 

Wit respect,

Nusret
[Results.xlsx](https://github.com/melexis/mlx90640-library/files/3820080/Results.xlsx)

slavysis, 2019-11-07:
Hi,

The file does not calculate anything - it simply holds the output based on a sample data. That is why your input does not change the output. You could read how it is supposed to be used in the sheet 'Input data'. I can extract manually the data from the EEPROM that you provided, but I want to verify that the ExtractParameters function is working properly. That is why it would be nice if you could provide the printout of the paramsMLX90640 structure. 
Note: after the reduced RAM footprint update some of the parameters in the Example data file will not be correct.

Best regards

nusret103, 2019-11-07:
Hi

In file bellow, you have the extracted parameters

With respect,

Nusret
[ExtraxtedParameters.txt](https://github.com/melexis/mlx90640-library/files/3820406/ExtraxtedParameters.txt)

nusret103, 2019-11-07:
Hi,

Please find the extracted parameters in the file below

With respect,

Nusret

[ExtraxtedParameters.txt](https://github.com/melexis/mlx90640-library/files/3820434/ExtraxtedParameters.txt)

slavysis, 2019-11-08:
Hi,

I see that you are not using the latest version of the driver. For the alpha values could you print out more digits - 6 digits is not enough to get the float value. I think %0.15f would be good enough.

Best regards

nusret103, 2019-11-08:
Hi

I am sending the new parameters

With respect,

Nusret
[ExtraxtedParameters_New.txt](https://github.com/melexis/mlx90640-library/files/3823621/ExtraxtedParameters_New.txt)

slavysis, 2019-11-08:
Hi, 
The EEPROM data is properly read and the extracted parameters are OK.
One thing that I find strange is that alpha[i] is correctly extracted, but in your first comment you report it as 'nan' or some strange number. Could you please check the memory handling? Is it possible that at some point you are overwriting the parameters?

Best regards

nusret103, 2019-11-08:
Hi
My first comment is when I try just to console out the alpha directly from inside of the method MLX90640_CalculateTo
Respect,

slavysis, 2019-11-08:
Hi,

That is what I also thought. I assume this is still the paramsMLX90640->alpha[i] values that you are reporting in the 1st comment. Is that right?

Best regards

nusret103, 2019-11-08:
From my perspective the EEPROM look ok.

The only thing where I have the issue is:
The normal flow software should run on the loop correct? On my side, that is the problem. 

When I call method MLX90640_GetImage some time I get the image some time not. and I think the software is hanging somewhere:

`    for( int pixelNumber = 0; pixelNumber < 768; pixelNumber++)
    {
        ilPattern = int((pixelNumber) / 32) - int(int((pixelNumber) / 32) / 2) * 2; 
        chessPattern = ilPattern ^ (pixelNumber - int(pixelNumber/2)*2); 
        conversionPattern = (int((pixelNumber - 2) / 4) - int((pixelNumber - 1) / 4) + int((pixelNumber + 1) / 4) - int((pixelNumber) / 4)) * (1 - 2 * ilPattern);
        
        if(mode == 0)
        {
          pattern = ilPattern; 
        }
        else 
        {
          pattern = chessPattern; 
        }
        
        if(pattern == frameData[833])
        { `

let me know what are your thoughts?

With respect

nusret103, 2019-11-08:
Hi
On this part of the code strange behavior is happening 

` if(pattern == frameData[833])`

With respect,

Nusret

slavysis, 2019-11-08:
Hi,

This if statement 

> if(pattern == frameData[833])

only ensures that the appropriate subpage is being processed. Can you check what are the values of frameData[833]. Under normal circumstances it should toggle between 0 and 1.

Best regards

nusret103, 2019-11-08:
Hi

In the file below, you have the param frame when the software is hanging

With respect

[paramData.txt](https://github.com/melexis/mlx90640-library/files/3824525/paramData.txt)

slavysis, 2019-11-08:
Hi,

When you say 'hanging', do you mean that the program halts and does not respond?

Best regards

nusret103, 2019-11-08:
yes

slavysis, 2019-11-08:
Hi,

In this part of the code, there is a for loop which should not be possible to hang. The only way I can think of for the program to hang here is if somehow you overwrite the stack and the function does not return properly. I still think you should check the memory allocations for the different arrays that are holding the data.

Best regards

nusret103, 2019-11-08:
Hi

I am pretty not overwriting the stack? I sent the raw param data. did you think the camera can be defective?

With respect,

Nusret

slavysis, 2019-11-08:
Hi,

So far, I think it is safe to conclude that:

1. The camera is responding to I2C commands
2. The EEPROM data is properly read
3. The parameters are properly extracted
4. The frame data is properly read
5. The frame data looks normal

Based on this I see no reason to think that the camera is defective.

The two problems that I see are:
1. The strange 'nan' values of the alpha[i] in your first comment
2. The 'hanging' of the program

Both could be explained with improper memory handling.
If you can share your 'main' I could look for some possible problems in it.

Best regards

nusret103, 2019-11-08:
The code below is not causing the hanging but there is all calculation logic is inside

````````
int MLX90640_GetImage(uint16_t *frameData, paramsMLX90640 *params, float *result)
{
    float vdd;
    float ta;
    float gain;
    float irDataCP[2];
    float irData;
    float alphaCompensated;
    uint8_t mode;
    int8_t ilPattern;
    int8_t chessPattern;
    int8_t pattern;
    int8_t conversionPattern;
    float image;
    
    vdd = MLX90640_GetVdd(frameData, params);
    ta = MLX90640_GetTa(frameData, params);
    
//------------------------- Gain calculation -----------------------------------    
    gain = frameData[778];
    if(gain > 32767)
    {
        gain = gain - 65536;
    }
    
    gain = params->gainEE / gain; 
  
//------------------------- Image calculation -------------------------------------    
    mode = (frameData[832] & 0x1000) >> 13;
    
    irDataCP[0] = frameData[776];  
    irDataCP[1] = frameData[808];
    for( int i = 0; i < 2; i++)
    {
        if(irDataCP[i] > 32767)
        {
            irDataCP[i] = irDataCP[i] - 65536;
        }
        irDataCP[i] = irDataCP[i] * gain;
    }
    
    irDataCP[0] = irDataCP[0] - params->cpOffset[0] * (1.0f + params->cpKta * (ta - 25.0f)) * (1.0f + params->cpKv * (vdd - 3.3f));
    if( mode ==  params->calibrationModeEE)
    {
        irDataCP[1] = irDataCP[1] - params->cpOffset[1] * (1.0f + params->cpKta * (ta - 25.0f)) * (1.0f + params->cpKv * (vdd - 3.3f));
    }
    else
    {
        irDataCP[1] = irDataCP[1] - (params->cpOffset[1] + params->ilChessC[0]) * (1.0f + params->cpKta * (ta - 25.0f)) * (1 + params->cpKv * (vdd - 3.3f));
    }
    
    for( int pixelNumber = 0; pixelNumber < 768; pixelNumber++)
    {
        ilPattern = int((pixelNumber) / 32) - int(int((pixelNumber) / 32) / 2) * 2; 
        chessPattern = ilPattern ^ (pixelNumber - int(pixelNumber/2)*2); 
        conversionPattern = (int((pixelNumber - 2) / 4) - int((pixelNumber - 1) / 4) + int((pixelNumber + 1) / 4) - int((pixelNumber) / 4)) * (1 - 2 * ilPattern);
        
        if(mode == 0)
        {
          pattern = ilPattern; 
        }
        else 
        {
          pattern = chessPattern; 
        }
//        if(pattern == frameData[833])
//        {    
//            irData = frameData[pixelNumber];
//            if(irData > 32767)
//            {
//                irData = irData - 65536;
//            }
//            irData = irData * gain;
//            irData = irData - params->offset[pixelNumber]*(1 + params->kta[pixelNumber]*(ta - 25.0f))*(1.0f + params->kv[pixelNumber]*(vdd - 3.3f));
//            if(mode !=  params->calibrationModeEE)
//            {
//              irData = irData + params->ilChessC[2] * (2 * ilPattern - 1) - params->ilChessC[1] * conversionPattern; 
//            }
//            
//            irData = irData - params->tgc * ((1-pattern) * irDataCP[0] + pattern * irDataCP[1]);
//            
//            alphaCompensated = (params->alpha[pixelNumber] - params->tgc * ((1-pattern) * params->cpAlpha[0] + pattern * params->cpAlpha[1]));
//            
//            image = irData/alphaCompensated;
//            
//            result[pixelNumber] = image;
//        }
    }  
}
````````

slavysis, 2019-11-11:
Hi,

How did you declare the array that the result pointer is referencing?

Best regards

nusret103, 2019-11-11:
Hi,

Please find the main loop what I am using

```````````````````
#include "mbed.h"
#include <MLX90640_I2C_Driver.h>
#include <MLX90640_API.h>

#define FRAMES_NUM 15
#define TA_SHIFT 8

Serial pc(USBTX, USBRX);                                                        // tx, rx, Virtual serial port over USB
void PrintEEPROM(uint16_t *p);
void PrintRawData(uint16_t *p);
void PrintToF(float *p);
void PrintToU(float *p);

DigitalOut led_green(LED2);

char buffer[3076]={'S'};

int main()
{
    uint8_t printme=1;
    MLX90640_I2CFreqSet(1000);
    pc.baud(115200);
    paramsMLX90640 mlx90640;
    uint8_t slaveAddress = 0x33;
    uint16_t *pEE;
    uint16_t *pFrame;
    float *toFrame;
    static uint16_t eeMLX90640[832];
    static uint16_t mlx90640Frame[834];
    static float mlx90640To[768];
    float emissivity = 1;
    float eTa;          //Ta for emissivity compensation
    float vdd;          //voltage of the circuit inside the camera

    pEE = eeMLX90640;
    pFrame = mlx90640Frame;
    toFrame = mlx90640To;
    MLX90640_SetRefreshRate(slaveAddress,5);
    MLX90640_SetResolution(slaveAddress,0x03);
    MLX90640_SetInterleavedMode(slaveAddress);
    MLX90640_I2CFreqSet(400);
    MLX90640_DumpEE(slaveAddress, pEE);
    MLX90640_I2CFreqSet(1000);
    MLX90640_ExtractParameters(pEE, &mlx90640 ); 
    
    PrintRawData(pEE);
       
    pc.printf("kVdd = %d\r\n",mlx90640.kVdd);
    pc.printf("vdd25 = %d\r\n",mlx90640.vdd25);
    pc.printf("KvPTAT = %f\r\n",mlx90640.KvPTAT);
    pc.printf("KtPTAT = %f\r\n",mlx90640.KtPTAT);
    pc.printf("vPTAT25 = %d\r\n",mlx90640.vPTAT25);
    pc.printf("alphaPTAT = %d\r\n",mlx90640.alphaPTAT);
    pc.printf("gainEE = %d\r\n",mlx90640.gainEE);
    pc.printf("tgc = %f\r\n",mlx90640.tgc);
    pc.printf("cpKv = %f\r\n",mlx90640.cpKv);
    pc.printf("cpKta = %f\r\n",mlx90640.cpKta);
    pc.printf("resolutionEE = %d\r\n",mlx90640.resolutionEE);
    pc.printf("calibrationModeEE = %d\r\n",mlx90640.calibrationModeEE);
    pc.printf("KsTa = %f\r\n",mlx90640.KsTa);
    
//    for(int i = 0; i < 4; i++){pc.printf("ksTo[%d] = %f\r\n",i, mlx90640.ksTo[i]);}
//    for(int i = 0; i < 4; i++){pc.printf("ct[%d] = %d\r\n",i, mlx90640.ct[i]);}
//    for(int i = 0; i < 768; i++){pc.printf("alpha[%d] = %0.15f\r\n",i, mlx90640.alpha[i]);}
//    for(int i = 0; i < 768; i++){pc.printf("offset[%d] = %d\r\n",i, mlx90640.offset[i]);}
//    for(int i = 0; i < 768; i++){pc.printf("kta[%d] = %f\r\n",i, mlx90640.kta[i]);}
//    for(int i = 0; i < 768; i++){pc.printf("kv[%d] = %f\r\n",i, mlx90640.kv[i]);}
//    for(int i = 0; i < 2; i++){pc.printf("cpAlpha[%d] = %f\r\n",i, mlx90640.cpAlpha[i]);}
//    for(int i = 0; i < 2; i++){pc.printf("cpOffset[%d] = %d\r\n",i, mlx90640.cpOffset[i]);}
//    for(int i = 0; i < 3; i++){pc.printf("ilChessC[%d] = %f\r\n",i, mlx90640.ilChessC[i]);}
//    for(int i = 0; i < 5; i++){pc.printf("brokenPixels[%d] = %d\r\n",i, mlx90640.brokenPixels[i]);}
//    for(int i = 0; i < 5; i++){pc.printf("outlierPixels[%d] = %d\r\n",i, mlx90640.outlierPixels[i]);}

        while (true){
            wait_us(500000);
            led_green = 1;    // turn on LED until initialization is complete
            MLX90640_GetFrameData(slaveAddress, pFrame);
            led_green = 0;    // turn on LED until initialization is complete
            //eTa = MLX90640_GetTa(pFrame, &mlx90640) - TA_SHIFT;
            //MLX90640_CalculateTo(pFrame, &mlx90640, emissivity, eTa, mlx90640To);
            MLX90640_GetImage(pFrame, &mlx90640, toFrame);
//            for(int i = 0; i < 768; i++){pc.printf("alpha[%d] = %0.15f\r\n",i, mlx90640.alpha[i]);}
//            for(int i = 0; i < 768; i++){pc.printf("offset[%d] = %d\r\n",i, mlx90640.offset[i]);}
//            for(int i = 0; i < 768; i++){pc.printf("kta[%d] = %f\r\n",i, mlx90640.kta[i]);}
//            for(int i = 0; i < 768; i++){pc.printf("kv[%d] = %f\r\n",i, mlx90640.kv[i]);}
            
            if (printme)
            {            
                PrintRawData(pFrame);
            }
        }
    //}

}

void PrintToU(float *p)
{
    uint16_t idx=1,i;
    uint16_t d;    
    for(i=0; i<768; i++) 
    {
        d = 10 * *p++;
        sprintf(buffer+idx,"%04x",d);
        idx+=4;
    }
    buffer[idx++] = 'E';
    buffer[idx++] = '\n';
    buffer[idx++] = 0;
    pc.printf(buffer);
}

void PrintToF(float *p)
{
    for(int i=0; i<768; i++) 
    {
        pc.printf("%0.3f,",*p++);
    }
    pc.printf("\n");

}

void PrintRawData(uint16_t *p)
{
//    for(int i=0; i<832; i++) {
//        pc.printf("%04X,",*p++);
//    }
//    pc.printf("\r\n");
    
    for(int i = 0; i < 26; i++)
    {
        for(int j = 0; j < 32; j ++)
        {
            pc.printf("%04X,",*p++);
        }
        pc.printf("\n");
    }

}

```````````````````

slavysis, 2019-11-11:
Hi, 
The code looks OK.
Can you print out the mlx90640 struct from inside the GetImage subroutine?
The goal is to check if the parameters are still the same as after the ExtractParameteres call.

Best regards

nusret103, 2019-11-11:
Hi

I can't do that because the Serial is defined on the main loop. But I check that one yesterday and after calling the MLX90640_GetFrameData(...) the structure bellow is mess up.
That means when i MLX90640_GetFrameData(...)  i gett the value for: alpha= nan or something
alpha= -340279649078051570000000000000000000000.000000

```
  typedef struct
    {
        int16_t kVdd;
        int16_t vdd25;
        float KvPTAT;
        float KtPTAT;
        uint16_t vPTAT25;
        uint16_t alphaPTAT;
        int16_t gainEE;
        float tgc;
        float cpKv;
        float cpKta;
        uint8_t resolutionEE;
        uint8_t calibrationModeEE;
        float KsTa;
        float ksTo[4];
        int16_t ct[4];
        float alpha[768];    
        int16_t offset[768];    
        float kta[768];    
        float kv[768];
        float cpAlpha[2];
        int16_t cpOffset[2];
        float ilChessC[3]; 
        uint16_t brokenPixels[5];
        uint16_t outlierPixels[5];  
    } paramsMLX90640;
```

nusret103, 2019-11-11:
Hi,

This definitely has to do with pointers and i can't figure out what is the issue

find the file of values of alpha before calling method MLX90640_GetFrameData(..) and after.

With respect,

Nusret

[alpha_after.txt](https://github.com/melexis/mlx90640-library/files/3831990/alpha_after.txt)
[alpha_before.txt](https://github.com/melexis/mlx90640-library/files/3831991/alpha_before.txt)

slavysis, 2019-11-12:
Hi,

Did you try commenting out the line where you are actually writing to the memory using the pointer?

> result[pixelNumber] = image;

 Best regards

nusret103, 2019-11-12:
Hi

When I am I remove the comment of the previous line the software is hanging.

Did you have any hello world where I can see this is working arduino, rasbery pi, mbed to start from there?

With respect,

Nusret

slavysis, 2019-11-12:
Hi,

I am running your code since yesterday on a MBED LP1768 and it is still working.
The problem might be with some float alignment that results in a error causing the MCU to halt. Perhaps, you could try the newest version of the driver. It uses 16bit ints to store the alpha.

Best regads

nusret103, 2019-11-12:
Hi

please, can you send me the your working code for LPC 1768 I have one board here?

With respect,

Nusret

slavysis, 2019-11-12:
Hi,

I just copy->pasted your main code https://github.com/melexis/mlx90640-library/issues/46#issuecomment-552358258.

Best regards

slavysis, 2019-11-12:
Hi,

I still have  the feeling that you are actually overwriting the stack and therefore you 
ruin the extracted parameters data.
Did you try declaring the parameters as  

> static paramsMLX90640 mlx90640;

Best regards

nusret103, 2019-11-12:
Hi,
I am testing that one now,

The temperature should be in Kelvin? I will come back to you

With respect,

Nusret

slavysis, 2019-11-12:
Hi,

The temperature is in °C. When you use the GetImage function, the returned values are not temperatures, but values relative to the pixels temperatures so that you still get the thermal image.

Best regards

nusret103, 2019-11-13:
Hi

The Temperature from function CalculateTo is in K but we fixed that one from the main loop part.

Thanks for everything.

With respect,

Nusret

closed 2019-11-13 by nusret103

eleogrande, 2023-04-21:
Hi, where i can find all the files needed to run this project on mbed on STMF429 with mlx90640? I saw that there is the "main" in the comments, but I would like to know which ".h" and ".cpp" files you used