melexis/mlx90640-library #47: Subpage 0 pixel values are all 0 for first few frames

Patel-Apurva, 2019-11-06:
Hello! I am using the MLX90640-BAA with ESP32. Everytime I power on the ESP32, the first few frames shows 0s for subpage 0 pixels. After a couple of frames, all pixels show temperature values. I2C frequency is set to 100kHz and refresh rate is set to 64Hz. 
When I2C frequency is set to 999990Hz and refresh rate is set to 32Hz, the problem only occurs for the very first frame then resumes normal operation. Why does the problem persist longer with the initial settings? And is this normal behaviour?

slavysis, 2019-11-07:
Hello! 
The readout of the data at I2C set to 100KHz would take around 75ms, while there is a new frame data available every 16ms. This would cause the GetFrameData function to return error -8 every time. Error -8 means that the data is not reliable. In order to get reliable and predictable performance I would first sort out the timings. Note that the readout time and the data processing time should be less than the refresh time so that you do not skip frames. If skipping frames is OK for you, you should at least have the readout time to be less than the refresh time. Perhaps this comment from a previous issue would be helpful
[(https://github.com/melexis/mlx90640-library/issues/33#issuecomment-484881185)](url)

Best regards

Patel-Apurva, 2019-11-07:
Thank you for your explanation. I have understood why the problem happens but could you explain how you calculated the 75ms readout time for frequency of 100kHz?

slavysis, 2019-11-08:
Actually, at 1KHz the readout time is about 150ms as the function always reads the whole frame.
It is rather simple calculation: every time you read out a data frame, you have to receive 834 16-bit words of data. I2C transmits a byte (8 bits) plus ACK. That makes 834x2x9 = 15012 bits that should be transmitted. There is also some overhead because of the command that needs to be send. At 100KHz, one bit is being transmitted every 10us. So 15012bits x 10us = 150.012ms + some microseconds for the command.

Patel-Apurva, 2019-11-08:
Do you mean 100kHz instead of 1kHz? Thank you for your help anyways! It has helped me to determine the I2C frequency and refresh rate I need to set.

slavysis, 2019-11-13:
Yes, indeed. Sorry for the typo.

Best regards

closed 2019-11-13 by Patel-Apurva