melexis/mlx90640-library #51: chessboard pattern data

LCAR979, 2019-11-22:
The question has confused my for a long time. After I read data frames, the temperature values see a  strong chessboard-pattern style. See images below (raw temperature data)

[image]

Part of measured values:
[image]

I marked some obviouslly abnormal points. 

And here is my core measuring code :
```
        int expectFrame = 0;
        int subPagesRead = 0;

        while (subPagesRead < 2) {
            vTaskDelay(250 / portTICK_RATE_MS);
            status = MLX90640_GetFrameData(0x33, mlx90640Frame);
            if (status != expectFrame) {
                continue;
            }
            expectFrame = 1 - expectFrame;
            subPagesRead++;
            float tr = MLX90640_GetTa(mlx90640Frame, &mlx90640) - TA_SHIFT;
            MLX90640_CalculateTo(mlx90640Frame, &mlx90640, emissivity, tr, mlx90640To);
        }
        // output mlx90640To
        memset(mlx90640Frame, 0, sizeof(mlx90640Frame));
        memset(mlx90640To, 0, sizeof(mlx90640To));
        subPagesRead = 0;
    }
```

Am I measuing in a wrong way? Or any other possible reasons? 

(This frame is calculated at 4Hz refrash rate on a esp32 chip)

LBuydens, 2019-11-22:
Often this is due to missing power decoupling capacitors which have to be close to the senosr.

LCAR979, 2019-11-25:
@LBuydens . Thanks for your help. After adding capacitors the data looks quite normal now.  I will close this issue.

closed 2019-11-25 by LCAR979