melexis/mlx90640-library #52: Not detecting device

jluus, 2019-11-27:
I have the Sparkfun MLX90640 sensor connected to Teensy 3.6 over I2C. I can not connect to the sensor. The message I get is "MLX90640 not detected at default I2C address". I am running the basic example from the supplied code. The I2C scanner does also not detect the device. 

Any suggestions?

slavysis, 2019-12-02:
Looking at the sparkfun code, it seems like you are getting this message when the MLX90640 device does not acknowledge a communication attempt. As the rest of the message suggests, you should first check the wiring. If the supply is OK and the pull-ups are properly connected, the device should respond and you should not get that message.

Best regards

jluus, 2019-12-02:
Thanks for the reply. I have checked the connections multiple times. I noticed that others on the sparkfun forum have the same issue. Other sensors are working and I have tried many versions of the libraries. And according to the website, "The MLX90640 IR Array has pull up resistors attached to the I2C bus", so no extra resistors were used. [Website](https://www.sparkfun.com/products/14844)

slavysis, 2019-12-03:
Did you check the communication with some oscilloscope and/or logical analyzer? The pull-ups are there, but it is still possible that something went wrong there. Did you check the supply voltage?

Best regards

jluus, 2019-12-03:
Supply voltage is good, I will try to get my hands on an oscilloscope, thanks.

jluus, 2019-12-16:
The problem was a hardware issue. Sparkfun said that it is a known issue and were very helpful. I bought the sensor from Pimoroni and it works well. Thanks for the help.

closed 2019-12-16 by jluus