melexis/mlx90640-library #54: Connecting multiple MLX90640 to one Raspberry Pi 4

mariongbr, 2019-12-08:
Dear MLX90640 experts!

Currently we face some issues while we connect multiple MLX90640-device (actually via a Qwiic eval board) to one RPI. The goal is to connect 4 different devices to one master. We tested in different setups:
1. We managed to change the I2C addresses. Monitoring the addresses with i2cdetect at the RPI 3 B+ also shows us that there are different I2C deviced connected on the bus. The Qwiic-board are connected in a chained way to each other such that there is a line of 3 or 4 modules. Between the modules there is 20cm Qwiic cable. So it not too long such that the capacity might increase too much.
2. We issued the i2cdetect several times to check if all devices are present. Sometimes we get all addresses correct and then there might be situations when we only get garbage in all addresses as a i2cdetect result. We doublechecked many times and we are sure that there is no loose contact in the connection. So this is a really serious problem since we do not see stability with several devices at one I2C.
3. The drivers for reading out the data from the MLX90640 are working well. We also used the test-programm from https://github.com/pimoroni/mlx90640-library/blob/master/examples/src/test.cpp with which it is possible to get out an ASCII-like thermal picture.
4. The suspicion was that the problem has something to do with the clock stretching bug in RPI 3. So we changed to the RPI 4 and after some update problems got it to work again. But the stability is still poor we can not really rely on the devices if there are multiple devices connected.
5. On both platforms, the RPI 3 and 4 we experimented with the baud rate. 100000 and 400000 made no difference in the stability.
6. The next idea was that there might be something with the bus, the capacitance or some other strange electrical behaviour. So we removed the pull-up resistors on Qwiic board, since the RPI 3 and 4 have their own pull-up resistors on board. Again, this did not lead to any stability improvement.
7. We monitored the I2C with oscilloscope. The shape of the signal is almost perfect like in the textbook for I2C. There is only a very short increase phase to see, so actually nothing which points to electrical bus issues. While monitoring the bus we realized that sometimes the messages for i2cdetect are not present on the bus. And this then leads to garbage as a result in the i2cdetect result. Sometimes this state last forever and only reducing the number of devices connected on the bus from 4 to 3 or from 3 to 2 removes the problem.
8. We already were able to shoot 3 pictures from 3 different MLX90640 at once and got the picture out of our software. But this is not possible repeatedly and therefor very instable.

So finally, this boils down to our questions:
Has anybody seen such kind of stability problems when working with the RPI 3/4 and multiple MLX90640?
Is there any known issue for the I2C implementation of Raspbian which might lead to such a strange behaviour?
Is there any known issue in connecting multiple MLX90640 (maybe for other masters than RPI) to the I2C?
Are there better ways to acquire pictures with about 2-3 Hz max from four MLX90640 at the same time?

Thank you in advance for your help!
mario

slavysis, 2019-12-09:
Hi,

How do you connect the devices on the bus? Are you using some sockets, or you are soldering the parts?

> While monitoring the bus we realized that sometimes the messages for i2cdetect are not present on the bus
Does that mean the SDA and SCL signals stay high?
From what I read, it seems like you might be having troubles with the power supply.
Did you check that there are no shorts between the pins of the devices?

Best regads

Abbad-Zafar, 2020-11-19:
How did you add multiple mlx90640 to raspbbery ? can you send me a code ?