melexis/mlx90640-library #55: frame data is full of NAN

natyshemer, 2019-12-26:
Hi,

I have been trying to get the MLX90640 to work on the up-board Linux based.
I keep getting NAN and non valid values from the sensor.
I have tested all functions according to the excel document and they seem to be working.
I have tested the hardware by using a Python code and that also worked fine.
I have tried to minimize the buffer size  in the I2C and only sample 32 characters at each read and that didn't change anything.

I am attaching the EEPROM data which I think looks fine and follows by that the frame data which is not ok.
[paramResults.txt](https://github.com/melexis/mlx90640-library/files/4002691/paramResults.txt)

Would really appreciate some ideas on how to solve this..

Thanks

slavysis, 2020-01-06:
Hi,

It seems like the extracted parameters are not correct. Is it possible to log the raw EEPROM data?

Best regards

natyshemer, 2020-01-06:
Hi,
I am not sure what you mean by the raw EEPROM data, 
Below is my code for extracting the parameters,
Thanks you for your assistance
Naty

`

    float mlx90640To[768];
    int status;
    uint16_t eeMLX90640[832];

    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);
    if (status != 0)
    {
        printf("Failed to load system parameters");
        return 0;
    }

    status = MLX90640_ExtractParameters(eeMLX90640, &mlx90640);
    if (status != 0)
    {
        printf("Parameter extraction failed %d\n", status);

        return 0;
    }
    printf("Got Params");
    if (1)
    {
        std::cout << "kVdd:" << mlx90640.kVdd << std::endl;
        std::cout << "vdd25:" << mlx90640.vdd25 << std::endl;
        std::cout << "KvPTAT:" << mlx90640.KvPTAT << std::endl;
        std::cout << "KtPTAT:" << mlx90640.KtPTAT << std::endl;
        std::cout << "vPTAT25:" << mlx90640.vPTAT25 << std::endl;
        std::cout << "alphaPTAT:" << mlx90640.alphaPTAT << std::endl;
        std::cout << "gainEE:" << mlx90640.gainEE << std::endl;
        std::cout << "tgc:" << mlx90640.tgc << std::endl;
        std::cout << "cpKv:" << mlx90640.cpKv << std::endl;
        std::cout << "cpKta:" << mlx90640.cpKta << std::endl;
        std::cout << "resolutionEE:" << (int)mlx90640.resolutionEE << std::endl;
        std::cout << "KsTa:" << mlx90640.KsTa << std::endl;
        for (int i = 0; i < sizeof(mlx90640.ksTo) / sizeof(float); i++)
            std::cout << "ksTo[" << i << "]:" << mlx90640.ksTo[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.ct) / sizeof(int16_t); i++)
            std::cout << "ct[" << i << "]:" << mlx90640.ct[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.alpha) / sizeof(float); i++)
            std::cout << "alpha[" << i << "]:" << mlx90640.alpha[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.offset) / sizeof(int16_t); i++)
            std::cout << "offset[" << i << "]:" << mlx90640.offset[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.kta) / sizeof(float); i++)
            std::cout << "kta[" << i << "]:" << mlx90640.kta[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.kv) / sizeof(float); i++)
            std::cout << "kv[" << i << "]:" << mlx90640.kv[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.cpAlpha) / sizeof(float); i++)
            std::cout << "cpAlpha[" << i << "]:" << mlx90640.cpAlpha[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.cpOffset) / sizeof(int16_t); i++)
            std::cout << "cpOffset[" << i << "]:" << mlx90640.cpOffset[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.ilChessC) / sizeof(float); i++)
            std::cout << "ilChessC[" << i << "]:" << mlx90640.ilChessC[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.brokenPixels) / sizeof(float); i++)
            std::cout << "brokenPixels[" << i << "]:" << mlx90640.brokenPixels[i] << std::endl;

        for (int i = 0; i < sizeof(mlx90640.outlierPixels) / sizeof(int16_t); i++)
            std::cout << "outlierPixels[" << i << "]:" << mlx90640.outlierPixels[i] << std::endl;
    }
`

slavysis, 2020-01-06:
Hi,

I mean the data returned from MLX90640_DumpEE().

Best regards

natyshemer, 2020-01-06:
Hi,
hoo, Ok, should have known that... 
Is the format ok?
Thanks
[eepromRawData.txt](https://github.com/melexis/mlx90640-library/files/4025259/eepromRawData.txt)

slavysis, 2020-01-06:
Hi,

The EEPROM content does not seem right. Could you try reading it several times and compare the values that you get?

Best regards

natyshemer, 2020-01-06:
Hi, 

ran the following code (no delay between eeprom readings), got the same results each time
[eepromRawData5.txt](https://github.com/melexis/mlx90640-library/files/4025594/eepromRawData5.txt)

 ```
  status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);
    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);
    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);

    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);

    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);
    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
    status = MLX90640_DumpEE(MLX90640_address, eeMLX90640);

    for (int i = 0; i < sizeof(eeMLX90640) / sizeof(uint16_t); i++)
    {
        std::cout << eeMLX90640[i] << ";";
    }
    std::cout << "end" << std::endl;
```

slavysis, 2020-01-06:
Hi,

Could you then please check with an oscilloscope or a logical analyzer.
Some of the FF values in the MSbyte are suspicious. For example at address 00, the expected value is 0x00A3 instead of 0xFFA3 that you read, address 0x06 I would expect to read 0x03E0 instead of 0xFF30 that you read.

Best regards

natyshemer, 2020-01-07:
Hi,

Yeah , i see what you mean. I have a got python library that is working so I'll compare the cpp and python signal and see what i find.
Thank you for your assistance.
Naty

natyshemer, 2020-01-23:
Hi,
So I have compared the two I2C inputs, one from a python script that is working and the other from my cpp code which does not work. I can see the difference but unfortunately I don't really understand enough I2C protocol to define the problem.
Any chance you could help me define my signal problem?

That's from my CPP non working library:
[image]
[T0001.txt](https://github.com/melexis/mlx90640-library/files/4101835/T0001.txt)

That's from my working Python library:
[image]
[T0000.txt](https://github.com/melexis/mlx90640-library/files/4101843/T0000.txt)

closed 2020-11-03 by slavysis