melexis/mlx90640-library #56: Strips of pixels that are ~2 degrees different to nearby pixels

zdakun, 2020-01-08:
I used MLX90640-BAA with raspberry pi to read temperature data.
I have about two dozen such sensors, and on all of them, my temperature reading have such vertical strip when I aim at a flat surface with same temperature. I have put C1=1u and C2=100n, and R1 R2 both at 1K, as in datasheet 13.1.
Measuring VDD-GND, the voltage is very stable.
[image]
[image]

Temperature difference between the stripes are ~ 2degrees celsius
[image]
[image]
[image]

Any idea how to solve this problem?

slavysis, 2020-01-10:
Normally the C1 capacitance should be 10uF.

Best regards

zdakun, 2020-01-11:
I have changed capacitor to larger ones up to 47uF, without much effect. Any specific requirement for the type of capacitor? I used ceramic capacitor

hosseinzadeh88, 2020-05-04:
I am having exactly the same issue with my sensor and I'm also using raspberry pi (4). Tried changing the I2C SCL but nothing! The stripes are almost always there, they occasionally disappear for a frame or two and come back again!

hosseinzadeh88, 2020-05-04:
> I have changed capacitor to larger ones up to 47uF, without much effect. Any specific requirement for the type of capacitor? I used ceramic capacitor

I have had the same issue. Until reading a couple of the posts here. 
One suggested that it's the cabling (that did not help me at all, I even soldered the wires and made them as short as possible). https://github.com/melexis/mlx90640-library/issues/9

The second suggestion on this thread was to check the capacitors. I checked them as much as I could, even desoldered one of them and measured it. They were all ok and within the expected range. Eventually added a 100uf electrolytic caps (as far as I remember it is rated for 16 volts) between the GND and VDD of the sensor and that did it for me. No more stripes.

I tried to replace the cap with a bunch of SMD ceramic caps (as they were smaller and made things look more elegant!) but they didn't work and had to resort back to the electrolytic cap (16v 100uf).

I just hope that this doesn't mess with the temperature measurement accuracy ....