melexis/mlx90640-library #57: RuntimeError("Too many retries")

matiasandina, 2020-02-06:
I have been using this camera for an application at 32 Hz. It works fine, however, I get `RuntimeError("Too many retries")` quite often (camera can't run for 5 minutes straight).
Any advice on how to catch this error and restart the camera or any chance you can increase the `cnts` value in `_GetFrameData`?

slavysis, 2020-02-07:
The GetFrameData function returns -8 when valid data can not be acquired for 5 'frames'. This could help you to 'catch' the error and deal with it. If you want to, you could simply call the function until you get valid data. Be careful though to not create an endless loop and hang the MCU. The reason for getting this error is that the I2C read-out time is too much compared to the refresh rate of the sensor. Even if the cnt is increased you are still dropping frames.

Best regards

closed 2020-04-08 by matiasandina

slavysis, 2020-04-13:
New version uploaded with no multiple retries.

Best regards