melexis/mlx90640-library #58: Mlx90640 frame 0 and frame 1

HelloBirthday, 2020-02-25:
Hello, I'm debugging the mlx80640 sensor. Question in getting frame data? Mlx90640 ﹐ getframedata() obtains the frame data. The return value of 0 indicates that the data of subpage 0 has been read, and the return value of 1 indicates that the data of subpage 1 has been read. Does MLX90640_GetFrameData() only read one frame of data at a time? Do you need to call MLX90640_GetFrameData() several times to get frame 0 and frame 1 respectively, and then synthesize the acquired 2 frames of data?

I looked at the sample data book and called it only once.

slavysis, 2020-02-26:
Hi,

The MLX90640_GetFrameData() function reads the whole frame data and indicates which is the most recent subframe. If the returned value is 0, this means that the most recent subframe is frame 0 and if the returned value is 1 - it is sub-frame 1. Note that when you call the CalculateTo function, it will calculate only the temperatures for the most recent subframe. The best would be if your code takes less time (reading + data processing) than the refresh time of the sensor. You should call the GetFrameData once and then call the CalculateTo function (or GetImage if you do not need absolute temperature values).  

Best regards

HelloBirthday, 2020-02-27:
Glad to see your reply.

I have understood what you mean. I will first mlx90640 ﹐ getframedata once until the data of frame 0 is returned, and then delay 50ms, and then call mlx90640 ﹐ getframedata to get the data of frame 1 (cycle to get, frame 1 needs twice in most cases). I will save the data of frame 0 and frame 1 respectively. After the two readings are complete, I will use calculateto Deal with them separately, and finally merge frame 1 into frame 0 to get the data I want. Based on this processing, my program gets 768 temperature values perfectly.

Thank you for all your assistance.

closed 2020-02-27 by HelloBirthday