melexis/mlx90640-library #61: EEPROM have gotten corrupted

motorslug, 2020-04-08:
Hi everyone,

I have two MLX90640 sensors that seem to have gotten their EEPROM corrupted after being connected to the Nvidia Jetson. Currently I'm using a Raspberry Pi to try to recover them. Attempting to read a frame results in "frameData timeout error waiting for dataReady". I can still dump the data out of the EEPROM, and I noticed that if I run the "step" example (which fails) and then the "rawrgb" example, I can see an image but the temperature values are miscalibrated. Seems to be related to the MLX90640_StartMeasurement(MLX_I2C_ADDR, 0); function which allows me to get data from the sensor. Any suggestions on how the EEPROM can be restored? This seems to be related to this issue: https://github.com/pimoroni/mlx90640-library/issues/38

motorslug, 2020-04-08:
Here's a dump of the EEPROM (can rename to CSV)
[eeprom.txt](https://github.com/melexis/mlx90640-library/files/4451380/eeprom.txt)

slavysis, 2020-04-08:
Hi, the driver never writes to the EEPROM. It sounds like some I2C issue where a read command is misinterpreted and the MLX90640 writes data to the EEPROM. Since both the MLX90640 and the Nvidia are expecting the other device to take control of the SDA, no device does that and the SDA stays high all the time = the MLX90640 writes FF to the EEPROM address instead of reading the data from it. You should first make sure that you can get proper I2C communication with the Nvidia Jeston. If you can fetch the SDA and SCL of a faulty communication I could help you identify the problem.

Best regards

motorslug, 2020-04-08:
Thanks for your reply. At the moment I'm not planning to use it with the Jetson anymore, have it connected to a Pi 3B instead, planning to keep using it on the Pi platform. 

Unfortunately it still exhibits the same behaviour at the moment. I2C communications don't seem to be the problem as I can swap the sensor out for another one (verified to be working) and confirm that it works with the same code and settings.  

For the malfunctioning sensor, I noticed that I have to write to the Status register (0x8000) to reset the bit before starting to reading frames, then I can read frames from it. Temperature readings are still off, the entire temperature range seems to be between 23 and 50 degrees Celsius. (see attached screenshot, should be a range of at least 15-36 degrees as per my other sensor)

I suspect that the Status register bit is stuck (I have to reset it after every restart before I can read frames) and some calibration data may be missing (seems so due to the narrow temperature range the sensor outputs). 

Here's the function I'm using to write to the registers (taken from PiMoroni). After using this I can start getting data from the sensor:

void MLX90640_StartMeasurement(uint8_t slaveAddr, uint8_t subPage)
{
    uint16_t controlRegister1;
    uint16_t statusRegister;
    MLX90640_I2CRead(slaveAddr, 0x800D, 1, &controlRegister1);
    controlRegister1 &= 0b1111111111101111;
    controlRegister1 |= subPage << 4;
    MLX90640_I2CWrite(slaveAddr, 0x800D, controlRegister1);
    MLX90640_I2CRead(slaveAddr, 0x8000, 1, &statusRegister);
    statusRegister &= 0b1111111111110111; // Clear b3: new data available in RAM
    statusRegister |= 0b0000000000110000; // Set b5: start of measurement
                                          // Set b4: enable RAM overwrite
    MLX90640_I2CWrite(slaveAddr, 0x8000, statusRegister);
}

[image]

slavysis, 2020-04-08:
Hi,

The code is working properly on the Pi, but I suspect the I2C functions do not work properly on the Nvidia. The damage to the sensor is already done and the Calibration parameters in the EEPROM were overwritten with 0xFF. The configuration register value is also 0xFF which is not correct. You can try to write the default value (0x1901) to the control register directly (instead of reading the value from the EEPROM). I still do not think you will be able to get correct values as there is 0xFF all over the EEPROM.

Best regards

motorslug, 2020-04-08:
Thanks, will keep in mind to keep the Jetson away from these sensors. Writing the default value to the control register should be okay. For the calibration values, are they unique for every sensor? Wondering if it's worth trying to get the values from a working sensor and try to write those to the EEPROM.

slavysis, 2020-04-08:
The values are unique for every sensor. Copy pasting values from another sensor might improve the performance, but do not expect the sensor to perform within the specification. Can you try to upload the attached file into the EEPROM?
[0523-0218-0187.xlsx](https://github.com/melexis/mlx90640-library/files/4452391/0523-0218-0187.xlsx)

motorslug, 2020-04-09:
I uploaded the values to the EEPROM, looks better now, thanks! Seems somewhat more sensitive now. Temperature values are still showing as off, but I can live with that.  Are those the default values?

slavysis, 2020-04-13:
This should be the original EEPROM content. If the temperature is off, there should be something else to it. If you have some hardcoded values or setting.

Best regards

motorslug, 2020-04-25:
I'll test further, may be something with my code. I'll close the issue. Thanks for your help!

closed 2020-04-25 by motorslug

sarvesh-edgeneuralai, 2020-07-06:
I m also facing the same issue. I have two waveshare MLX90640 sensors connected to Jetson. First one was connected, now getting bad temperature values, but Ta values are correct, but temperature pixels values are off by around 20 degree celsius. The new MLX90640 waveshare sensor, works well and gives correct temperature values. Same code works for both, but one gives bad temperature pixels and other one gives good one. How do I fix it? Whats the way to upload EEPROM data, as suggested in the one of the above comment?

vikasbucha, 2023-02-05:
> I uploaded the values to the EEPROM, looks better now, thanks! Seems somewhat more sensitive now. Temperature values are still showing as off, but I can live with that. Are those the default values?

This could be a newbie question for MLX90640.
How to upload new values to EEPROM ?