melexis/mlx90640-library #62: Query for MLX90640 and MLX90632 IR Thermal Sensors

Ritesh236, 2020-04-16:
Hello Everypone,

Hope you are doing well and everyone is same at their end.

Actually, We are thinking to create IR Thermal Camera System and we felt to use Melexis Sensors for same.

We looked and found that either MLX90640 or MLX90632 can be used into our application.

But, I want to know much details from accuracy and measurement point of practical details which have been measured on those modules.

So, can anyone provide those kind of such details which will be beneficial for us to move further for same?

Regards,
Ritesh Prajapati

slavysis, 2020-04-17:
Hi,

MLX90632 is a single pixel device, so this is not really suitable for  IR Thermal camera.
The accuracy for the devices is specified in the datasheets.

Best regards

Ritesh236, 2020-04-17:
@slavysis ,

Thanks for your quick response.

So, Would you please suggest which one is best suitable module to use for IR Thermal Camera to measure non-contact body temperature?

Regards,
Ritesh Prajapati

Ritesh236, 2020-04-17:
@slavysis ,

I have also check MLX90614 Data-sheet which is also for single pixel device nut mentioned to use it for Body temperature measurement. So, I am little bit confused as multiple pixel supported module is better to use then single pixel module.

Regards,
Ritesh Prajapati

LBuydens, 2020-04-17:
Dear sir,

This channel is not monitored by Melexis for the questions you are asking.  
Please contact sales@melexis.com

Regards.

closed 2020-11-03 by slavysis