melexis/mlx90640-library #66: Lower than expected temperature readings

pritsios, 2020-05-02:
Use the library with a SAMD21 microcontroller.
Trying to measure forehead temperature, I get 4.5 degrees less than expected. After looking at the source of the driver I have the following questions

1. In "Restoring the KsTo coefficient" Why the driver calculate 5 values and the manual only 4
2.The same for ct restoring
3. In MLX90640_GetFrameData why use the data 0x0030 instead of 0x0000 to clear the "new data available flag" in the status register?
4. When measuring the ambient temperature I get 35.5 degrees, at list 9 more than the actual one 
3. When I change the emissivity to 0.8 I get correct readings, but is this ok? 

[image]

slavysis, 2020-05-03:
I will try to answer you questions:

1. The extra parameters are for temperatures above 400C and are sort of a future-proof thing. Note that this is well beyond the temperatures you are interested in
2. To ensure proper function
3. This is normal - the Ta of the sensor is not the same as the 'room' temperature. The difference depends on the enclosure. In open air it should be around 8C.
4. No, that is not correct as this is not the emissivity of the human skin.

Some general remarks:
1. I am not sure how you end up with the conclusion you have 4.5C lower then expected. If you were expecting to measure 36-36.5C forehead temperature, I would say your expections are too high. There is a relation between the forehead temperature, the room temperature and the body core temperature, but you need to apply a formula in order to get the body core temperature. As an example at room temperature around 23C a healthy (body core temperature 36.5C) person's forehead temperature would probably be around 33C.  
2. I am not sure if you are trying to build a medical device or you are just playing around, but note that the MLX90640 is **not** a medical grade sensor.

pritsios, 2020-05-03:
Thank you for your reply.

I am not trying to build a medical device but a device for tracking human budy in rooms and discover heat sources for an alarm system. I did not know that there is a formula to calculate core body temperature from forehead temperature.
 
1. Is it possible to measure the real tempereture of an object that does not cover all the FOV of the MLX90640?. In my attached picture the maximum temperature is 32.1C. If I completely fill the FOV its 34C and if I move back 1,5 meter the temperature drops at 28.5C. I know that this is due to the specs of the device (FOV, resolution etc), but I wonder if there is a method to cope with those differencies and calculate temperature regardless of the distance.
2. Do all the pixels of my MLX90640 have the same FOV=110 or is the FOV of every pixel something like: pixel FOV = Total FOV/resolusion?
3. What is the relation between sensitivity and FOV of the MLX90640. 100% sensitivity is at zero degrees, but what about the rest of the -90 to +90 area? Is there a formula for that?

Thank you in advance

kkaryono, 2020-07-14:
Hi, just want to share something that I found on the human temperature.
The human skin temperature is 4° or 5° C lower than the core temperature (based on A. C. Burton, "Human Calorimetry: II. The Average Temperature of the Tissues of the Body: Three Figures," The Journal of Nutrition, vol. 9, no. 3, pp. 261-280, 1935, doi: 10.1093/jn/9.3.261.). 

The effect of the ambient temperature in IR thermometer reading is shown in these works (F. Suarez, A. Nozariasbmarz, D. Vashaee, and M. C. Öztürk, "Designing thermoelectric generators for self-powered wearable electronics," Energy & Environmental Science, 10.1039/C6EE00456C vol. 9, no. 6, pp. 2099-2113, 2016, doi: 10.1039/C6EE00456C and
P. Webb, "Temperatures of skin, subcutaneous tissue, muscle and core in resting men in cold, comfortable and hot conditions," European Journal of Applied Physiology and Occupational Physiology, vol. 64, no. 5, pp. 471-476, 1992/09/01 1992, doi: 10.1007/BF00625070)

Kind Regards,

tegaete, 2020-12-30:
Although there is a difference between core temperatures and skin temperatures, the sensor does drop in accuracy when you distance an object.

If i measure my hand in the front of the sensor, the maximum point in the array will measure around 36.6 ºC. When i put my hand about 4ft away the sensor's max point is just 33.3 ºC.

I did a bunch of readings and for 4 ft I adjust the measurement by {(T - 38 + 4.2)*1.3 + 38} for the range of 35 - 40 ºC (with an error of about +- .3 ºC)

As slavysis said, this sensor doesn't seem to be good enough for many medical grade applications.
I still have a lot to try tho.