melexis/mlx90640-library #68: Incorrect (very hot) temperature occures some frames

moggiozzi, 2020-06-03:
Hi, melexis.
I use mlx90640 (MLX90640ESF-BAA-000-TU) with ESP32 controller. Test code available on https://github.com/moggiozzi/mlxtest.
Test environment is objects in room with 20-36 degrees of Celsius.
1) When set B2 bit of Control register (Enable data hold) I got low temperature on some pixels. See 
[err_log1.txt](https://github.com/melexis/mlx90640-library/files/4720906/err_log1.txt)
Is this normal behavior?
2) When I use other chip peripheral I got very hot temperature 300-1000 degrees of Celcius. See
[err_log2_uart.txt](https://github.com/melexis/mlx90640-library/files/4720907/err_log2_uart.txt)
I try
a) Set i2c frequency 100/400 KHz
b) Reread data when anomal temperature detected (with Enable data hold) - the data remains the same.
The matrix is sensitive to i2c delays? How can I fix this error?

slavysis, 2020-06-04:
1. You should avoid using the device with data hold bit enabled. If you really need to use it, you should make sure that the sampling + data processing time is fast enough so that you do not leave the sensor idle for more than few ms.

2. Can you please check your communication. It looks like the endianness of the wrong values is not correct e.g. you have 0x5cff when it should be 0xff5c.

Best regards

mythbuster5, 2020-09-30:
So, can you send the error log.txt again? Thx.~

moggiozzi, 2020-09-30:
> So, can you send the error log.txt again? Thx.~
Log output files attached to my first message.

mythbuster5, 2020-09-30:
Still some problems, the err_log2_uart.txt is unavailable, would you please to send it again? The second, as your description, i2c went wrong when you communicate over uart and i2c at the same time. So, have you tried use i2c independently? Because I want to make it sure that there is no crush with different peripherals(i2c and uart), thanks

moggiozzi, 2020-09-30:
> Still some problems, the err_log2_uart.txt is unavailable, would you please to send it again?

https://drive.google.com/drive/folders/14WCyNtXlwZHpS8y67I3vSPC521D89Myv?usp=sharing

> The second, as your description, i2c went wrong when you communicate over uart and i2c at the same time. So, have you tried use i2c independently? Because I want to make it sure that there is no crush with different peripherals(i2c and uart), thanks

When I use i2c independently (without uart and other peripheral) problem (byte swap) not occur.

moggiozzi, 2020-09-30:
I think that problem in esp32 i2c driver.

mythbuster5, 2020-10-12:
How to launch your project without bt, idf.py -p COMX is unavailable...

moggiozzi, 2020-10-12:
> How to launch your project without bt, idf.py -p COMX is unavailable...
Project can be run only on ESP32 board.

mythbuster5, 2020-10-13:
[image]

is this normal in your project?