melexis/mlx90640-library #69: Ta value negative

sixwac, 2020-06-03:
Hello,
I extract my EEPROM but i have a bad TA value.
Is negative and the data after the CalculateTo appear very bad:

GETFRAME AFTER CALCULATETO:
0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;232.0;nan;nan;nan;232.1;nan;219.8;238.8;129.9;nan;nan;nan;nan;nan;nan;nan;nan;nan;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;nan;204.5;nan;nan;129.4;245.5;nan;104.3;240.8;217.4;230.9;254.5;179.7;197.3;nan;nan;nan;nan;nan;nan;nan;nan;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;nan;nan;nan;nan;nan;nan;nan;130.8;nan;nan;nan;220.8;nan;nan;232.6;245.5;-127.8;213.9;260.9;118.6;260.0;234.6;248.1;227.3;nan;nan;220.6;nan;106.6;nan;nan;nan;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;nan;nan;nan;nan;nan;nan;nan;178.2;nan;197.7;nan;257.4;nan;nan;257.2;238.1;nan;177.1;220.3;169.0;202.1;99.3;219.3;nan;nan;nan;219.3;nan;nan;69.5;nan;nan;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;0.0;nan;nan;nan;nan;89.5;258.1;252.2;244.3;nan;nan;nan;218.6;nan;nan;249.4;236.7;nan;191.9;253.7;166.6;228.5;160.3;238.1;225.0;150.

Please help me to understand the problem.
Tks

[eeprom.txt](https://github.com/melexis/mlx90640-library/files/4725470/eeprom.txt)

slavysis, 2020-06-04:
Did you try to modify values in the EEPROM? It seems like you changed the EEPROM value at address 240C to 0xffff. This is the configuration value and should be 0x1901. As a start you could try to set that value right. If you did not change other values in the EEPROM, you should be good to go.

Best regards

sixwac, 2020-06-04:
> Hi,
> I can add to begin Adafruit function the command:
>    MLX90640_I2CWrite(i2c_addr, 0x240C, 0x1901);    //EMILIANO
>
> I attach a picture:
> [image: image.png]
>
>
> In this manner I should write the correct "Cont reg 1" value.
> Is correct?
>
>
> Tks
> Emiliano

hi,
i try to make this change on eeprom but i obtain more 0x1901 in the EEPROM:
[image: image.png]

Why i obtain this?

TKS

0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901,
0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901,
0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901,
0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901,
0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901, 0x1901,
0x1901, 0xff11, 0xccee, 0x188a, 0x2fac, 0x2d55, 0xa384, 0x7666, 0x1d5,
0x413c, 0x3c3b, 0x2452, 0x161, 0x17b5, 0x629, 0xf600, 0x9797, 0x9797,
0x2afa, 0x3bff, 0x790, 0x1bde, 0x3ae, 0x7ce, 0x410, 0x239e, 0x3ee, 0xb6e,
0xfc20, 0x1fbe, 0xf47e, 0x3de, 0x780, 0x18ae, 0xfc5e, 0x40, 0xfc50, 0x18be,
0xf43e, 0xfca0, 0xfcd0, 0x188e, 0xf03e, 0xfc10, 0xfbce, 0x143e, 0xf3de,
0xf8ae, 0xfbf0, 0x870, 0xebde, 0x840, 0x79e, 0xfbce, 0x1790, 0x790, 0xe,
0x39e, 0xffff, 0xb62, 0xffff, 0xfbbe, 0x1060, 0x3e0, 0x77e, 0xf8ae, 0x1840,
0x32, 0xfc10, 0xf8ae, 0x1020, 0x92, 0xfcbe, 0xfc7e, 0xc30, 0x2, 0xffce,
0xfc2e, 0xffff, 0x490, 0x3ee, 0xf05e, 0xbb0, 0x3bff, 0xb70, 0x1bce, 0xfbfe,
0xbc0, 0x410, 0x23be, 0x3fe, 0x7d0, 0x10, 0x1f9e, 0xf86e, 0x7c0, 0x7c0,
0x1c6e, 0xf8be, 0xfc40, 0xfc60, 0x189e, 0xf09e, 0xfcb0, 0xde, 0x1880,
0xf07e, 0x3f0, 0x3b0, 0x1450, 0xf3fe, 0x480, 0xfc20, 0x1050, 0xeffe,
0x1ba2, 0x360, 0xfbae, 0x13d0, 0xbb2, 0x3f0, 0x39e, 0x17e0, 0x7c2, 0xfc10,
0xff8e, 0x1060, 0xba2, 0x790, 0xfc3e, 0x1490, 0x12, 0xfc20, 0xfc8e, 0xc60,
0x72, 0xbe, 0xfc5e, 0xc60, 0xbd4, 0x3a0, 0xfc3e, 0xfde, 0x862, 0x1e,
0xf43e, 0xbe0, 0x1c00, 0xbb2, 0xffff, 0x3ee, 0x1ba0, 0x870, 0xffff, 0x6e,
0x1b70, 0x830, 0x273e, 0xfc9e, 0x1740, 0xf50, 0x1c8e, 0xae, 0x820, 0x70,
0x18fe, 0xf86e, 0xd0, 0x4a0, 0xffff, 0xf49e, 0xbf2, 0x7c0, 0x14ae, 0xf430,
0x870, 0x420, 0x1430, 0xf05e, 0x17e2, 0x380, 0xff5e, 0x13d0, 0x1394,
0xfc40, 0xf80e, 0x1440, 0x1352, 0x3f0, 0x70e, 0x1050, 0x1322, 0xb30,
0xfc5e, 0x1470, 0xbf2, 0xfc30, 0xf8be, 0x1010, 0xfcb2, 0x60, 0xfc90, 0xc60,
0xbc4, 0x38e, 0xf87e, 0xc00, 0x852, 0x2, 0xf800, 0x420, 0x1000, 0xbc0,
0x23b0, 0x3e0, 0xbe0, 0xc10, 0x2040, 0x60, 0xfe0, 0x452, 0x1fe0, 0xf8ce,
0x7e0, 0xbe0, 0x18ce, 0xfc8e, 0x400, 0x30, 0x1c60, 0xf84e, 0x442, 0x830,
0x1890, 0xf81e, 0x402, 0x10, 0x14a0, 0xf43e, 0x1010, 0xffff, 0x1470,
0xec6e, 0xbd2, 0xfb80, 0xf78e, 0xffff, 0x3c2, 0x3d0, 0xf41e, 0x1030, 0x3b2,
0xf820, 0xf7be, 0x890, 0xffa2, 0xffb0, 0xf08e, 0x1040, 0xffc2, 0xffff,
0xf82e, 0x800, 0x12, 0x3f0, 0xf43e, 0xfd0, 0x3c4, 0xffc0, 0xf45e, 0xbf0,
0x13d4, 0xfc30, 0xf02e, 0x420, 0x1020, 0xfb2, 0x1c0e, 0xfc50, 0x460, 0x502,
0x18e0, 0xbe, 0x800, 0x812, 0x2750, 0xfc60, 0x7b0, 0x420, 0x18a0, 0xfc7e,
0xbf2, 0x30, 0x18a0, 0xf430, 0x72, 0x840, 0x1880, 0xf48e, 0x2, 0x3f0,
0x1460, 0xfbd0, 0xc52, 0xc0, 0x1450, 0xf06e, 0x7e2, 0xf770, 0xefce, 0x420,
0xf822, 0xf0c0, 0xe89e, 0x470, 0xffb2, 0xfbc0, 0xf6fe, 0x820, 0xfb82,
0xfbc0, 0xec5e, 0x830, 0xffa4, 0xf7f0, 0xf06e, 0x7f0, 0xf822, 0xfff0,
0xf04e, 0x430, 0xffff, 0xfba0, 0xf01e, 0xb90, 0x804, 0xf480, 0xf000, 0x30,
0xffff, 0xf72, 0x1fc0, 0xfc40, 0xc10, 0x8a0, 0x2010, 0x70, 0xfd2, 0xbe2,
0x1f30, 0xfc40, 0xffff, 0x7c0, 0x142e, 0xfc60, 0xb92, 0x7c0, 0x1ff0,
0xf800, 0x412, 0x422, 0x1440, 0xfba0, 0xffc2, 0xb10, 0xc10, 0xf400, 0x90,
0x7f0, 0xc20, 0xf020, 0x442, 0xf3b0, 0xebfe, 0xfc70, 0xf852, 0xf0d0,
0xec3e, 0x4a0, 0xfc02, 0xf000, 0xf35e, 0x70, 0x372, 0xf7e0, 0xe85e, 0x490,
0xffb2, 0xf7e0, 0xf02e, 0x30, 0xf842, 0xf450, 0xe87e, 0x7f0, 0xf404,
0xf7b0, 0xec30, 0xfc30, 0xf8b2, 0xf810, 0xe84e, 0xfc50, 0x3bff, 0xf92,
0x1030, 0xf8a0, 0xf870, 0x492, 0x146e, 0xf8c0, 0xf850, 0x52, 0x17a0,
0xf0de, 0x710, 0x7d2, 0x1440, 0xf4b0, 0xffc2, 0xfc20, 0x1460, 0xffff,
0xf842, 0x12, 0x1020, 0xec60, 0xfbd2, 0xffc2, 0x880, 0xf000, 0xf8d0,
0xffff, 0x852, 0xe89e, 0x1042, 0x7c0, 0xf44e, 0x10c0, 0xa2, 0x8a0, 0xfc9e,
0xffff, 0x470, 0x470, 0xffde, 0x14f0, 0x1342, 0xbf0, 0x6e, 0x18d0, 0xff2,
0x840, 0x8e, 0x1830, 0xc72, 0xc30, 0x6e, 0x1490, 0xfe4, 0xfe0, 0xfcae,
0x1c30, 0x1102, 0x1060, 0x70, 0x14c0, 0xf820, 0xb82, 0xbe0, 0xf050, 0xfbe0,
0x452, 0x870, 0xf0c0, 0xec60, 0xf870, 0xfb0, 0xf7e0, 0xff00, 0x372, 0x850,
0xf010, 0xf782, 0xfbf2, 0x850, 0xe47e, 0xf042, 0xf830, 0xbe0, 0xefc0,
0xf742, 0xfba0, 0x50, 0xebf0, 0xf460, 0x3e0, 0xfc80, 0xe820, 0xfc42,
0xfba0, 0xe80e, 0x460, 0x3f2, 0xfc60, 0xec9e, 0xcd0, 0xf462, 0xf890,
0xf3be, 0x1000, 0x722, 0x790, 0xffff, 0x1060, 0x7b2, 0xfc00, 0xf46e, 0xc70,
0x52, 0x440, 0xfbfe, 0x1000, 0xb62, 0x3b0, 0xffff, 0x17f0, 0xc72, 0xc00,
0xf480, 0x1030, 0xf460, 0x3a2, 0xffff, 0xec70, 0xf7e2, 0xf862, 0x40,
0xe8c0, 0xf7b2, 0xfbb0, 0x310, 0xe880, 0xf750, 0xff62, 0xfc30, 0xf3c0,
[... 1 lines cut ...]
0xf762, 0xf45e, 0xdc90, 0xec30, 0xf832, 0xf430, 0xe7f0, 0xec62, 0xf3b2,
0xe78e, 0xf470, 0xf7e2, 0xec60, 0xe03e, 0xfcc0, 0xfbc2, 0xf7d0, 0xeb2e,
0xfc80, 0xff62, 0xfb80, 0xe45e, 0xbe0, 0x744, 0xf400, 0xec5e, 0x410,
0xf862, 0x2, 0xec7e, 0x850, 0xf062, 0xff70, 0xe86e, 0xfca0, 0x442, 0x30,
0xec4e, 0xbf0, 0xfc90, 0x1402, 0x1020, 0xa0, 0xf890, 0xd42, 0x10f0, 0x130,
0xfc60, 0xffff, 0x17f0, 0xfd10, 0x3e0, 0x13c2, 0x1490, 0xfcf0, 0x7d2,
0xc20, 0x14c0, 0xfc70, 0x20, 0xce0, 0x1850, 0xffff, 0x22, 0xfd0, 0x10a0,
0x3e0, 0x470, 0xffff, 0xc70, 0xfc50, 0xf492, 0xf800, 0xe82e, 0x8a0, 0xf482,
0xf920, 0xe8fe, 0xd20, 0xfc72, 0x80, 0xf00e, 0x1110, 0x3f2, 0xfd0, 0xf4be,
0x1500, 0xbf2, 0x820, 0xf8ce, 0x1880, 0x1012, 0x1060, 0x5e, 0x1ca0, 0x1022,
0x17d0, 0xfc9e, 0x2bd0, 0x1862, 0x1480, 0x7e, 0x2050, 0xf7ff, 0xbb2,
0xf82e, 0xf460, 0xe820, 0x822, 0x50, 0xf49e, 0xe840, 0x22, 0xb70, 0xf470,
[... cut ...]
0x13a0, 0x7e0, 0xbd0, 0xf79e, 0x17b0, 0xd8ce, 0xffff, 0xec40, 0xe4e0,
0xe430, 0xfc52, 0xf42e, 0xe8a0, 0xe410, 0xf842, 0xf3ce, 0xe8ae, 0xe01e,
0xffc2, 0xed3e, 0xecce, 0xe080, 0xf840, 0xfcce, 0xec30, 0xec40, 0x60,
0x420, 0xf020, 0xdc80, 0x3a0, 0xec8e, 0xefce, 0xe46e, 0x70, 0xe86e, 0xec7e,
0xe8b0, 0xfbf0, 0xd43e, 0xfcc0, 0xf420, 0x30, 0xe40e, 0xc90, 0xfff0,
0xfc30, 0xefce, 0x1090, 0xfc10, 0xfbe, 0xe93e, 0x14d0, 0xfc90, 0xc30,
0xf8ce, 0x2020, 0x1042, 0x1850, 0x41e, 0x2810, 0x460, 0x1b8e, 0xf87e,
0x23b0, 0x1450, 0x184e, 0x5e, 0xffff,