melexis/mlx90640-library #70: mbed.h no such file in directory

Googlesmagic, 2020-06-10:
I have been getting this error when trying to compile the code for an arduino duo. I am fairly new to this and I am not sure what I am doing wrong. I am not sure if I need to download the mbed,h file online somewhere as I am having trouble finding it.

Cheers

slavysis, 2020-06-11:
Hi, the MLX90640_I2C_Driver.cpp should be modified for any specific MCU. The included file is an example for MBED. For more details, please refer to MLX90640 driver.pdf

Best regards

kkaryono, 2020-07-14:
Hi, I am fairly new also, but I've managed to run the module. You can use the code which is already available like Adafruit code https://github.com/adafruit/Adafruit_MLX90640 or Spark Fun https://github.com/sparkfun/SparkFun_MLX90640_Arduino_Example.
Kind Regards,

closed 2020-11-03 by slavysis