melexis/mlx90640-library #72: EEPROM - Dump eeMLX90640 read problem..

jinoid, 2020-08-05:
Hi, I have a question.
I can read form eeMLX90640[0] to eeMLX90640[831], but the result is strange from eeMLX90640[64] to eeMLX90640[831].
result value is always 0.
Is this ok? or wrong?

status = MLX90640_DumpEE(eeMLX90640);

int MLX90640_DumpEE(Uint16 *eeData)
{
     return MLX90640_I2CRead(0x33, 0x2400, 832, eeData);

}

[eeMLX90640(txt).txt](https://github.com/melexis/mlx90640-library/files/5026855/eeMLX90640.txt.txt)

slavysis, 2020-08-05:
Hi, 
This is not normal. Is it possible that you have some kind of a buffer when you read I2C? Can you check with an oscilloscope or
a logical analyzer that the I2C data for the addresses after 64 is indeed 0000?

Best regards

jinoid, 2020-08-05:
Thanks for your quick answer!
I check it.
result is also 0.

I think read and write function is normal. because data that from 0 to 63 is normal.

[image]

jinoid, 2020-08-05:
The MLX90640 sensor I have is an old model.
The model was purchased in February 2017. Do you know if the contents stored in the eeprom may be different from the models that these days?

The value stored in the address 0x2404 of the EEPROM is similar to the slave address, so there seems to be a problem.
If you look at the content of July 7, 2017 among the revisions in the datasheet, the storage address of the slave address has changed.

slavysis, 2020-08-05:
The sensor being old should not be an issue. Can you please give some zooms for the scope traces so that I can be able to clearly see the signal transitions? 
Also, if I read the traces correctly, you are doing full command transmission for each slave address which is very inefficient. 
I would recommend reading the full EEPROM data with minimum overhead - send the SA and the starting EEPROM address just once and
then take advantage of the MLX90640 internal auto increment to get all the data (terminate with NAK).
Can you also share the EEPROM contents you are getting?

Best regards

jinoid, 2020-08-05:
The scope trace I uploaded, that is made by the code, temporarily for testing.
//========================================================
// Test block
//--------------------------------------------------------------------------------------------
		int n;
		if(check_para[1] ==999){
		    n=64;
            for(i= 0x2440; i<0x273f; i++){

                status = _90640_I2CRead(i, 1, &eeMLX90640[n]);
                DELAY_US(5000);
                n++;
            }
		    check_para[1] = 0;
		}

Below is my codes, 
Master MCU is TMS320F280049,

int MLX90640_DumpEE(Uint16 *eeData)
{
     return _90640_I2CRead(0x2400, 832, eeData);
}

int _90640_I2CRead(Uint16 startAddress, Uint16 nMemAddressRead, Uint16 *data)
{
    int Read_Address_Write[2] = {0,0};

    int cnt = 0;
    int i= 0;
    char i2cData[1664] = {0};

    Uint16 *p;

    p = data;

    //-----------Read Command write Resister set -----------
        I2caRegs.I2CMDR.bit.IRS = 1;                // reset I2C
        while(I2caRegs.I2CSTR.bit.BB == 1);
        I2caRegs.I2CSTR.bit.SCD = 1;                // Clear the SCD bit (stop condition bit)
        while(I2caRegs.I2CMDR.bit.STP == 1);

        Read_Address_Write[0] = startAddress >> 8;
        Read_Address_Write[1] = startAddress & 0x00ff;

        I2caRegs.I2CSAR.all = slaveAddr;                     // I2C slave address

        while(I2caRegs.I2CSTR.bit.BB == 1);      // busy loop

        _90640_Read_Address_Write_Register_Setting(2,0);

        _90640_Read_Address_Write(Read_Address_Write, 2); //Address, count

        while(!I2caRegs.I2CSTR.bit.ARDY);

        _90640_ReadBytes((nMemAddressRead << 1), i2cData);  //(nMemAddressRead*2, *data)

            for(cnt=0; cnt < nMemAddressRead ; cnt++)
            {
                i = cnt << 1;
                *p++ = (int)i2cData[i]*256 + (int)i2cData[i+1];
            }

        return 0;
}
//========================================================

The scope's trace reading 832 at once is difficult to enlarge.
The shape of the signal looks crushed.
I will try again with other equipment tomorrow.

[image]
[image]
[image]
[image]

jinoid, 2020-08-06:
The framedata is read normally from the sensor's RAM.
Even in EEPROM, 64 data from 0 to 63 are read normally, but 768 from 64 to 832 are read as 0.
I think there is no data in the sensor's eeprom.

Below is the scope waveform of EEPROM.
[image]

jinoid, 2020-08-07:
This is the memory read after writing 0x1234 to the address 0x2440 to 0x2445 of EEPROM and executing the DumpEE function.

It is written and read normally.
[eeMLX90640_1(txt).txt](https://github.com/melexis/mlx90640-library/files/5038728/eeMLX90640_1.txt.txt)

slavysis, 2020-08-07:
Can you please check if there is a green PCB with some capacitors on the bottom of the sensor (pin side)?

Best regards

jinoid, 2020-08-10:
I thought that there could be a problem with the EEPROM because there was a lot of code that ran this and that while testing, so when I tested it with a new sensor, different results came out.
All the data that can be read from EEPROM are as follows.

[eeMLX90640_20200810(txt).txt](https://github.com/melexis/mlx90640-library/files/5048734/eeMLX90640_20200810.txt.txt)

Please check the picture.
[image]

slavysis, 2020-08-10:
These are sort of a prototype devices and the coefficients are not stored in the EEPROM but in a file - you should have those files if you got the sensors from Melexis. Note that those devices are not supported as their performance is inferior to the standard/current ones. Also note that they do not correspond to the current datasheet. So my advice would be to not evaluate/work with them.

Best regards

closed 2020-11-03 by slavysis