melexis/mlx90640-library #76: Pixel flooding-Most of the pixels (in Image mode) appear to have high value with no object in FOV

Yogeshkb07, 2020-11-18:
We are working on a prototype with Melexis 90640 thermal imaging sensor.
We are operating sensor on 4Hz in Pixel Image mode and not in Temperature Image mode in Chess frame capture mode.

Many a times the sensor generates a frame in which the pixel value appear to have high values even when there is no object in FoV.
We are calling it as "Frame Flooding". 
It was observed that, on Power Reset the flooding vanishes.

Can anyone help me to explain this behavior of sensor and what measures can be taken to avoid it.

slavysis, 2020-11-23:
Hi,

Is it just a single frame from time to time (good frame -> 'flooded' frame -> good frame) or once you get the 'flooded' frame is stays on?
Are you doing some post-processing or you get the 'flooded' frame right off the GetImage function?
Can you share the raw frame data and the GetImage resulting frame for consecutive good<->'flooded' frames?
The "power reset' that you say fixes the issue - is it a power reset on the MLX90640 only, or is it a power reset of the whole system (including your MCU)?

Best regards

Yogeshkb07, 2020-11-23:
Hi Slavysis,

1. We are getting continues flooded frames
2. We are processing the frames for object detection
3.  We are getting flooded frames from GetImage function
4. I am attaching a screen shot of the flooded frame. This frame is after we run a single step that highlights the pixels which is higher in number
[image]

5. We are performing power reset for entire system including MCU and sensor

slavysis, 2020-11-23:
Hi,

Just to make sure I am getting it right:
1. You power on the system and it is working fine.
2. For some time the frames are OK.
3. After some time you get 'floded frame'
4. You reset the whole system and you are back at 1.

Is that correct?

At what frequency are you reading the I2C? How much time does it take to read out the data and process it?
Regarding the flooded frame you showed: there is no object in front of the sensor, right? Was there an object in front of the sensor before? When you say 'a single step' what do you mean?
Excuse me for asking so many questions, but I am trying to get better understanding of what might be happening.

Best regrads

Yogeshkb07, 2020-11-23:
Hi,
All the 4 assumptions are right.

1. Frequency of I2C read- 8Hz
2. Time taken to read data -40ms Time to process a complete frame- 105 ms
3. The sensor was pointing the floor (in an AC room) during test
4. By Single step I mean after we receive two halves of frame, we combine them and then highlight cells with higher pixel values

Its absolutely fine about the questions. Let me know if there is anything more you want me to share

slavysis, 2020-11-23:
Hi

> Frequency of I2C read- 8Hz
I guess this means you are trying to read out data every 125ms (8Hz). Is the device set to 8Hz?
What is the I2C clock frequency?
What is the return value of the MLX90640_GetFrameData function when you get the 'flooded frame'?
Have you tried setting the device at 0.5Hz to check if you would have the same problem?

Yogeshkb07, 2020-12-04:
Hi,
Sorry for delayed response!
1. Yes, the device is set to 8Hz
2. I2C clock freq is 400KHz
3. We dont really capture the return values but the frame
4. We did not try for 0.5Hz but then it will be two slow for our purpose

slavysis, 2020-12-11:
Hi,

Then the time needed to capture and process the data is a bit more than the refresh rate of the device. Could you please test if you will run into the same issue if the MLX90640 is set to 0.5 or 1Hz (just for the sake of the test)? Could you also check if the return value of the GetFrame data function is 0 or some error code.

Best regards