melexis/mlx90640-library #79: Can't detect i2c with EVB90640

Kirei28, 2021-02-23:
While connecting my EVB90640 directly to the i2c pins of the raspberry pi, the raspberry pi can't detect it. However, connecting it directly to the PC using USB interface and the provided software by the melexis, the sensor works and it's indicated address is i2c 0x33. The raspberry pi i2c pins are working since I've tried it using a different module which is ds3231 and it works. Can somene help me

mlx-kva, 2021-02-23:
Are you still using mlx90641?

mlx-kva, 2021-02-23:
Referring to: https://github.com/melexis-fir/mlx9064x-driver-py/issues/6

Kirei28, 2021-02-23:
Oh hello, I've tried both the mlx90641 and the mlx90640

mlx-kva, 2021-02-23:
This library is intended to embed on a microcontroller.
It expects you to implement the low lever I2C routines. For this see the pdf in the root folder of this repo.
The python driver for MLX90641 is progressing well.
I do not have the EVB working there, but since you have a raspberry pi, you might give this a try:
https://github.com/melexis-fir/mlx90641-driver-py

mlx-kva, 2021-02-23:
I will close this ticket as EVB and this lib are not supposed to be compatible.

mlx-kva, 2021-02-23:
@slavysis, can you close?

closed 2021-02-24 by Kirei28