melexis/mlx90640-library #81: Strange temperature distribution when creating a histogram

spamekkel, 2021-03-11:
Hi,

I'm using the mlx90640 with the current mlx90640-library and have some serious issues with the reported temperatures.
When plotting a histogram of the calculated temperatures I observed something strange: not all temperatures are represented.

To analyze this, instead of floats I changed the MLX90640_CalculateTo() function to return the temperatures in Celsius * 100 and store them in an int16_t array. So, 12.34 Celsius becomes 1234. These temperatures are indexes in a huge array of 4000 entries to create a histogram for 0..40 Celsius using 1000+ frames.

Below are example histograms for full frame and two sub frames (not from the same footage).

[image]

[image]

From the histograms it can be observed that
- many consecutive bins that are zero
- the values on the right and left side of such bins are not very close. E.g. at 24 Celsius the values to the right are over 300, while at the left the values are close to zero.

Did anyone run into this?

spamekkel, 2021-03-18:
Hi,

I opened this issue a week ago, then closed it as I wanted to be sure.

While examining the pixel values I noticed there an unexpected distribution of the temperaures. With the latest mlx90640 driver I created a histogram of pixel #400. Below is the outcome after ~40.000 frames @4Hz. I'd expect a sort of normal distribution of the pixel values.

[image]

Did anyone investigate this?

slavysis, 2021-03-22:
Did you check if your object temperature changed during that long measurement? Can you share the linear plot of the data?

Best regards

spamekkel, 2021-03-26:
Hi,

Thanks for your reply, your help is mucj appreciated.

There probably was a temperature change during that measurement of 3 hours. The histogram was recorded on the embedded system without storing the while data set. I will create another histogram of 30 minutes and record the data from the whole image. I'm currently too busy to pick it up now, but will do this next week.

Kind regards,
Spam.

spamekkel, 2021-03-29:
I should have said why the distribution seems strange. It is not the second bump to the right.

Looking at the device's targeted temperature range (-40 to +300'C) and a resolution of 18-bits I expected much smaller discrete temperature steps. E.g. for this range 340'C / 2^18 = 0.0013'C. But the frame data is only 16-bits, so perhaps steps of 340'C / 2^15 = 0.0052'C are more reasonable. But the temperature steps in the recorded histogram (2650 equals 26.50'C) are approx. 0.08'C, much higher then I'd expected.
I admit I haven't gone through the device manual's complex explanation of the temperature calculation. But I'm aware of the noise figures, I'm not expecting a 16-bit accuracy of the final To.
The datasheet does not explain the ADC resolution vs the end result. Perhaps this and all the compensations for Vdd, Ta etc. lead to the resultion of 0.08'C. ?

slavysis, 2021-03-31:
Well, the thing is that the step is not solely defined by the resolution. The sensitivity of the sensor also plays major role as this parameter would define more or less what change of the calculated temperature would be induced by changing the raw data. So for example, if your sensor has such a sensitivity that  when the raw data changes by 1LSB, the calculated temperature would change by 0.08C then whatever the resolution is, you just would not be able to see smaller step.

Best regards