melexis/mlx90640-library #83: Corrupted EEPROM values - every second pixel is FFFF

n3o59hf, 2021-04-13:
I am getting weird looking data from `MLX90640_DumpEE`, it seems like each second column of pixels is `0xFFFF`.
Data is consistent between multiple reads. I believe my issue is similar to https://github.com/melexis/mlx90640-library/issues/61
EEPROM probably became corrupted while implementing i2c reading (SWI2C) as sensor worked normally before (I have not saved EEPROM) and during development it stopped responding on address `0x33`. I was able to locate sensor on address `0x7f` and write correct address into EEPROM as it was set to `0xFF` if I remember correctly.

What can be done to restore values to factory values or to sane "defaults" (accepting precision loss)?

Data:

```
        00   01   02   03   04   05   06   07   08   09   0A   0B   0C   0D   0E   0F   10   11   12   13   14   15   16   17   18   19   1A   1B   1C   1D   1E   1F 
2400: 00B6 099F 0000 2061 0004 0320 03E0 0917 4B21 0189 048D 0000 FFFF 0000 FFFF BE33 67FF FFBA FFFF 0202 FFFF E1F1 FFFF A0C0 FFFF FF11 FFFF DFF1 FFFF D1F4 FFFF B3F4 
2420: 7FFF 2A48 FFFF 2100 FFFF 2233 FFFF BCEE FFFF 00FF FFFF 2222 FFFF 0112 FFFF CCEE 77FF FFFF FFFF A48C FFFF 09CD FFFF 4E4D FFFF 0463 FFFF 0631 FFFF F300 FFFF FFFF 
2440: FFFF 0432 FFFF 0812 FFFF 0870 FFFF 001E FFFF 041E FFFF FB8E FFFF 0BF0 FFFF F81C 67FF FFFF FFFF FB9E FFFF 0430 FFFF FFFF FFFF FFFF FFFF F0CE FFFF FFF0 FFFF F4C0 
2460: 67FF F420 FFFF 13D4 FFFF F83E FFFF 0BE0 FFFF F7EE FFFF 076E FFFF FBDE FFFF 07FE FFFF F02E FFFF 076E FFFF F80E FFFF 0820 FFFF F790 FFFF 0080 FFFF F7B0 FFFF 0490 
2480: 6FFF 0034 FFFF FC1E FFFF 0462 FFFF F44E FFFF 0040 FFFF FFFF FFFF 0050 FFFF F45E FFFF FC70 FFFF F7DE FFFF 0040 FFFF F47E FFFF FFB0 FFFF F070 FFFF FBC0 FFFF F0A0 
24A0: 77FF 03D0 FFFF 17D0 FFFF 0430 FFFF 1000 FFFF 001E FFFF 0FAE FFFF 043E FFFF 1020 67FF 003E FFFF 13BE FFFF 041E FFFF 1440 FFFF 0770 FFFF 1030 FFFF FFFF FFFF 1050 
24C0: 6FFF 07A4 FFFF FFC0 FFFF 0442 FFFF F820 FFFF FFFF FFFF F3EE FFFF 0070 FFFF F46E 67FF FC70 FFFF FBBE FFFF FC70 FFFF F46E FFFF FBB0 FFFF EC50 FFFF FB60 FFFF E460 
24E0: 77FF 03D2 FFFF 13F0 FFFF 007E FFFF 0860 FFFF FC8E FFFF 042E FFFF 00BE FFFF FFFF 67FF FCCE FFFF 13F0 FFFF FCBE FFFF 10A0 FFFF F800 FFFF 0880 FFFF FF90 FFFF 0480 
2500: 77FF 07A6 FFFF FFE0 FFFF 0072 FFFF FFFF FFFF FCA0 FFFF F01E FFFF FCC0 FFFF F0CE FFFF FCAE FFFF F03E FFFF FC80 FFFF FFFF FFFF FFB0 FFFF E87E FFFF FB60 FFFF EC30 
2520: 77FF FF24 FFFF 0B72 FFFF F82E FFFF 0020 FFFF F45E FFFF 07D0 FFFF F87E FFFF 087E FFFF FC60 FFFF 0BE0 FFFF 0030 FFFF 0C40 FFFF FF60 FFFF FFFF FFFF FB10 FFFF 07D0 
2540: 67FF FBB2 FFFF F000 FFFF 0042 FFFF F06E FFFF FC90 FFFF EC3E FFFF F8F0 FFFF F0DE F7FF F500 FFFF EC4E FFFF F8D0 FFFF ECBE FFFF FBB0 FFFF E860 FFFF F762 FFFF EFF0 
2560: 77FF FB40 FFFF 0FA0 FFFF 07D0 FFFF 0C00 FFFF 0420 FFFF 0FDE FFFF FFFF FFFF 1070 67FF 00B0 FFFF 13F0 FFFF 047E FFFF 1450 FFFF 0750 FFFF 13F0 FFFF 06E0 FFFF 1390 
2580: 6FFF 0AE6 FFFF F78E FFFF 03F2 FFFF EC2E FFFF FFFF FFFF F3C0 FFFF F890 FFFF F09E 67FF F4A0 FFFF EFFE FFFF F860 FFFF EC80 FFFF F790 FFFF E400 FFFF F702 FFFF FFFF 
25A0: 77FF 06E4 FFFF 0B92 FFFF 0002 FFFF 0430 FFFF FC50 FFFF 0BD0 FFFF 00B0 FFFF 08AE 6FFF FCBE FFFF 1000 FFFF 007E FFFF 1090 FFFF FF90 FFFF 0810 FFFF FF10 FFFF FFFF 
25C0: 67FF 1306 FFFF 0390 FFFF 0C02 FFFF FC10 FFFF 0450 FFFF FBFE FFFF 0880 FFFF F89E FFFF 00B0 FFFF F81E FFFF 00A2 FFFF FFFF FFFF 0750 FFFF F400 FFFF 0312 FFFF FB90 
25E0: FFFF F304 FFFF FF92 FFFF F400 FFFF FC10 FFFF F040 FFFF 03F0 FFFF FC7E FFFF 0880 F7FF F4B0 FFFF 0800 FFFF F4AE FFFF 0C60 FFFF FF50 FFFF 0800 FFFF FB10 FFFF 0FA0 
2600: 67FF 0F74 FFFF FFFF FFFF 0C22 FFFF FC40 FFFF 0860 FFFF F41E FFFF 04C0 FFFF F8BE 67FF 00D0 FFFF F440 FFFF 00C0 FFFF F8BE FFFF 03B0 FFFF F050 FFFF FF72 FFFF FBE0 
2620: 6FFF FFFF FFFF 0BA2 FFFF 0010 FFFF 0830 FFFF 0040 FFFF 0BF0 FFFF 00A0 FFFF 10A0 6FFF FFFF FFFF 1010 FFFF FFFF FFFF 14A0 FFFF 07A0 FFFF 1040 FFFF 0370 FFFF 17E0 
2640: 67FF 1766 FFFF FFD0 FFFF 0830 FFFF F440 FFFF 0440 FFFF F7E0 FFFF 00D0 FFFF F880 FFFF 00A2 FFFF FBFE FFFF 0480 FFFF F89E FFFF FFFF FFFF EC70 FFFF 0362 FFFF F030 
2660: FFFF 0346 FFFF 07B2 FFFF F810 FFFF 0020 FFFF F820 FFFF FFFF FFFF F890 FFFF FFFF 67FF FC70 FFFF 0FD0 FFFF 0070 FFFF 1070 FFFF FF90 FFFF 0450 FFFF FF50 FFFF 0810 
2680: EFFF FFE2 FFFF FBC0 FFFF 0430 FFFF F010 FFFF 0002 FFFF F780 FFFF 0420 FFFF F42E F7FF F880 FFFF FFFF FFFF 0050 FFFF F450 FFFF 0392 FFFF E870 FFFF FFFF FFFF F040 
26A0: F7FF EC40 FFFF 0420 FFFF FC70 FFFF 0060 FFFF 0040 FFFF 13C0 FFFF 0460 FFFF 1460 67FF 00C0 FFFF 1020 FFFF 08A0 FFFF 1890 FFFF 0BE0 FFFF 10C0 FFFF 0BE0 FFFF 18A2 
26C0: F7FF 0C32 FFFF 0402 FFFF 1052 FFFF FC20 FFFF 13F2 FFFF FF60 FFFF 0C32 FFFF FFFF 67FF 0C32 FFFF 03BE FFFF 0C62 FFFF FC70 FFFF 0FB2 FFFF F8A0 FFFF 0BE2 FFFF FCB0 
26E0: FFFF F412 FFFF 0FE2 FFFF 0010 FFFF 0BE0 FFFF 07B0 FFFF 1720 FFFF 0FE0 FFFF 1BC0 6FFF 0BF0 FFFF 1F60 FFFF 0830 FFFF 2032 FFFF 1380 FFFF 1872 FFFF 0FC0 FFFF 2090 
2700: F7FF 0C34 FFFF FC60 FFFF 0C52 FFFF FFFF FFFF 0402 FFFF FF30 FFFF 0FE2 FFFF 03C0 FFFF 0FD2 FFFF 0360 FFFF 0C42 FFFF FC70 FFFF 0FE2 FFFF FFFF FFFF 0832 FFFF FD00 
2720: EFFF EC12 FFFF F8B2 FFFF F8B0 FFFF 0080 FFFF F840 FFFF 0B70 FFFF 0010 FFFF 1000 FFFF 0020 FFFF 1BA0 FFFF 0480 FFFF 14C0 FFFF 0420 FFFF 0930 FFFF 0090 FFFF 10F0 
```

Sensor marking `0AA1903301`.

sachinbs1995, 2021-09-02:
I am also having the same issue if you would have solved plz let me know.

mdehbozorgi, 2022-03-08:
I don't know if it still helps or not. But I believe the problem is SCL speed.if you increase the scl it might help with the problem.

Hank-lu-88, 2023-11-17:
Hi, I would like to ask how to output those parameters. I use rasberry pi to control mlx90640. I encountered the same problem but I can only output temperature values. And like your question, the entire row of values ​​​​is 0.