melexis/mlx90640-library #87: Device operation in mode Step mode

saplatov, 2021-08-13:
Hello!
In order to quickly receive data from the sensor, I try to work in Step mode. I wrote a program that works according to the algorithm in the figure. I receive incorrect data. What could be the reason for the failure?
[image]

slavysis, 2021-08-23:
Step mode can/should be used in rear case and in a special way. That is why it is no longer supported. Note that step mode does not really help you to 'to quickly receive data from the sensor'. The fastest would still be continuous mode.

Best regards

amitkriit, 2025-04-25:
@slavysis my use-case requires me to operate the sensor in the step mode. I am adhering to the measurement flow chart). The sensor outputs wrong object temperature data if the refresh rate is set above 0.5Hz. Like you said, there is a "special way". Kindly guide me.

Mulukulum, 2025-09-03:
@slavysis could you clarify whether step mode was completely removed from the newer MLX90640 sensors? Is it something not supported by newer hardware or just a driver setting that was removed?