melexis/mlx90640-library #89: mlx90641 drivers work, mlx90640 drivers don't...

luigi-grim, 2021-11-17:
Hello. I'm trying to implement the mlx90640 on a CC2538 texas instrument and a Contiki os. 
I successfully implemented the mlx90641 getting an output like this: 

DumpEE:
b0	899f	0	2031	5	320	3e0	161b	f6e3	6186	ad9	0	901	0	0	be33	
0	7f6	e	0	0	28b	222	281	164	1ad	1ad	1ad	4dd	525	542	554	
534	4da	7da	200	f0	1d	677	79a	164	4	12c	8	540	321	23	7ee	
1c	313	197	420	16	4b9	4b9	4b9	4b9	4b9	190	4b9	1f4	4b9	258	4b9	
7f9	7fd	4	14	b	9	e	d	3	1	7fc	0	7e1	7e8	7dc	7d6	
28	15	20	1a	1e	1f	28	1a	1e	1a	1e	e	9	7fe	7fc	7e9	
7fe	8	12	16	13	d	e	1e	b	d	8	13	7f7	7f7	7ef	7e3	
26	1b	21	21	20	27	2b	23	23	1f	27	19	15	c	8	7f9	
0	8	e	18	14	12	17	1f	d	16	c	15	7fc	7ff	7f5	7f4	
1d	10	1d	16	1d	21	26	1f	20	21	23	15	17	10	10	1	
7eb	7ff	7f9	c	5	5	9	14	a	6	8	15	7fd	0	7f9	2	
5	7fd	9	2	8	c	19	d	12	14	1c	13	14	f	10	2	
7d0	7db	7e1	7f7	7f1	7f0	7f2	4	7f6	7f8	7f9	d	7f5	7f6	7f6	4	
7e2	7db	7e7	7e6	7ec	7f1	7fa	7f5	7ff	2	10	1	4	1	6	6	
7a1	7b4	7bc	7ca	7c8	7c8	7d3	7db	7d5	7dc	7dc	7f4	7e8	7e9	7e8	7f8	
791	790	79d	797	7a3	7ac	7bc	7b1	7bf	7cb	7df	7d9	7e4	7e9	7f5	7ef	
581	619	67c	6d0	74f	78d	78b	7b7	7ae	7b3	76e	747	72c	6b6	641	5c0	
5cd	670	6db	74a	7a7	7e1	7dc	7f9	7f4	7fc	7af	78c	774	728	695	600	
5bf	659	6bb	740	7a6	7cd	7dc	7c4	7eb	7c2	7a5	762	74e	708	685	60d	
5e3	67e	6f7	743	7aa	7cc	7d9	7ed	7fc	7ff	7b5	784	765	730	6a9	617	
5d5	668	6e5	756	7b8	7ca	7d7	7d6	7ff	7c5	7c6	786	776	720	6af	61d	
5ee	685	6df	765	7a7	7db	7e0	7f4	7fe	7fd	7bd	794	768	732	69b	624	
5d0	649	6ea	73f	7af	7d8	7ed	7da	7ee	7ff	7ae	778	765	713	68c	609	
5ba	644	6aa	724	794	7d0	7b9	7e0	7de	7da	799	76c	72f	6df	660	5f1	
5aa	64e	6be	711	794	7d5	7ed	7e5	7ff	7f7	7c0	769	760	6ea	667	5dc	
568	5f9	668	6d6	732	786	7a5	7bb	7a9	7b2	755	73f	6ef	6a5	614	584	
58b	610	67f	6ef	76a	7ad	7c5	7f5	7fc	7e4	7bf	757	730	6af	633	5b5	
524	5b8	60a	69a	6dc	741	73c	782	77d	76d	715	704	6b1	65e	5bc	53c	
7a0	6e0	6bf	700	700	720	71f	6df	77e	71f	6df	79f	6fd	740	1f	61	
7a7	6e3	6c3	721	760	6e2	722	71f	77f	721	6e1	6ff	780	6e1	7a1	2	
7c0	7c1	7a1	7c0	7e0	73f	75e	7df	7be	77e	77d	bf	79d	3f	7df	61	
67	784	784	7a1	7a1	762	722	77f	77f	701	761	73e	79f	7a1	7a1	41	
21	7c2	7a0	0	1f	7ff	7de	1e	7fd	1e	79d	3e	77c	3e	1e	a1	
89	7a5	7e4	741	1	7c2	7e2	77f	7df	781	0	77e	7df	760	761	7e1	
83	c3	41	81	20	9f	3e	3e	7c	7e	79d	7e	7bc	3e	7be	a0	
12a	45	64	41	62	1	61	7be	7de	7a0	7ff	7be	1e	760	7a0	760	
105	144	e2	121	e0	9f	5e	fe	bc	5d	7fc	fd	3c	7fd	7de	7f	
1cc	87	65	c2	101	81	a0	5d	fd	7df	5f	7fd	7fe	79f	75f	7df	
128	185	162	161	140	15e	13d	fc	7b	9c	3c	7d	5c	79d	77d	7e0	
16c	147	143	e0	13e	11f	11d	9b	7c	7c	5d	7fc	75d	6ff	6ff	71f	
7f9	7fb	4	15	b	7	d	d	4	7ff	7fc	0	7e2	7e7	7db	7d6	
28	13	20	1a	1f	1e	28	1b	1f	18	1e	e	8	7fd	7fc	7e9	
7ff	7	12	16	14	c	e	1e	c	b	7	13	7f7	7f5	7ee	7e3	
27	1a	21	21	21	26	2b	23	24	1e	26	19	16	b	7	7f9	
1	7	e	18	15	12	16	20	e	15	d	16	7fd	7fe	7f5	7f4	
1e	f	1d	16	1e	20	26	1f	21	1f	24	16	18	f	f	1	
7ec	7fd	7f8	d	6	4	9	14	a	6	8	14	7fd	7fe	7f8	1	
6	7fc	9	3	8	b	19	c	12	13	1b	13	14	e	f	2	
7d1	7da	7e0	7f7	7f2	7ee	7f1	4	7f7	7f8	7f8	e	7f5	7f5	7f6	4	
7e4	7da	7e7	7e6	7ec	7f1	7fb	7f5	0	0	f	2	5	0	6	7	
7a2	7b3	7bc	7ca	7c9	7c7	7d2	7dc	7d6	7dc	7dd	7f5	7e9	7e8	7e8	7f9	
792	78f	79e	798	7a4	7ab	7bc	7b2	7c0	7cb	7e0	7db	7e6	7e8	7f6	7f0	
status Frame-synch:0
Get frame!
Status get_frame:0
Frame:
fe93	fe8f	fe92	fe9d	fe92	fe89	fe8d	fe8c	fe83	fe87	fe84	fe83	fe68	fe74	fe72	fe7b	
febe	fea6	feac	fea2	fea1	fe9f	feaa	fe98	fe9c	fe9f	fea5	fe90	fe8e	fe86	fe94	fe8f	
fe95	fe95	fe99	fe99	fe91	fe89	fe88	fe9a	fe82	fe92	fe8d	fe92	fe79	fe82	fe87	fe8c	
feed	fee1	fee3	fede	fed9	feda	fed7	fec5	febc	feb8	febd	fea2	fea0	fea1	fea7	fea0	
feec	fefa	ff05	ff11	ff0d	ff06	ff0f	ff14	fef7	fefd	fefb	fef7	fec8	feb9	feb0	feb1	
ff0d	ff06	ff1c	ff16	ff22	ff28	ff2d	ff27	ff03	ff0a	ff26	feff	feee	fee3	feea	fee7	
fec1	fede	fee2	fef8	feed	feeb	fefc	ff0b	fedb	fee7	fef3	fee7	fec0	fec2	fec9	fedc	
feb5	feaf	febe	feb2	feb3	febe	fed4	fec6	feba	feca	fed4	fec6	fec4	fec6	fecd	fec0	
fe83	fe8b	fe90	fea9	fea1	fea6	fea2	feb1	fe9c	fead	fea6	fecf	fec5	fecc	feb8	febb	
fe94	fe9f	fe9f	feb9	feb8	fef3	fec5	fecc	ff01	ff09	feea	ff4e	ff3b	fefa	feba	feb4	
[... 120 lines cut ...]
28 28 -99 185 29 28 28 28 24 22 -16 17 22 22 22 22 103 22 22 22 0 108 113 22 22 33 22 206 23 23 23 23 
28 28 28 0 28 28 28 27 255 116 129 24 22 22 22 22 22 22 251 187 22 22 22 22 22 22 0 207 23 23 23 23 
28 28 28 28 28 28 27 26 22 22 22 22 22 22 22 22 22 22 22 22 18 17 18 17 27 22 23 23 23 22 23 23 
28 28 28 28 28 27 26 25 22 22 0 184 22 22 22 22 22 22 22 22 22 22 0 192 22 22 22 22 23 23 154 242 
28 28 28 28 28 27 25 180 213 209 22 120 208 206 22 121 22 22 22 22 0 207 22 130 0 100 124 22 28 22 23 23 
28 28 0 206 27 27 24 23 22 22 22 22 22 22 0 86 22 22 0 194 22 22 262 198 6 101 141 23 63 23 23 23 
28 28 160 192 27 39 23 22 184 179 22 122 8 18 17 17 47 43 45 49 22 22 0 189 23 22 0 205 23 23 0 235 
28 28 0 204 26 26 0 193 22 22 0 188 22 22 0 189 22 22 0 190 22 23 0 198 22 22 0 211 23 23 0 244 
28 27 0 190 26 25 0 184 22 22 0 179 22 22 0 181 22 22 0 184 23 22 0 192 22 23 0 207 23 23 0 239 
28 28 0 208 26 24 0 199 22 22 0 195 22 22 0 194 22 22 0 195 22 22 0 202 22 23 0 219 23 23 0 250 
28 28 0 197 24 23 0 188 22 22 0 186 22 22 0 186 22 22 0 188 22 22 0 196 22 22 0 213 23 23 0 246 
28 27 0 217 24 23 0 207 22 22 0 201 22 22 0 200 22 22 0 202 22 23 0 209 22 23 0 225 23 23 0 256 
28 28 0 204 23 22 0 195 22 22 0 192 22 22 0 192 22 22 0 195 22 22 0 204 23 23 0 221 23 23 0 254 
28 28 0 227 22 22 0 214 22 22 0 209 22 22 0 208 22 22 0 210 22 22 0 218 23 23 0 235 23 24 0 267 
28 27 0 213 22 22 0 203 22 22 0 200 22 22 0 198 23 22 0 202 23 23 0 212 23 23 0 230 24 24 0 264 
28 27 0 247 22 22 0 235 22 22 0 231 22 22 0 228 22 22 0 228 23 23 0 238 23 23 0 258 23 24 0 292 
========================

eeprom and frame looks like reasonable values, but after the To conversion only some pixel are getting 22-23 degree that is correct reading, all the other are strongly out of range. I also tried "lookup table method" but i'm getting same results.
I have several sensors and several boards, so i already tried to change batteries, board or sensors, but I got same mistakes in same pixels.
CC2538 has only 32kB ram but "size" command gives:
 text 	   data	    bss	    dec	    hex	filename
  81868	   1970	  22452	 106290	  19f32	host.elf
So it looks like ok.
Any suggestions?

slavysis, 2021-11-23:
Hi,

I tried the driver on several random MLX90640 devices and it is working. 
I also used the EEPROM and the frame data that you provided to feed it in the driver functions and check the outcome.
I also did manual calculations as described in the datasheet. Both the manual calculations and the driver agree that based on that input data the temperatures should be:

Vdd = 3.305
Ta = 25.431
To = 28.763,28.694,28.991,28.945,28.804,28.724,28.632,28.502,28.323,27.966,27.542,26.696,23.479,22.934,22.378,22.758,22.697,22.404,22.676,23.043,22.640,22.874,23.113,23.062,23.201,23.211,23.235,23.437,23.614,23.597,23.598,24.736,28.719,28.244,29.382,28.696,28.575,28.522,28.642,28.664,28.472,28.138,26.866,26.346,22.657,22.649,22.475,22.636,22.402,22.477,22.782,22.540,22.720,22.584,22.622,22.914,22.880,22.889,23.332,23.049,23.685,23.680,24.214,24.496,28.863,28.706,28.944,28.961,28.197,28.197,28.814,28.449,28.186,28.001,26.274,24.932,22.699,22.388,22.352,22.713,22.721,22.433,22.277,22.817,22.775,22.825,22.878,22.870,23.359,23.132,23.181,22.966,23.482,23.265,23.715,24.044,28.414,28.705,28.765,28.734,28.712,28.730,28.984,28.761,28.181,28.007,25.304,23.579,22.407,22.451,22.773,22.582,22.605,22.651,22.195,22.687,22.661,22.534,22.767,23.090,23.022,22.794,23.059,23.016,23.526,23.555,23.811,23.591,28.701,28.557,28.447,28.246,28.642,28.207,28.371,28.538,27.768,27.612,23.367,22.928,22.520,22.493,22.427,22.691,22.626,22.592,22.419,22.445,22.892,22.557,22.291,22.801,22.903,23.073,22.713,23.328,22.978,23.201,24.058,24.122,28.478,28.562,28.811,28.600,28.284,28.553,28.733,28.536,27.600,26.636,23.108,22.472,22.401,22.205,22.193,22.579,22.359,22.476,22.507,22.831,22.929,22.435,22.544,22.310,22.762,22.941,22.800,22.978,23.466,23.695,23.663,23.912,28.713,28.585,28.338,28.873,28.203,28.322,27.948,28.116,26.895,25.110,22.487,22.523,22.922,22.380,22.362,22.694,22.097,22.263,22.678,22.843,22.661,22.460,22.544,22.866,22.792,22.820,22.844,23.394,23.466,23.458,23.437,23.745,28.535,28.419,28.546,28.335,28.214,28.005,28.472,28.119,26.109,24.168,22.238,22.237,22.656,22.449,22.456,22.422,22.295,22.470,22.583,22.553,22.889,22.533,22.452,22.732,22.493,22.708,22.932,23.241,23.317,22.901,24.167,23.781,28.430,28.355,28.265,28.609,28.620,28.438,28.022,27.722,24.144,22.684,22.054,22.449,22.497,22.340,22.599,22.637,22.593,22.340,22.264,22.595,22.525,22.376,22.121,22.765,22.690,23.076,22.533,23.241,23.128,23.320,23.286,23.571,28.620,28.385,28.468,28.451,28.313,28.311,28.058,27.407,23.388,22.428,22.444,22.643,22.540,22.218,22.502,22.345,22.325,22.245,22.358,22.627,22.575,22.589,22.352,22.627,22.732,22.605,23.159,23.103,23.172,23.165,23.567,23.821,28.019,28.140,28.070,28.558,27.806,27.919,27.062,26.099,22.606,22.228,21.952,22.461,22.171,22.220,22.177,22.204,22.165,22.037,22.764,22.610,22.434,22.457,22.486,23.021,22.923,22.584,22.975,23.518,23.157,22.795,23.112,23.812,28.224,28.191,28.117,28.235,28.164,27.632,26.629,24.986,22.334,22.596,22.352,22.501,22.528,22.579,22.089,22.395,22.523,22.407,22.525,22.803,22.480,22.351,22.560,22.714,22.805,22.302,22.857,22.824,23.009,23.230,23.400,23.631,28.174,28.125,28.069,28.371,27.901,27.425,25.442,23.964,22.419,22.462,21.962,22.685,22.384,22.255,22.361,22.386,22.361,22.386,22.142,22.506,22.261,22.306,22.817,22.986,22.922,22.608,23.137,22.791,23.010,22.817,23.349,23.581,27.867,28.200,28.121,28.227,27.450,27.474,24.685,23.168,22.299,22.360,22.514,22.412,22.421,22.301,22.416,22.410,22.548,22.107,22.691,22.557,22.466,22.513,22.880,23.013,22.784,22.650,23.017,23.176,23.243,23.062,23.841,23.824,28.026,27.769,28.220,28.072,26.962,26.776,23.172,22.646,22.319,22.275,22.312,22.466,22.220,22.446,22.194,22.564,22.367,22.371,22.276,22.495,22.283,22.586,22.665,22.641,22.908,22.793,23.319,22.963,23.168,23.008,23.560,24.010,27.716,27.849,28.108,28.276,26.666,26.170,22.920,22.521,22.363,22.332,22.531,22.663,22.401,22.186,22.428,22.920,22.725,22.413,22.657,22.532,22.641,22.954,23.054,22.838,22.774,22.501,23.008,23.167,23.006,23.246,24.067,23.832,27.901,27.643,27.752,27.743,25.967,24.961,22.612,22.481,22.556,22.605,22.746,22.509,22.358,22.528,22.478,22.531,22.638,22.511,22.306,22.802,22.911,22.566,22.637,22.857,22.772,22.918,23.114,23.281,23.758,23.309,23.736,23.978,27.959,27.892,27.814,27.431,25.846,24.035,22.856,22.353,22.597,22.654,22.644,22.863,22.235,22.108,22.402,22.262,22.529,22.556,22.693,22.359,22.631,22.624,22.711,22.727,22.825,22.997,23.175,23.115,23.013,23.558,23.801,23.801,28.139,27.850,27.822,27.250,24.613,23.228,22.802,22.609,22.346,22.350,22.686,22.731,22.307,22.347,22.612,22.670,22.804,22.329,22.407,22.579,22.723,22.852,22.765,22.945,22.854,22.520,23.791,23.226,23.282,23.264,23.477,24.401,28.210,27.583,27.372,26.780,23.963,23.118,22.514,23.158,22.395,22.230,22.423,22.754,22.357,22.398,22.675,22.701,22.675,22.528,22.444,22.768,22.775,22.923,23.010,22.823,22.893,23.121,23.287,23.054,23.516,23.321,24.232,23.977,27.693,27.800,26.984,26.231,23.041,22.662,22.596,22.726,22.617,22.666,22.449,22.975,22.891,22.537,22.708,22.715,22.589,22.190,22.845,22.795,22.841,22.496,23.241,23.040,22.944,23.288,23.740,23.550,23.843,23.623,24.348,24.370,27.774,27.905,26.895,26.111,22.560,22.545,22.839,22.416,22.521,22.416,22.853,22.343,22.605,22.608,22.612,22.919,22.314,22.579,22.918,23.009,22.910,22.755,22.944,23.597,23.157,23.328,23.803,23.778,23.247,24.107,23.952,24.655,27.779,27.514,26.303,25.118,22.728,22.314,22.401,22.224,22.591,22.379,22.591,22.799,22.652,22.644,22.959,22.642,22.980,22.465,22.762,23.119,23.117,23.083,23.116,23.299,23.441,23.007,24.269,23.917,24.458,23.974,24.530,25.328,27.895,27.461,25.653,24.638,22.801,22.599,22.510,22.485,22.320,22.646,22.538,22.561,22.725,22.233,22.707,22.891,22.536,22.545,22.850,23.181,23.003,22.982,23.029,23.185,23.491,23.475,23.526,23.772,23.657,24.072,24.891,25.171

You mentioned that you started off of the MLX90641 deriver working. Is it possible that you somehow mixed-up the two drivers?

Best regards

luigi-grim, 2021-11-24:
Thank you very much for your time, I could focus on libraries since my reading is correct and I realized the T0 computation screws calculation when it starts using kTa and Kv arrays. In fact those arrays doesn't look right at all. I will look better into their calculation, maybe my C dialect makes something wrong, but I have a big doubt: according to the excel file with example values, kTa and kV should be both 0.something, so float, but in the params struct their type is actually int8_t. If I force those data to zero in the T0 calculation the reading is very good, probably with little errors here and there. Any idea why they are int in the params struct instead of float, as the excel example suggest they should be?

slavysis, 2021-11-24:
HI, indeed the xls file is not valid anymore - since the update for reducing the RAM footprint.
You first need to calculate the correct Vdd and Ta values. For the input data you shared Vdd should be 3.305 and Ta should be 24.31.
In your case you have error in Vdd (which maybe explains the error in Ta). So I would suggest that you first look into Vdd.

Best regards