melexis/mlx90640-library #92: strange Behaviour of MLX90640

Chimsons, 2022-02-25:
Hi, 
I use Mlx90640 with a SAMD21 microcntroller. My sensor works fine sometimes but it also behaves strangely:

1- the maximum temperature measured when there are no people and no hot objects in the room is 33, so there is no difference when there is a person in the room.  It's normal?
2- the maximum temperature sometimes rises to 50 and the minimum to 29, although the ambient temperature in the room is 23. I don't understand how.

I would appreciate any help. 
Thank you in advance

slavysis, 2022-03-09:
Hi,

This does not sound right. Can you share some more details and real data?
When you say 'the maximum temperature' do you refer to the average temperature of the whole frame of to the
temperature of a single pixel?
Are you sure there is no object in front of the sensor or some reflections (from mirrors, windows or other reflective materials)?

Best regards

Chimsons, 2022-03-09:
Hi,
I am referring to the temperature of a single pixel.
There was no person in the room.  There are no reflective materials, because I always did the test in the same room and I had to get good values.
Attached are the temperature values and details of the test conditions.
 
Thanks
[MLX90640_temp_value.txt](https://github.com/melexis/mlx90640-library/files/8213169/MLX90640_temp_value.txt)
 again

slavysis, 2022-03-09:
Hi,

Can you also share the full EEPROM and the frame data that resulted in these calculated temperatures?

Best regards

slavysis, 2022-03-09:
This is the image that you see:
[image]

It does not seem like a random spike of a single pixel, but like there is indeed some hot object in the field of view.
How much time after power-on did you get that measurement?
Do you have the capacitors between Vdd and Gnd pins as recommended in the datasheet?

Best regards

Chimsons, 2022-03-09:
I really can't remember how long after the power-on I took the measurement. 
Is there a minimum amount of time we have to wait after power-on before taking the measurements?

yes there are cpacitors between Vdd and Gnd. I used the Sparfun model.

Thanks

slavysis, 2022-03-09:
In order to achieve the specified accuracy one should wait 4min after POR. 

Best regards