melexis/mlx90640-library #97: Fisheye effect in image from MLX90640 (110FOV)?

ramkumarabhishek, 2022-08-03:
Hello,

We have started to notice a kind of fisheye effect in the IR image we capture from the MLX90640 (110FOV) unit, when the measured object is placed ~18mm away from the lens - Is this expected? If yes, what is the focal length of the lens used within the MLX90640 (110FOV) unit?

Setup: 
RPi4 connected to [Adafruit MLX90640 24x32 IR Thermal Camera Breakout - 110 Degree FoV](https://www.adafruit.com/product/4469#technical-details)
Code being used is based on that written by [Joshua Hrisko](https://makersportal.com/blog/2020/6/8/high-resolution-thermal-camera-with-raspberry-pi-and-mlx90640)

Any feedback / help would be much appreciated.

Thank you,

Abi

LBuydens, 2022-08-04:
Yes, this is normal for the integrated lens.  Lens parameters are confidential.

ramkumarabhishek, 2022-08-04:
> Yes, this is normal for the integrated lens.  Lens parameters are confidential.

Thanks for your feedback. 

Abi

ramkumarabhishek, 2022-08-04:
Question for @LBuydens -

Not knowing the exact confidential lens parameters, would calculating the focal length using the 110 degrees FOV and assuming an Equisolid / Equal Area Fish Eye projection of the image on the lens be a reasonable approximation? I don't need the calculation to be precise but accurate to a reasonable degree.

Please let me know your thoughts. 

Best,

Abi

ramkumarabhishek, 2022-08-05:
Thanks for your response, @LBuydens.

My objective is to correct for the fisheye distortion on an IR image of an object placed ~18 mm away the lens of the MLX90640 unit. Reason we need this correction is because we are trying to measure & average the temperature on specific areas of the object (to control the temperature on those areas specifically using a PID per area) and would like to map the real-word coordinates of the object with that seen in the distorted image. Specifically, since we are trying to measure & average the temperatures over 4 rectangles, we would like to be able to transform these rectangles into corresponding distorted quadrilaterals and use them for the real-time averaging as part of the PID feedback loop.

Any thoughts on my projection approximation approach or any alternative approaches would be much appreciated.

Best,

Abi

karelv, 2022-08-05:
@abi, if you have individual control of the heaters, why not firing them up one by one and monitor with mlx90640 the reaction. Doing so you account not only for distortion, but also for potential misalignments. Both in X - Y positioning and in rotation error around Z-axis (perpendicular to the PCB).

ramkumarabhishek, 2022-08-05:
Thanks @karelv. 

Since we are trying to control the temperature in ROI areas to be ~37-40C adjacent to the heaters (i.e. the heaters are not within the ROI), it is difficult to get a discrete definition of the ROI (or the heater for that matter) in the IR image. Also, since the object is plastic and not a PCB/metal, this makes the ROI definition in the image further difficult to discern, which is why we would like to geometrically calculate the distortion to define the ROI in the IR image.

Best,

Abi

P.S. I think you referenced the wrong user in your previous comment.