melexis/mlx90640-library #103: Parameter extraction failed, No useful data returned

maximmax-crypto, 2023-04-13:
Hi. 

I have the same issue as already asked: https://github.com/melexis/mlx90640-library/issues/13. For communication the STM32F407G is used. I have the ability to give you the needet data, so we can try to solve the problem together. The Function DumpEE returns 0, so i think the communication is astablished.  
After ExtractParameters i have the followind Parameters calculated: 
[image]

Are the ExtractParameter incorrect?

slavysis, 2023-04-19:
Hi, as a first step, could you please share the EEPROM dump data.

maximmax-crypto, 2023-04-19:
I cant sent you all the Data. But here a couple examples: 
[image]
[image]

maximmax-crypto, 2023-04-19:
I cant send you all 834 Words. But here some examples
[image]
[image]

slavysis, 2023-04-20:
Some of those values indeed look strange, Based on the EEPROM data you sent, I would say that this is a MLX90641 device that you are using (not MLX90640). Can you please check the marking to make sure what device you have.

P.S. You can log the full data in a csv file and attach it. It would be much easier than series of pictures

closed 2023-04-21 by maximmax-crypto