melexis/mlx90640-library #104: All values from the EEPROM are "FFFF", excepted for the first that is "00BF"

eleogrande, 2023-04-21:
Hi, i'm using the MLX90640 thermal camera on mbed on the STM32F429-Disc. The main function is as follows: 

#include "mbed.h"
#include <MLX90640_I2C_Driver.h>
#include <MLX90640_API.h>

#define FRAMES_NUM 15
#define TA_SHIFT 8

Serial pc(USBTX, USBRX);// tx, rx, Virtual serial port over USB
DigitalOut greenLED(LED1);

void PrintRawData(uint16_t *p);

int main()
{
    uint8_t printme=1;
    pc.baud(115200);
    greenLED = !greenLED;
    MLX90640_I2CFreqSet(1000000);
    
    paramsMLX90640 mlx90640;
    uint8_t slaveAddress = 0x33;
    uint16_t *pEE;
    uint16_t *pFrame;
    float *toFrame;
    static uint16_t eeMLX90640[832];
    static uint16_t mlx90640Frame[834];
    static float mlx90640To[768];
    float emissivity = 0.95;
    float eTa;  //Ta for emissivity compensation
    float vdd; //voltage of the circuit inside the camera
    int status;
    
    pEE = eeMLX90640;
    pFrame = mlx90640Frame;
    toFrame = mlx90640To;

    status=MLX90640_SetRefreshRate(slaveAddress,0x07);//64Hz
    printf("status_SetRefreshRate: %d\n", status);
    status=MLX90640_SetResolution(slaveAddress,0x03);
    printf("status_SetResolution: %d\n", status);
    status=MLX90640_SetInterleavedMode(slaveAddress);
    printf("status_InterLMode: %d\n",status);
    status=MLX90640_DumpEE(slaveAddress, pEE);
    printf("status_DumpEE: %d\n", status);
    PrintRawData(pEE);

    status=MLX90640_ExtractParameters(pEE, &mlx90640); 
    
    PrintRawData(pEE);
       
        while (true){
            wait_us(500000);
            greenLED = 1;    // turn on LED until initialization is complete
            MLX90640_GetFrameData(slaveAddress, pFrame);
            greenLED = 0;    // turn on LED until initialization is complete
            eTa = MLX90640_GetTa(pFrame, &mlx90640) - TA_SHIFT;
            MLX90640_CalculateTo(pFrame, &mlx90640, emissivity, eTa, mlx90640To);
            MLX90640_GetImage(pFrame, &mlx90640, toFrame);
            
            if (printme)
            {            
                PrintRawData(pFrame);
            }
        }

}

void PrintRawData(uint16_t *p)
{
    for(int i=0; i<832; i++) {
        pc.printf("%04X,",*p++);
    }
    pc.printf("\r\n");
    
}

I added to the project the files "MLX90640_API.h", "MLX90640_API.cpp", "MLX90640_I2C_Driver.h", and "MLX640_SWI2C_Driver.cpp" provided by you. When I try to debug the code, it seems to hang when I call the function "MLX90640_ExtractParameters(pEE, &mlx90640)", so I thought the problem was in the parameter "pEE", which is the EEPROM data, and which actually turns out to be incorrect. Is it possible that the code hangs when I call the function "MLX90640_ExtractParameters(pEE, &mlx90640)" due to incorrect EEPROM data?

slavysis, 2023-04-24:
Hi, 
The code should not hang, unless there is something awfully wrong with the memory allocation and you somehow mess-up the stack. I would suggest that you start by fixing the communication and getting proper data.
You are using the SW I2C lib and this one requires that you trim the SCL frequency.
The I2C speed when reading the EEPROM should be no more than 400KHz, So you should do something like:
MLX90640_I2CFreqSet(400000);
status=MLX90640_DumpEE(slaveAddress, pEE);
MLX90640_I2CFreqSet(1000000);

eleogrande, 2023-04-24:
Hi, thanks a lot for the answer!!
I tried to do what you say, having also seen it in the "main" of one of the
posts here on github, but it still doesn't work. What i explained at the
beginning of this posts happens during debugging, while during the "run" i
get problems before getting the data back from the EEPROM. In particular,
the set refresh rate and resolutions functions return values such as
-1(NACK), -2 ("the data is not the intended one" what does it means?) or
1(what does it means?). During debugging, the same functions return
correctly 0.
I'm using the mbed online platforms, could there be any problems related to
this?

slavysis, 2023-04-24:
Hi,

-2 means that the value written via I2C in a register is not the one that you wanted to write.
Using the mbed on-line compiler is not a problem.
It does seem like you have issues with the i2c communication.

Best regards

eleogrande, 2023-04-24:
Hi,
i'm sorry but i don't have proper instrumentation. Can i do it on mbed on-line compiler?

slavysis, 2023-04-24:
Hi, 
I am not sure I can think of a way to check the if the I2C waveforms are correct without having some sort of a visualization. Do you have a logical analyzer?

Best regards

eleogrande, 2023-04-24:
Hi,
sorry my instrumentation consist only in the camera and the board. So i think i cannot have some visualization. Is there a way to do it without an external device? Do you reccommend to use the "MLX90640_I2C_Driver.cpp" or "MLX90640_SWI2C_Driver.cpp" file? I take this opportunity to ask if the other files are correct. This is the MLX90640_API.cpp: 
 
#include <MLX90640_I2C_Driver.h>
#include <MLX90640_API.h>
#include <math.h>
#include <stdio.h>
#include <chrono>

void ExtractVDDParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractPTATParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractGainParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractTgcParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractResolutionParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractKsTaParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractKsToParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractAlphaParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractOffsetParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractKtaPixelParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractKvPixelParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractCPParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
void ExtractCILCParameters(uint16_t *eeData, paramsMLX90640 *mlx90640);
int ExtractDeviatingPixels(uint16_t *eeData, paramsMLX90640 *mlx90640);
int CheckAdjacentPixels(uint16_t pix1, uint16_t pix2);
float GetMedian(float *values, int n);
int IsPixelBad(uint16_t pixel,paramsMLX90640 *params);

int MLX90640_DumpEE(uint8_t slaveAddr, uint16_t *eeData)
{
     return MLX90640_I2CRead(slaveAddr, 0x2400, 832, eeData);
}

int MLX90640_CheckInterrupt(uint8_t slaveAddr)
{
    uint16_t statusRegister;
    MLX90640_I2CRead(slaveAddr, 0x8000, 1, &statusRegister);
    return (statusRegister & 0b1000) > 0;
}

void MLX90640_StartMeasurement(uint8_t slaveAddr, uint8_t subPage)
{
    uint16_t controlRegister1;
    uint16_t statusRegister;
    MLX90640_I2CRead(slaveAddr, 0x800D, 1, &controlRegister1);
    controlRegister1 &= 0b1111111111101111;
    controlRegister1 |= subPage << 4;
    MLX90640_I2CWrite(slaveAddr, 0x800D, controlRegister1);
    MLX90640_I2CRead(slaveAddr, 0x8000, 1, &statusRegister);
    statusRegister &= 0b1111111111110111; // Clear b3: new data available in RAM
    statusRegister |= 0b0000000000110000; // Set b5: start of measurement
                                          // Set b4: enable RAM overwrite
    MLX90640_I2CWrite(slaveAddr, 0x8000, statusRegister);
}

int MLX90640_GetData(uint8_t slaveAddr, uint16_t *frameData)
{
    int error = 0;
    uint16_t statusRegister;
    uint16_t controlRegister1;

    // Get page data
    error = MLX90640_I2CRead(slaveAddr, 0x0400, 832, frameData);

    // Get status reguster
    MLX90640_I2CRead(slaveAddr, 0x8000, 1, &statusRegister);

    // Get control register
    MLX90640_I2CRead(slaveAddr, 0x800D, 1, &controlRegister1);

    frameData[832] = controlRegister1;
[... 1622 lines cut ...]

    int MLX90640_SetDeviceMode(uint8_t slaveAddr, uint8_t deviceMode);
    int MLX90640_SetSubPageRepeat(uint8_t slaveAddr, uint8_t subPageRepeat);
    int MLX90640_SetSubPage(uint8_t slaveAddr, uint8_t subPage);
    int MLX90640_CheckInterrupt(uint8_t slaveAddr);
    void MLX90640_StartMeasurement(uint8_t slaveAddr, uint8_t subPage);
    int MLX90640_GetData(uint8_t slaveAddr, uint16_t *frameData);
    int MLX90640_InterpolateOutliers(uint16_t *frameData, uint16_t *eepromData);
    int MLX90640_SynchFrame(uint8_t slaveAddr);
    int MLX90640_TriggerMeasurement(uint8_t slaveAddr);
#endif

This is "MLX90640_I2C_Driver.h":

#ifndef _MLX90640_I2C_Driver_H_
#define _MLX90640_I2C_Driver_H_

#include <stdint.h>
#include "MLX90640_API.h"
    extern void MLX90640_I2CInit(void);
    extern int MLX90640_I2CGeneralReset(void);
    extern int MLX90640_I2CRead(uint8_t slaveAddr,uint16_t startAddress, uint16_t nMemAddressRead, uint16_t *data);
    extern int MLX90640_I2CWrite(uint8_t slaveAddr,uint16_t writeAddress, uint16_t data);
    extern void MLX90640_I2CFreqSet(int freq);
#endif

eleogrande, 2023-04-24:
The most stranger thing is that during debugging i have correctly 0 from the calibration functions, but during the run returns other values. Could you explain why? I'm sorry for the many questions and thank you a lot for the help.
Best regards.

slavysis, 2023-04-26:
What do you mean when you say 'debugging'? Is it step-ing through the code?
My guess would be that there is something wrong with the timings.

eleogrande, 2023-04-26:
Hi,
Thanks for the answer. Yes, i mean that when i run the code in debug mode
(step-ing through the code) the calibration's functions return correctly 0;
but if i don't run it in debug mode, they're turn -1.

eleogrande, 2023-04-26:
I tried to use the oscilloscope but i can't see any waveform with the file i uploaded here. But using this code: 

#include "mbed.h"

Serial pc(USBTX, USBRX);
#define WAIT_TIME_MS 1000
I2C i2c(PC_9, PA_8); // sda, clk

int main() {
    
    pc.baud(115200);
    i2c.frequency(100000);     
    
    while (1) {
        
        pc.printf("\033[0m\033[2J\033[HI2C Searching!\n\n\n"); 
        int count = 0;
        pc.printf("Starting....\n\n");
        
        for (int address=0; address<256; address+=2) {
            if (!i2c.write(address, NULL, 0)) { // 0 returned is ok                
                pc.printf("I2C address 0x%02X\n", address);
                count++;
            }
        }        
        pc.printf("\n\n%d devices found\n", count);
        WAIT_TIME_MS;       
    }
}

i get these two waves: the first refers to the SCL pin and the second to the SDA pin. 

[image]

Going back to the main project, i have another question. When i run the code in debug mode, as mentioned above, the calibration functions correctly return 0, but the "MLX90640_ExtractParameters(pEE, &mlx90640)" functions seems to crash. Is it always due to timing?

Thank you so much for your availability and help, any suggestion on what i could do is always welcome. 
Best regards.

slavysis, 2023-04-27:
This is using the hardware I2C of the MCU, and you were trying to use the SW I2C driver. Make sure that your configuration is correct. If you want to use the HW I2C module of the MCU you should use the "MLX640_I2C_Driver.cpp" and implement the i2c read/write functions accordingly. You should definitely first make sure that your communication is working. Now that you have an oscilloscope, this should be a rather straight forward thing to do.