melexis/mlx90640-library #106: MLX90640 eeprom data 0xFFFF after 20 words

BobPutten, 2023-05-23:
Hi,
I converted the mlx90640-library from the original C code to C#. Testing at RPi. with I2C running at 400kbit/s
I notice that after the MLX90640_DumpEE() function, the first part of the eeprom buffer (unsigned integers) do have a value.
I recognise eeprom [16] = 49691 => 0xBE33. This is the device I2C address (0x33) and according the MLX90640-Datasheet-Melexis, table 8.
However after address 20, I can see only 0xFFFF.
ExtractVDDParameters() will result in a kVDD of -32 and vdd25 of -8224. That seems odd to me..
Also the ExtractAlphaParameters() function will freeze in 
`while(temp < 32767.4)
    {
      temp = temp*2;
      alphaScale = (byte)(alphaScale + 1);
    }
 `
Any hints what could be wrong here?

conara, 2023-10-18:
I have have the same issue with the original C code. Did you found the problem? Or can somebody help me debugging this problem?

karelv, 2023-10-18:
It might be an issue with the I2C buffer length.
For an arduino based solution see here:
https://github.com/melexis/i2c-stick/blob/main/i2c-stick-arduino/mlx90640_i2c_driver.cpp#L57
Maybe it can inspire you for C#.