melexis/mlx90640-library #107: MLX90640 Parameter Check - MLX90640_BROKEN_PIXELS_NUM_ERROR

ryanlederhose, 2023-07-05:
When checking the parameters initially I get a MLX90640_BROKEN_PIXELS_NUM_ERROR. Through debugging I found the brokenPixCnt=4 in the ExtractDeviatingPixels function. I am using the Teensy 4.1 and I have also attached a dump of EEPROM dump:

[image]

Any ideas on how to fix this?

Thanks

slavysis, 2023-07-05:
Is it possible to give the EEPROM dump in a format that would allow to easily work with it? Extracting over 800 values from a picture is a bit too inconvenient.

ryanlederhose, 2023-07-05:
Of course. Thanks for the reply. Might this be easier?

eeMLX90640 Array:
Index 0  194
Index 1  31135
Index 2  0
Index 3  8289
Index 4  5
Index 5  800
Index 6  992
Index 7  2346
Index 8  46834
Index 9  393
Index 10  1177
Index 11  0
Index 12  0
Index 13  0
Index 14  0
Index 15  48691
Index 16  17184
Index 17  65465
Index 18  257
Index 19  257
Index 20  257
Index 21  61937
Index 22  57585
Index 23  53200
Index 24  0
Index 25  1
Index 26  2
Index 27  61698
Index 28  61698
Index 29  61698
Index 30  57586
Index 31  49377
Index 32  35221
Index 33  14850
Index 34  61148
Index 35  4111
Index 36  4369
Index 37  4386
Index 38  273
Index 39  65280
Index 40  0
Index 41  15
Index 42  8738
Index 43  13107
Index 44  8755
Index 45  4386
Index 46  60928
Index 47  43997
Index 48  6197
Index 49  12201
Index 50  8535
Index 51  42369
Index 52  30294
Index 53  63940
Index 54  13877
Index 55  13105
Index 56  9298
Index 57  3193
Index 58  6067
Index 59  1573
Index 60  62976
Index 61  38807
Index 62  38807
Index 63  11002
Index 64  9152
Index 65  4304
Index 66  8398
[... 740 lines cut ...]
Index 807  63504
Index 808  60178
Index 809  63184
Index 810  58254
Index 811  62448
Index 812  60304
Index 813  63312
Index 814  57086
Index 815  864
Index 816  63394
Index 817  62400
Index 818  55326
Index 819  3040
Index 820  65458
Index 821  62416
Index 822  58254
Index 823  2880
Index 824  65458
Index 825  65264
Index 826  60382
Index 827  0
Index 828  834
Index 829  65296
Index 830  60238
Index 831  5008

closed 2023-09-03 by ryanlederhose