melexis/mlx90640-library #110: How interpreting data from MLX90640_GetImage ?

AsdyCorp, 2023-10-12:
If CalculateTo gives us float values of object temperature in celisus of "pixels" how to interpreting data from GetImage ? what are ranges and what they mean?

LBuydens, 2023-10-12:
The values obtained from GetImage indicate only if something is hot(ter) or cold(er). Positive values are hotter than the sensor, negative are colder. But this is not calibrated and not a linear scale.

AsdyCorp, 2023-10-12:
> The values obtained from GetImage indicate only if something is hot(ter) or cold(er). Positive values are hotter than the sensor, negative are colder. But this is not calibrated and not a linear scale.

So we can compare them to Ta ? and theoretically - if we put dry ice or some other cold agent near sensor it must improve it sensitivity, especially in GetImage  function?

LBuydens, 2023-10-12:
It's relative to Ta. You will not gain that much by cooling.

AsdyCorp, 2023-10-12:
> It's relative to Ta. You will not gain that much by cooling.

but with coolling all values from GetImage will be positive then, right?

LBuydens, 2023-10-12:
Yes, but if that's the goal, better just add enough.

closed 2023-10-12 by AsdyCorp